genodelabs/genode #3963: platform_drv: support restarting drivers

alex-ab, 2020-12-03:


alex-ab, 2020-12-10:
On the issue_3963 branch the commits are ready for review. The commits named tmp or debug in the beginning just ignore. Some of the commits we may also squash together, depending on your suggestions. Functionally wise, I tested the restarting now on X201, FUJI3 and T460p. Drivers like nic, wifi, intel graphic, usb and ahci still work after restarting them forcefully. The audio driver is still fickle, on X201 now it works ever after restart, but don't like to cooperate on the other test machines by now.

m-stein, 2020-12-10:
@alex-ab It's really cool to hear that! I'm looking forward to give restarting the NIC driver a try next time I test Sculpt with the WIP Uplink-session commits.

nfeske, 2021-01-07:
The patch series looks already very very nice! It is great that you split up the change into multiple commits, e.g., adding the timing facility as a separate step.

I left a few annotations inline. These are merely cosmetic. They do not necessarily need to be applied to the existing patches (which would probably cause a rebasing hassle). I'd be perfectly fine with applying those touches as a follow-up commit.

alex-ab, 2021-01-08:
Thanks. I updated the commits in the issue_3963 branch where possible in place, and added a separate commit to use not the Register framework in the Bdf type. Using your tag-value mimic causes trouble I could not resolve. In the DEBUG commit I added a very similar structure where the assignment succeeds, but using the Bdf type the compiler don't accept it.

nfeske, 2021-01-08:
Thanks for the quick response.

> In the DEBUG commit I added a very similar structure where the assignment succeeds, but using the Bdf type the compiler don't accept it.

It only works if the struct has no constructor. So please try to remove the constructor and construct the struct via the syntax I sketched above.

In cases where you need to construct a `Bdf` object differently, like with a bdf value in your example, use a static class function. I suggest to call the function `from_value`. So at the caller side, it looks like:

```
Bdf const bdf = Bdf::from_value(0x1234);
```

alex-ab, 2021-01-08:
Thanks, that helped. I updated the commit accordingly.

nfeske, 2021-01-08:
Cool! I merged your commit series to staging now.

commit 0e1a042d (2021-01-08): platform_drv: add support to power PCI devices

- depending on available PCI power cap power off and on
- save and restore PCI bars if required
- reset PCI devices after power on if supported

Fixes #3963
commit b185f3fa (2021-01-25): platform_drv: add support to power PCI devices

- depending on available PCI power cap power off and on
- save and restore PCI bars if required
- reset PCI devices after power on if supported

Fixes #3963
closed 2021-01-25 by nfeske