genodelabs/genode #3964: vm_session(x86): support cstar register in VMMs

alex-ab, 2020-12-03:


alex-ab, 2020-12-03:
d4d821ccf2cb030b7ebabad00a635163117caf98 adds the support

nfeske, 2020-12-04:
Would you mind sharing the motivation behind this change?

alex-ab, 2020-12-04:
If you have a 64bit guest running applications in [compatibility mode](https://en.wikipedia.org/wiki/X86-64#:~:text=In%2064%2Dbit%20mode%2C%20instructions,bit%20operating%20system%20supports%20them), this register is used and must be stored/restored by the VMM.

nfeske, 2020-12-04:
> If you have a 64bit guest running applications in compatibility mode, this register is used and must be stored/restored by the VMM.

This information I can easily get from Google. Actually, I did. But that is not my point.

What I don't get from Google:

- Why wasn't that a problem until now?
- What is the practical problem *you* encountered?
- Why is it worth solving?
- Which concrete scenario (including guest / VMM / hardware) has triggered the problem?

When opening an issue - any issue - it is very tactful to supply this information upfront.

alex-ab, 2020-12-04:
The story is that during investigating another issue I noticed that we don't handle this 64bit register, on the one hand not in the vm_session interface nor by some supported kernels. After reading the CPU specs, recognizing that this is required in some cases and later noticing that my Linux VM runing with VBox5 on my Intel machine actually uses/writes this register (see https://elixir.bootlin.com/linux/latest/source/arch/x86/kernel/cpu/common.c#L1708) I decided to address that. This feature issue is meant to avoid any headaches or weeks for bug hunting to whomever. For Seoul the utcb header was adjusted so that the register layout is the same as for usage with VBox5/nova.

nfeske, 2020-12-04:
Thank you for the explanation. This makes the issue/commit tangible to me.

nfeske, 2020-12-07:
Merged to staging.

commit 02ebfabd (2020-12-07): vm_session(x86): support cstar register

Fixes #3964
closed 2020-12-09 by nfeske