genodelabs/genode #3965: vbox5: add support for 64bit SVM

alex-ab, 2020-12-06:
The feature is up to now not enabled nor tested. With the event of Sculpt OS I would like to use it also in my private context, which is dominated by AMD machines. The issue is WIP and will collect all adjustments required to get eventually to the desired goal. 

Regarding the current SVM state support: Solely Seoul works for me on AMD with Sculpt 20.08, albeit it solely supports 32bit guests, e.g. the Firefox browsing VM. VBox5 on Sculpt 20.08 quits (either 32/64 bit guest) quickly with unimplemented VM exits.

alex-ab, 2020-12-06:
Back from (online) concert is right time to push a working state.

alex-ab, 2020-12-06:
With the commits I can run vbox5_ubuntu_16_04 and vbox5_vm_ubuntu_16_04 and  surf a bit (Ryzen 1800X, MSI B450-A pro board). The timing seems still to be odd, it not ever boots to end - but a good intermediate result.

alex-ab, 2021-01-16:
The commits on issue_3965_staging are ready for review. I could run successful vbox5_*_64.run on a native AMD machine and was able to install a 64bit VM on AMD on a backported Sculpt 20.08 image.

chelmuth, 2021-02-02:
I was about to merge 084660c466ea19fbfbc779ebca676f6127f299f2 but discovered that it updates NOVA with two commits.

98b501b Make FPU eager switching default
24bef22 AMD: add asid allocator for vCPUs

Is the *eager FPU switching* change closely related to this issue? Do I understand correctly that this change affects our tests (and Sculpt as well) as we didn't set the `fpu_eager` command-line switch before?

alex-ab, 2021-02-02:
> Is the _eager FPU switching_ change closely related to this issue?

Yes, without it the VMs don't run stable.

> Do I understand correctly that this change affects our tests (and Sculpt as well) as we didn't set the `fpu_eager` command-line switch before?

No. For Intel CPUs the fpu_eager was ever taken, independent on any command line. The commit does not affect our tests.

chelmuth, 2021-02-02:
Thanks for the clarification. Merged all commits to staging.

chelmuth, 2021-02-03:
The current staging unexpectedly reboots in Qemu (version 4.2.1) on *run/seoul-auto*.

alex-ab, 2021-02-03:
I will check asap.

chelmuth, 2021-02-03:
> I will check asap.

No sweat, I put it here for the records only - I did not expect your response before Monday for reasons. ;-)

alex-ab, 2021-02-09:
> The current staging unexpectedly reboots in Qemu (version 4.2.1) on _run/seoul-auto_.

Qemu 4.2.1: Even before the commit, it reboots occasionally if one would choose fpu_eager on commandline.
Qemu 3.1.1: It works, with this commit and before the commit with fpu_eager.
AMD Hardware 2x: It works.

With Qemu 4.2.1 it works for me when using 1 CPU but 2 vCPUs. I fear a bit that debugging Qemu to find the diff between 3.1.1 and 4.2.1 will take time, similar as I had to do like in https://github.com/alex-ab/seoul/issues/4 resp #3373.

For now, I would suggest to use 109ca7ad866a1271ed07604a06298714a8c48485. Using 2 CPUs in Qemu accidentally slipped in during 9812799b242758df412bdc1b53d325d8eb46b016.

chelmuth, 2021-02-09:
Even though I was impressed by your persistency and skills when you fixed alex-ab/seoul#4 in the Qemu sources, I share your sentiment that we should be economical with our resources. Therefore, I followed your suggestion and merged 7d8e95ab532a1a835377f6edfc99104e2a94ff1a.

commit ae0090ad (2021-02-02): vbox5: enable 64bit SVM support

Fixes #3965
closed 2021-02-23 by nfeske