genodelabs/genode #3966: Sculpt doesnt boot on Lenovo Ideacentre 3 07ADA05

ucodelukas, 2020-12-08:
Host: 90MV009MGE IdeaCentre 3 07ADA05 

CPU: AMD 3020e with Radeon Graphics (2) @ 2.600GHz 
GPU: AMD ATI 03:00.0 Picasso 
Memory: 1411MiB / 9963MiB

alex-ab, 2020-12-08:
The machine does not fulfill the [hardware requirements](https://genode.org/documentation/articles/sculpt-20-08#Hardware_requirements_and_preparations) for Sculpt 20.08. I can only advise to choose a machine which are closer to the requirements. Optionally, you may try the image from me from #3918 which contains some (inofficial) updates for 20.08 which helped already according to @dckc. If it still not works, we would need your support in form of serial output from your machine (either you have a serial connector already or you have to plugin a PCI-to-serial card). If you need help with that, please ask.

nfeske, 2021-02-26:
I close this issue because there is currently no plan to officially support AMD.

closed 2021-02-26 by nfeske