genodelabs/genode #3967: libc: integration test

trimpim, 2020-12-09:
I created a test program which stresses the interplay between `libc`, `pthreads` and `vfs_pipe` and may detect regressions.

The program starts a thread that spawns a bunch of workers, sends and receives data to them through a pipe. When all data of a worker is collected, the worker is finished (`join()`) and a new worker is started. Both the number of bytes sent to the worker and the size of its answer are random.

The program can also be easily compiled for native Linux (see `README`).

On 20.08, the test hanged after a few hundred workers (on `base-linux`). With the changes regarding the libc after 20.08, the program now runs for more than three days and starts/stops several hundred thousand workers during that time (also on `base-linux`).

Things I discovered using this program are:

- on `base-hw` (both `imx8q_evk` and `x86_64`) the program exhausts capabilities after about 12500 started/stopped workers
- on `base-nova` (`x86_64`) the program hangs after about 750 started/stopped workers
- both `base-hw` and `base-nova` hangs right in the beginning if 23 workers are started in parallel

In order to be able to use this test in the continuous testing infrastructure, the following things have to be considered in particular:

- I did not create a depot-autopilot test yet. Some things would need to be adjusted (number of parallel workers, total number of workers etc.).
- Currently, both the total number of workers and the test timeout are way to big to be useful in a nightly test, as I wanted to be sure, that the program can run a long time.

nfeske, 2020-12-16:
Thank you @trimpim for this interesting report and for providing such a comprehensive test.

Your findings regarding the differences of the base platforms are particularly interesting and should motivate us to track down the differences.

I think that a stress test like this should better stay at a dedicated run script so it can be tuned for different kernels and run separately from the nightly depot-autopilot tests (e.g., at a lower rate, like once a week). In contrast, the depot-autopilot test suite should retain the nature of relatively "quick" running tests.

@alex-ab I'd appreciate you reviewing @trimpim's commit, drafting a plan of how we could potentially integrate it into our testing, and in particular looking into the described effects that occur on NOVA and base-hw. Would you be up to that?

alex-ab, 2020-12-16:
> @alex-ab I'd appreciate you reviewing @trimpim's commit, drafting a plan of how we could potentially integrate it into our testing, and in particular looking into the described effects that occur on NOVA and base-hw. Would you be up to that?

Yes, i will take care of it and investigate the effects.

alex-ab, 2020-12-17:
> * both `base-hw` and `base-nova` hangs right in the beginning if 23 workers are started in parallel

I could track down the reason for this behavior. A pthread (A) creates a lot of worker and acquires/releases a the lock in [add_worker()](https://github.com/trimpim/genode/commit/9df249ab5afd2307d1e6b3c136c211ab0b701661#diff-46fb23004f5e8041078d1feefba2246a1d95f9cadc46fb4144085f6da62e31a5R94). Concurrently to this startup process, another pthread (B) blocks eventually in a pipe with a write() while holding the very same lock in [run()](https://github.com/trimpim/genode/commit/9df249ab5afd2307d1e6b3c136c211ab0b701661#diff-46fb23004f5e8041078d1feefba2246a1d95f9cadc46fb4144085f6da62e31a5R117). It seems that pthread A will do after the creation of all worker the select() and read of the pipes. Since pthread A is however blocked on the hold lock by pthread B, no more progress will be done. @trimpim: For me it looks like, that you should not hold a lock during going into a blocking write, or the startup worker & read pipe procedure must be restructured to avoid this deadlock or you need a non blocking pipe (e.g. pipe2 with O_NONBLOCK).  Can you please adjust the test to your desired behavior.

I will try, in the meantime, to examine the reason for the exhausts capabilities, which may take a while.

trimpim, 2020-12-17:
@alex-ab thanks for the report, I will look into this, and will post a patch that corrects this.

trimpim, 2020-12-18:
@alex-ab I have changed the pipes to be `O_NONBLOCK`. the tests (`base-nova`, base-hw` and `base-linux` on x86_64) are currently running on my systems. I will post if I have further findings.

trimpim, 2020-12-21:
I did let the tests run over the weekend (`base-hw`, `base-linux` and `base-nova` all on `x86_64`).
`base-nova` (358k workers) and `base-linux` (3.5M workers) are still running, on `base-hw` there was a resource request after about 10k workers.

alex-ab, 2021-01-04:
It seems that in 'base-hw' is an constant (but small) cap leakage. The (seldom) increase of consumed caps arises from https://github.com/genodelabs/genode/blob/master/repos/base-hw/src/lib/base/env_deprecated.cc#L86. Since the process is long running and some kind of release is not available, the test must run out of caps eventually. The reason for the steady increase is still unclear.

alex-ab, 2021-01-05:
At https://github.com/genodelabs/genode/blob/master/repos/base-hw/src/core/kernel/thread.cc#L695 deletions of capabilities in use are denied by the kernel and skipped. According to instrumentation these skipped caps are not deleted at any time in the future (d037216a9a1f3b79f68bef6701a1cbd0935d6670 test commit). Is my analysis correct ?

skalk, 2021-01-06:
@alex-ab omg, how embarrassing! You're absolutely right. In the original commit, I've introduced the capability reference counting without the defered deletion.

alex-ab, 2021-01-06:
I think I found the place where some ack_cap syscall is missing, namely on thread construction. Because of the missing syscall, it seems that the in kernel cap structure stays in the in_utcb() state. With debb83bb5634ed30e19561b383c99ddf76b780f0 the debug numbers of the cap slab are now stable for me. I don't know whether 04852ac37b4d001d2aa62a44371a081caa34be8e is necessary, but the pattern looks similar. @skalk: what do you think ? @trimpim: please try the commit(s) whether it helps for your base-hw runs

trimpim, 2021-01-07:
@alex-ab I'm currently running the `base-hw` test on `x86_64` with debb83bb5634ed30e19561b383c99ddf76b780f0. So far it is at over 27k workers and is still running. I'll let it run while doing other work to see if an issue comes up later.

m-stein, 2021-01-07:
27k :-) :+1:

skalk, 2021-01-07:
@alex-ab very nice that you provide a solution here. I'm fine with the commit. The only question to me was in the fixup commit, whether the validity check is necessary at all. I would think those capabilities are always meaningful, but even if they are invalid, the kernel syscall should be robust against it. But this is to save few if condition branches only, and not really important.

alex-ab, 2021-01-08:
Ok, I added 9a445797f7739ab0e910ae2308cecc22897a5241 without the checks.

chelmuth, 2021-01-08:
@alex-ab would you mind to move the base-hw fix into a separate issue marked fixed when solved?

alex-ab, 2021-01-11:
> @alex-ab would you mind to move the base-hw fix into a separate issue marked fixed when solved?

Ok, see #3982.

trimpim, 2021-04-16:
How should I proceed with this issue? Can I support you in any way to bring it to staging/master?

alex-ab, 2021-04-16:
> Can I support you in any way to bring it to staging/master?

If you could adjust the test in a way that it terminates after some configurable amount of rounds, I can take care to integrate it into our weekend testings.

trimpim, 2021-04-22:
@alex-ab I have rebased to the latest master and added parameters to the test program. The current run scripts sets 10000 iterations, wich takes about 10 minutes on Linux and 2 hours on Nova.

alex-ab, 2021-04-22:
@trimpim: Thanks a lot, I'm going to integrate the test tomorrow.

alex-ab, 2021-04-23:
I adjust the run script parameters on the issue_3967 branch and restricted for the moment to some few platforms, to see how it does over the weekend. With commit 608587264d36c3e5c0bf11008135858fcc6b5f03 the runtime also got better for the microkernels. @chelmuth, @nfeske: please consider for staging

chelmuth, 2021-04-23:
> I adjusted the run script parameters on the issue_3967 branch and restricted for the moment to some few platforms, to see how it does over the weekend.

That's great. I merged the series and are looking forward to the results.

> With commit [6085872](https://github.com/genodelabs/genode/commit/608587264d36c3e5c0bf11008135858fcc6b5f03) the runtime also got better for the microkernels.

This sounds promising at first but I'm still uncertain that the pthread-placement mechanism works as transparent as required. Now an explicit placement strategy (ie., *policy*) is silently extended by automatic round-robin placement if pthread IDs become greater than what is expected in the configuration. In my opinion we have to clearly specify the mechanism (incl. corner cases) and also have to replace the statically-sized map array by a dynamic policy database that reflects the specification. As always I'd vote for the way of least surprise for users.

alex-ab, 2021-04-26:
@chelmuth: 251c599e1476f37a7d46ba66bc05ad9b1b0048c2 fixes the build issues for -O0

alex-ab, 2021-04-27:
The results over the weekend are mixed:

linux_64: succeed one time, failed another time with segmentation fault
linux_32: failed 2x times (timeout, early no progress in the log)
x86_64 (hardware): nova && hw - succeeded 2x times
imx8 (hardware): hw - succeeded 2x times
x86_32 (hardware): nova failed 2x times (spawn died unexpected)

The runtime on hardware was too short, so we may increase the round numbers (just 1minute for 1000 iterations)

alex-ab, 2021-04-27:
> This sounds promising at first but I'm still uncertain that the pthread-placement mechanism works as transparent as required. Now an explicit placement strategy (ie., _policy_) is silently extended [...]

Agree. I will come up with a fixup according to our offline discussion.

alex-ab, 2021-04-27:
> Agree. I will come up with a fixup according to our offline discussion.

@chelmuth, please check d427b0dc8066ff4db23e8b15ec1dfd12335b858e

chelmuth, 2021-04-28:
The commit looks quite good already (please see also my commit comment).

What do you think about combining all placement-policy related functions/types into one class in contrast to the current scattered implementation? For illustration, I imagine something like

``` c++
struct Placement_policy
{
  struct Placement { /* as already defined */ };
  enum class Policy { /* as already defined */ };

  Policy _policy;
  Registry<Registered<Placement>>  _placements;

  Placement_policy(Policy) { }

  unsigned placement(unsigned pthread) const { }

  void add_placement(unsigned pthread, unsigned cpu) { }
};
```

nfeske, 2021-04-28:
Maybe related, I noticed that the pthread test reproducibly fails for base-hw on the two ARMv7 boards imx7d_sabre and imx6q_sabrelite with the current staging branch. In both cases, there is no error but the test silently times out. It works fine on all other board / kernel combinations. Could this problem be related to the pthread-affinity change? I haven't bisected the problem but ten days ago, before we added this change to staging, the test used to be fine.

```
[init -> test-pthread] main thread: test without timeouts
[init -> test-pthread] signaller: started
[init -> test-pthread] waiter: started
[init -> test-pthread] waiter(main thread): started
[init -> test-pthread] waiter: started
```

alex-ab, 2021-04-28:
> The commit looks quite good already (please see also my commit comment).
>
> What do you think about combining all placement-policy related functions/types into one class in contrast to the current [...]

Ok, i added the replacement commit 4e11ea949549ff4bc65372586aa7dcb9073ccda1

alex-ab, 2021-04-28:
> Maybe related, I noticed that the pthread test reproducibly fails for base-hw on the two ARMv7 boards imx7d_sabre and imx6q_sabrelite with the current staging branch. In both cases, there is no [...]

With the change of the commit, all cpus are used all over the time for new pthreads, instead solely for the first 32 pthreads. I can check next time I'm in office.

chelmuth, 2021-04-28:
I merged the series to staging. Thanks @alex-ab for addressing my code-cleanup request.

alex-ab, 2021-04-28:
> The results over the weekend are mixed:
> 
> linux_x86_64: succeed one time, failed another time with segmentation fault
> linux_x86_32: failed 2x times (timeout, early no progress in the log)
> x86_64 (hardware): nova && hw - succeeded 2x times
> imx8 (hardware): hw - succeeded 2x times
> x86_32 (hardware): nova failed 2x times (spawn died unexpected)
> 
> The runtime on hardware was too short, so we may increase the round numbers (just 1minute for 1000 iterations)

In the x86_32 case, it seems that often a case triggers, where the sender (the dynamic spawned worker_threads) and the one receiver disagree about the transferred data amount. The receiver decides that it received enough, tries to clean up and finally make a pthread_join on the sender. The sender still believes to have to send data and never leaves the function scope to get to an joinable state.

alex-ab, 2021-04-29:
With commit 8d32f7958a8fddd0bc20ae6538892e152a707e16 the 32 platform make progress. Please add also 304568b5c875d62040824d5a0c7fcbc8becc3dfa to staging.

chelmuth, 2021-04-29:
Thanks, merge complete.

cproc, 2021-05-03:
I got a page fault with manual pthread placement in falkon on base-linux on the current staging branch. According to git bisect it is related to commit 40578f8. With commit 7d43f8d the page fault can be reproduced with `make KERNEL=linux BOARD=linux run/test PKG=test-pthread`. It appears to happen in `Genode::Mutex::acquire()` if I interpreted the objdump output correctly.

alex-ab, 2021-05-04:
@cproc: Thanks for reporting. Please use 8086f522fe84053753fb3303f46d0b74e7e45093 as fixup

cproc, 2021-05-04:
@alex-ab: thanks, no page fault anymore with falkon as well

nfeske, 2021-05-04:
Thanks. Merged to staging.

alex-ab, 2021-05-11:
f4b74cb820d8304d2f65364df38e25424724a9b6 fixes a pagefault/segmentation fault happening on our nightly weekend tests and adjust the hw/imx8 to a round number feasible in the given amount of test time.

trimpim, 2021-05-17:
e816d55  fixes a possible segfault that could occur which I was pointed to by a coworker.

alex-ab, 2021-05-17:
The commits fcba32d3488442e2c5b1b1f7c1f9493aacc7cd34 and 0cb8f3ef550691d2f63b586610e853fe990f7d73 fixes the std::runtime_exception for the libc_integration script happened on staging over the weekend. With the stdcxx update an exception is thrown on x86 if no sufficient hardware random source is available.

commit ab8183bf (2021-05-12): libc_integration: fix out-of-bound access

which leads to sporadic pagefaults/segmentation faults.

Fixes #3967
commit f8d0552d (2021-05-28): libc_integration: fix out-of-bound access

which leads to sporadic pagefaults/segmentation faults.

Fixes #3967
closed 2021-05-28 by chelmuth