genodelabs/genode #3969: libc: reading bogus data from file provided via vfs/rom plugin

sidhussmann, 2020-12-18:
During stress-testing a new feature, I noticed a strange error while reading data from a file provided by the vfs/rom plugin. Commit TODO contains a test that reproduces this error on base-linux usually within <1000 iterations. The test starts several pthreads. Within these threads, said file is opened and read twice. Sometimes, however, the data read is not the same. Analyzing with GDB shows that one of the buffers contain mostly zeroes, while the other contains the expected data.

chelmuth, 2020-12-18:
I slightly adjusted your test in a24ac8b918169c062110d2951c51218db6934bf0 to reproduce the error almost instantly. When replacing `fread` etc. by `read` and friends in 61c069fc84aefbe31c4e8819bbe07a4b6bf16558 the test runs without flaw. This observation underpins our suspicion that the libc stream API does not work as expected. A closer look into the implementation reveals that stream function use special locks to protect the stream data structures as well as global data (see `__sfp()` for example).

Long story short, it seems your test hits a documented shortcoming not completely addressed yet *libc: thread-safety #725*

sidhussmann, 2020-12-21:
@chelmuth thank you for your explanation. For now it is good to know of this shortcoming, so we can circumvent it by avoiding `fread` and friends. However, at some point in the future we might port existing code to Genode, which uses `fread` and friends. How involved is fixing this shortcoming, once it becomes necessary?

chelmuth, 2020-12-21:
I did not look at the libc from the *concurrency* angle thoroughly and, therefore, have more questions than answers.

- Why does `STDIO_THREAD_LOCK()` in *findfp.c* use a spinlock in contrast to a pthread mutex used by `FLOCKFILE()`? Do we have to address this in our port too or can we just use a mutex?
- What are the semantics of `__isthreaded` which conditionally enables the concurrency measures?
- Which other places use which measures for concurrency protection? (e.g., `_MUTEX_LOCK` in *localtime.c*)

My current knowledge hints that enabling `__isthreaded = 1` when the first additional pthread is created is a good starting point. Depending on the resulting issues I expect one to two weeks of test and implementation.

nfeske, 2021-02-26:
Since this issue is not related to the VFS/ROM and the immediate issue could be resolved by a work-around, I'm closing it.

We still have the issue https://github.com/genodelabs/genode/issues/725 to track the libc-thread-safety topic.

closed 2021-02-26 by nfeske