genodelabs/genode #3971: Remove obsolete mk/spec/ files for distinguishing boards

nfeske, 2021-01-04:
With the unified build directories in place (#3316, #2190), the board-specific spec files at https://github.com/genodelabs/genode/tree/master/repos/base/mk/spec should be removed. Otherwise, aliasing between boards can still happen at the build stage.

This cleanup should also consider the following places:

* If boards must be distinguished, run scripts must use the `[board]`, not the `[have_spec]` function. When removing a spec file, all run scripts should be grep'ed for the spec name and adjusted.

* The line at https://github.com/genodelabs/genode/blob/master/tool/builddir/build.mk#L93 should be removed. It magically adds the board name as spec value. I guess this is leftover from the transition phase.

@skalk Would you consider picking up this issue?

tomga, 2021-01-04:
@nfeske I think I provoked you to create this issue although I did not expect so far-reaching changes, so I'll allow myself to comment.

Removing board specific files from `repos/base/mk/spec` may not be a good idea as they define specs like `framebuffer`, `gpio` and probably have to stay defined.

The real problem are board specific entries that add to REP_INC_DIR and only those for which exist board specific include paths: `repos/os/include/spec/imx53` and `repos/os/include/spec/rpi` and of course the aforementioned line that add board to specs.

nfeske, 2021-01-05:
> I think I provoked you to create this issue although I did not expect so far-reaching changes, so I'll allow myself to comment.

I actually spotted the problem during a discussion with @chelmuth about issue #3773 where we need to distinguish ARM CPUs with and without NEON support. But once I saw the spec files, I could immediately draw the connection to our discussion.

> Removing board specific files from repos/base/mk/spec may not be a good idea as they define specs like framebuffer, gpio and probably have to stay defined.

Those specs are obsoleted by now, similar to the board-specific spec values. Today, the build directories only depend on the instruction-set architecture. If the spec values are still used somewhere - presumably in conditions used in run scripts - they should be removed (replaced by `[board]`-depending conditions).

tomga, 2021-01-05:
> I actually spotted the problem during a discussion ...

What a coincidence :-) Day after my mentioning build problems with board specific include paths. Sorry for the noise then.

> Those specs are obsoleted by now, similar to the board-specific spec values.

I missed that fact. Thank you for the information.

skalk, 2021-01-06:
@nfeske yes, I can pick up the issue.

skalk, 2021-01-12:
@nfeske commits 58cd065 and 0e81432 resolve the issue. I've used `[is_board]` instead of `[board]` for convenient return of a boolean value. The function `[board]` was already declared in the run-tool to return the BOARD string value that is used to select packages, or the correct core/bootstrap libraries. I hope you are however fine with the naming.

nfeske, 2021-01-12:
I'm not confident that the use of the `is_` prefix is a good idea because it is not consistent with our C++ coding style (which is for C++, but is nevertheless also in parts applied to our Tcl scripts). For reference, see the last point of the "Identifiers" section: https://genode.org/documentation/developer-resources/coding_style#Identifiers. I would rather not proliferate this prefix in our code base.

I'm not really able to see how `is_board` is much better than the existing `board` function.  If you dislike the prominent use of double quotes (`[board] == "pbxa9"`) I agree. But I'm not sold on `is_board` either. ;-)

Aside the style question, your patch introduces few inconsistencies (at https://github.com/skalk/genode/commit/58cd065dc05937addccc53c5aa6ca0634f5a1c54#diff-762a3143d1e675e3b833ae199d1b9a0f1ed71a88989ab9990bb2458fe83017bfR55, https://github.com/skalk/genode/commit/58cd065dc05937addccc53c5aa6ca0634f5a1c54#diff-29a1fa45fad218da6c768e46333e603b3475b5e718f36bad47505ae383c35aefR3)

skalk, 2021-01-12:
@nfeske, thank you for your review. I've fixed the two places you mentioned and an additional one in a fixup commit.

Regarding the naming: Indeed I did not want to use the double quotes approach, especially in more complex boolean expressions in our run scripts. That is why I introduced another method beside `[board]`. I'm aware that `is_` prefix is not common in our code base. But I thought it would be okay in the run-script context, as it is quite similar to `[have_spec]` somehow. On the other hand, I did not want to change the semantics of the already existent `[board]` function.
Well, I'm puzzled now what to do next? Changing `[is_board]` into another name, leaving it in this state, or using `[board]` with comparator and double quotes? I can change it, if you like, but personally dislike the last option.

nfeske, 2021-01-12:
> Well, I'm puzzled now what to do next? Changing [is_board] into another name, leaving it in this state, or using [board] with comparator and double quotes? I can change it, if you like, but personally dislike the last option.

I agree that changing the existing `board` function is risky.

My reservation against `is_board` could be addressed by calling the function `have_board`, which would mirror the `have_spec` pattern somehow, and side-step the `is_` prefix style issue. Could you live with that?

skalk, 2021-01-13:
I've reworked both commits accordingly. They are cbc0748 and 0f8b4b7.

nfeske, 2021-01-13:
Thank you for the update. Now I'm happy when browsing the commits. :-)

With this taste-issue solved, I can now truly appreciate your cleanup work. In particular, I like how the run scripts have become more self-descriptive. Previously, the meaning of spec values like `pbxa9` was not intuitively clear. Now, when seeing a condition like `[have_board pbxa9]` the reader knows that pbxa9 is a board. It's also great to see how many lines you were able to remove.

I think in the second commit, a file accidentally slipped in: https://github.com/skalk/genode/commit/0f8b4b7de84ad12648688505b1d998764b5d84de#diff-4bee8ad0a1c8f8b5ab8f762225c273087ea60a6ec099c9b90b035f9643ad0234

skalk, 2021-01-13:
@nfeske yes, it slipped in. I've re-pushed the second commit: 
4f7667f Sorry!

skalk, 2021-01-13:
@nfeske: @tomga directed me to an error I made in the commit. Therefore, I've added fixup commit 
b0931ea too.

nfeske, 2021-01-13:
Thanks @skalk. I merged your commits to staging now.

chelmuth, 2021-01-14:
The commits seem to break sel4 for arm_v7a.

```
> ./tool/depot/create UPDATE_VERSIONS=1 FORCE=1 -j4 genodelabs/bin/arm_v7a/base-sel4-imx7d_sabre
created genodelabs/api/base/2021-01-14 (new version)
removing stale binary archive genodelabs/bin/arm_v7a/base-sel4-imx7d_sabre/2021-01-14
created genodelabs/src/base-sel4-imx7d_sabre/2021-01-14 (new version)
checking library dependencies...
Library-description file syscall-sel4.mk is missing
Library-description file kernel-sel4-.mk is missing
Library-description file kernel-sel4-pc.mk is missing
Library-description file ld-sel4.mk is missing
Skip target timer/pit because it requires x86
make[5]: *** No rule to make target 'syscall-sel4.lib', needed by 'startup-sel4.lib'.  Stop.
Makefile:292: recipe for target 'gen_deps_and_build_targets' failed
make[4]: *** [gen_deps_and_build_targets] Error 2
.../tool/depot/mk/build_bin_archive:200: recipe for target '.../depot/genodelabs/bin/arm_v7a/base-sel4-imx7d_sabre/2021-01-14.build/bin' failed
make[3]: *** [.../depot/genodelabs/bin/arm_v7a/base-sel4-imx7d_sabre/2021-01-14.build/bin] Error 1
.../depot/var/build.mk:12: recipe for target 'genodelabs/bin/arm_v7a/base-sel4-imx7d_sabre/2021-01-14' failed
make[2]: *** [genodelabs/bin/arm_v7a/base-sel4-imx7d_sabre/2021-01-14] Error 2
.../tool/depot/build:134: recipe for target 'execute_generated_build_mk_file' failed
make[1]: *** [execute_generated_build_mk_file] Error 2
tool/depot/create:59: recipe for target 'build' failed
make: *** [build] Error 2

> ./tool/depot/create UPDATE_VERSIONS=1 FORCE=1 -j4 genodelabs/bin/arm_v7a/base-sel4-imx6q_sabrelite
created genodelabs/api/base/2021-01-14
created genodelabs/src/base-sel4-imx6q_sabrelite/2021-01-14 (new version)
checking library dependencies...
Library-description file syscall-sel4.mk is missing
Library-description file kernel-sel4-.mk is missing
Library-description file kernel-sel4-pc.mk is missing
Library-description file ld-sel4.mk is missing
Skip target timer/pit because it requires x86
make[5]: *** No rule to make target 'syscall-sel4.lib', needed by 'startup-sel4.lib'.  Stop.
Makefile:292: recipe for target 'gen_deps_and_build_targets' failed
make[4]: *** [gen_deps_and_build_targets] Error 2
.../tool/depot/mk/build_bin_archive:200: recipe for target '.../depot/genodelabs/bin/arm_v7a/base-sel4-imx6q_sabrelite/2021-01-14.build/bin' failed
make[3]: *** [.../depot/genodelabs/bin/arm_v7a/base-sel4-imx6q_sabrelite/2021-01-14.build/bin] Error 1
.../depot/var/build.mk:12: recipe for target 'genodelabs/bin/arm_v7a/base-sel4-imx6q_sabrelite/2021-01-14' failed
make[2]: *** [genodelabs/bin/arm_v7a/base-sel4-imx6q_sabrelite/2021-01-14] Error 2
.../tool/depot/build:134: recipe for target 'execute_generated_build_mk_file' failed
make[1]: *** [execute_generated_build_mk_file] Error 2
tool/depot/create:59: recipe for target 'build' failed
make: *** [build] Error 2
```

chelmuth, 2021-01-14:
Same for build-directory compilation.
```
> make -C build/arm_v7a KERNEL=sel4 BOARD=imx7d_sabre LIB=ld-sel4
checking library dependencies...
Library-description file syscall-sel4.mk is missing
make[1]: *** No rule to make target 'syscall-sel4.lib', needed by 'startup-sel4.lib'.  Stop.
make[1]: *** Waiting for unfinished jobs....
    COMPILE  emutls.o
    COMPILE  exception.o
    COMPILE  guard.o
  Library cxx
    COMPILE  malloc_free.o
    COMPILE  misc.o
    MERGE    supc++.o
    CONVERT  supc++.o
    MERGE    cxx.lib.a
Makefile:292: recipe for target 'gen_deps_and_build_targets' failed
make: *** [gen_deps_and_build_targets] Error 2
```

m-stein, 2021-01-14:
@chelmuth I would suggest a fixup:
b630a385d5 fixup: "base: remove SPEC variables of boards (fix #3971)"

skalk, 2021-01-15:
@chelmuth: the fixup commit e3b7fa0 shall address all compilation and packaging issues regarding base-foc and base-sel4. Sorry for the long delay, but it was a bit more complicated to solve the board-spec free issues here. I 've tested run-scripts, package building and `make LIB=...` without a BOARD variable.

chelmuth, 2021-01-15:
Great, thank you. Merged it instantly.

nfeske, 2021-01-19:
Just as a reminder, we need to adjust tool/run/boot_dir/hw in genode-world.

skalk, 2021-01-19:
@nfeske yes, as soon as the commit hits staging, there are still a lot of occurences in genode-world ;-)

chelmuth, 2021-01-20:
I discovered that the base-hw-muen build is broken because the kernel *target.mk* depends on `SPECS` containing `muen`.
https://github.com/genodelabs/genode/blob/b8b06e7135ade841a6faf2340c777172309e6c65/repos/base-hw/src/kernel/spec/muen/target.mk#L26-L55

During my investigations I also stumbled upon the following places where we must also remove the use of  board `SPECS` to close this issue. Otherwise other such hidden dependencies may stay scattered over the code base.
```
base-hw/recipes/src/base-hw-imx53_qsb/content.mk:7:	echo "SPECS += imx53_qsb" >> etc/specs.conf
base-hw/recipes/src/base-hw-rpi3/content.mk:7:	echo "SPECS += rpi3" >> etc/specs.conf
base-hw/recipes/src/base-hw-zynq_qemu/content.mk:7:	echo "SPECS += zynq_qemu" >> etc/specs.conf
base-hw/recipes/src/base-hw-imx53_qsb_tz/content.mk:7:	echo "SPECS += imx53_qsb trustzone" >> etc/specs.conf
base-hw/recipes/src/base-hw-virt_qemu/content.mk:7:	echo "SPECS += virt_qemu" >> etc/specs.conf
base-hw/recipes/src/base-hw-imx8q_evk/content.mk:7:	echo "SPECS += imx8q_evk" >> etc/specs.conf
base-hw/recipes/src/base-hw-pbxa9/content.mk:7:	echo "SPECS += pbxa9" >> etc/specs.conf
base-hw/recipes/src/base-hw-rpi/content.mk:7:	echo "SPECS += rpi" >> etc/specs.conf
base-hw/recipes/src/base-hw-muen/content.mk:7:	echo "SPECS += muen" >> etc/specs.conf
base-hw/etc/specs.conf:1:SPECS += genode hw
base-hw/etc/specs.conf:3:ifeq ($(filter-out $(SPECS),x86_64),)
base-hw/etc/specs.conf:4:SPECS += pci ps2 vesa framebuffer
base-hw/etc/specs.conf:6:ifneq ($(filter-out $(SPECS),muen),)
base-hw/etc/specs.conf:7:SPECS += acpi
base-hw/etc/specs.conf:11:ifeq ($(filter-out $(SPECS),imx53_qsb_tz),)
base-hw/etc/specs.conf:12:SPECS += imx53_qsb trustzone
base-hw/etc/specs.conf:15:ifeq ($(filter-out $(SPECS),usb_armory),)
base-hw/etc/specs.conf:16:SPECS += trustzone
base-hw/src/kernel/spec/muen/target.mk:26:ifneq ($(filter muen, $(SPECS)),)

```

Regarding Muen please consider Issue #3995.

skalk, 2021-01-20:
@chelmuth: I wonder that I left out so many board recipes in base-hw, because I know I started to remove several of it. Mea culpa! Regarding the declarations of pci, ps2, vesa etc. I've left them out by now, because those were part of the original x86 SPECS too, and I wanted to solve this as a follow-up issue, which is already in the pipe. But I can add it as a fixup too, if you like?!
Regarding the muen issue: I would have had removed the muen/SPECS filter rule. But this implies that in case of `make all KERNEL=hw` for x86_64 the muen kernel would be build, which does not really correspond to issue #3995. So I'm stumped now. Do you agree, when I do remove the "muen requirement" for now to solve this issue, and separate it from issue #3995?

skalk, 2021-01-20:
@chelmuth I've put two additional fixup commits to my merge_to_staging branch. The first a09dc7f removes the left board SPECS declarations from the base-hw recipes, and removes the muen kernel target restriction. The second removes all platforms-specifics left (like ps2 vesa framebuffer ...) that were for historical reasons part of the x86 declaration. If you like, I can move the second part into a distinct issue.

chelmuth, 2021-01-20:
Many thanks, I merged both commits without objections to staging.

chelmuth, 2021-01-20:
@alex-ab What do you think, should we just remove or adapt the following line?

https://github.com/genodelabs/genode/blob/c0ef9da4de7b2e16e22c4564904bbf8fcd7c6219/repos/libports/run/acpica.run#L1

chelmuth, 2021-01-20:
Hm, same here.

https://github.com/genodelabs/genode/blob/c0ef9da4de7b2e16e22c4564904bbf8fcd7c6219/repos/os/run/pci.run#L1

alex-ab, 2021-01-20:
I'm fine with just removing it.

chelmuth, 2021-01-20:
Great. I removed both lines.

commit 3c49632a (2021-01-13): fixup: "base: remove SPEC variables of boards (fix #3971)"
commit e8da1dcc (2021-01-13): base: remove SPEC variables of boards (fix #3971)

* Remove SPEC declarations from mk/spec
* Remove all board-specific REQUIRE declaratiions left
* Replace [have_spec <board>] run-script declarations with have_board where necessary
* Remove addition of BOARD variable to SPECS in toplevel Makefile
* Move board-specific directories in base-hw out of specs
commit 9da9783f (2021-01-14): fixup: "base: remove SPEC variables of boards (fix #3971)"
commit 65af7aff (2021-01-15): fixup: "base: remove SPEC variables of boards (fix #3971)"
closed 2021-01-25 by nfeske