genodelabs/genode #3972: Minor issues in mk files

tomga, 2021-01-04:
During my work on alternative build system I found some mistakes that are ignored by current build system but probably should be fixed. There are fixes for two types of issues:
 * wrong extension names of files
 * duplicated files in sources (duplicates with entries from included files)

Both those types of problems are silently _repaired_ by current build system.
Fixes for problem I found are in my [mk_fixes](/tomga/genode/commits/mk_fixes) branch.

chelmuth, 2021-01-05:
Merged as 17c13bf833bc56862fad86debf3e0f364910faab.

tomga, 2021-03-14:
@nfeske @chelmuth I have another set of fixes in make files in 91f700d634253cd004652ffa54cc3823d5c4de24 that I'd like you to consider for merge. I don't think they deserve a new issue, so I write here.

Fixes are for:
 * a duplicated source file
 * a wrong line continuation
 * few places where CC_CXX_OPT is set instead of being appended to - this causes ignoring CC_CXX_OPT set in spec makefiles in case of libraries but probably due to different include order not for programs - anyway I think it is wrong

nfeske, 2021-03-16:
Thank you @tomga. I merged your commit as https://github.com/genodelabs/genode/commit/2f0898d2a946e2b2acb034cd2f8240e6aaa6de77.

nfeske, 2021-03-23:
Fixed in master.

tomga, 2021-03-27:
@nfeske @chelmuth I have found another inconsistency in behavior. Steps to reproduce are as follow:
```
tool/create_builddir linux
sed -i -e 's/#\(.*dde_linux\)/\1/' build/linux/etc/build.conf   # enable dde_linux repository
make VERBOSE= VERBOSE_MK= VERBOSE_DIR= -C build/linux BOARD=linux drivers/wifi
```
Building `drivers/wifi` requires some libraries from `libports` that are not available and therefore nothing should be built. But it can be observed that `.ucode` are copied due to rules in `repos/dde_linux/lib/mk/iwl_firmware.mk` _assigned_ to `all`.

637f17a26941426ca84f2bd0a157a936ff7c41cc contains a local fix for the issue. It works but I don't like it as I had to use internal `.lib.tag` name directly (`$(LIB_TAG)` did not work as is not defined yet I think).

I think that clean solution to such cases would be introducing support for `CUSTOM_LIB_DEPENDENCIES` in `lib.mk` by adding them to library tag dependencies and define it here.

Of course the issue is small and you can also decide to ignore it completely. In such case I will silence this case in my consistency checks with my scons experiment.

tomga, 2021-03-28:
In cb94e7580fe596a2b61796598ebb49133cda3f19 I added a fixup  with a proposition to solve it cleaner in my opinion.

nfeske, 2021-03-29:
I like your suggestion, @tomga. In fact, while we're at it, I think we should consider taking this direction even a bit further to also cover places like https://github.com/genodelabs/genode/blob/master/repos/gems/src/app/menu_view/target.mk among others. You are touching a long-standing inconvenience of the build system.

Do you agree that such a generalization makes sense? If yes, what would be a sensible name that would be intuitive (when encountering it in a build-description file) while covering both cases (target.mk and libraries)?

In any case, I'd suggest splitting the introduction of the hook and the adaptation of the actual target(s) into two distinct commits, one that introduces the mechanism (with a short documentation of the hook in the commit message) and one that applies it.

tomga, 2021-03-29:
@nfeske, I expected that such cases exist for program targets (although I did not look for them and they did not hit me yet) and I also thought that a generalized solution should be implemented and I hoped that you'll propose a name for this variable ... :-) `TARGET_DEPS`? `CUSTOM_TARGET_DEPS`? `TARGET_DEPENDENCIES`?

As about splitting implementation of hook and adaptation I agree of course. Current commits were more like request for comment.

And one more thing - are you ok with adding more such fixes to this issue or would you prefer opening new each time? If using current is ok, then can you reopen it? I don't think I'm allowed to do this myself.

nfeske, 2021-03-29:
> And one more thing - are you ok with adding more such fixes to this issue or would you prefer opening new each time?

I'm generally fine with either way and will just go ahead with re-opening it.

Your suggestion `CUSTOM_TARGET_DEPS` sounds very good to me.

tomga, 2021-03-31:
Changes are in my [mk_fixes_2](/tomga/genode/commits/mk_fixes_2) branch.

First commit contains introduction of `CUSTOM_TARGET_DEPS` for programs and libraries with some small doc string that I assumed is needed. @nfeske feel free to modify it in any way or even remove if you decide it is not required.

Following two commits make use of this variable in `iwl_firmware` library and mentioned by you `app/menu_view`. I tested those changes briefly and they seem to work as expected.

nfeske, 2021-04-06:
Thank you @tomga for the commit series, which I merged to staging right now.

I am especially happy about your attention to detail and the added documentation. It is fantastic to receive such thoughtful contributions!

nfeske, 2021-04-20:
@tomga, with the commits having entered the master branch, are you fine with closing this issue?

tomga, 2021-04-20:
Of course. We can reopen it again if something new comes up.

nfeske, 2022-06-01:
Commit 6326774056b62ab8d0182400c90169009c8e19d9 addresses a corner case I observed with app/menu_view.

tomga, 2022-07-31:
I've encountered another corner case in build system that caused a surprising (for me) failure. I tried to build some simple scenario for rpi and used build directory that I created some time ago. Build failed with:
```
  Program core/hw/core-hw-rpi3
(echo    "create core-hw-rpi3.a"; \
           echo -e "\naddlib /projects/genode/genode/build/arm_v8a/var/libcache/base-hw-common/base-hw-common.lib.a \naddlib /projects/genode/genode/build/arm_v8a/var/libcache/core-hw-rpi3/core-hw-rpi3.lib.a \naddlib /projects/genode/genode/build/arm_v8a/var/libcache/cxx/cxx.lib.a \naddlib /projects/genode/genode/build/arm_v8a/var/libcache/syscall-hw/syscall-hw.lib.a"; \
           echo    "save"; \
           echo    "end"; \
          ) | /usr/local/genode/tool/21.05/bin/genode-aarch64-ar -M
/usr/local/genode/tool/21.05/bin/genode-aarch64-strip --strip-unneeded -o core-hw-rpi3.a.stripped core-hw-rpi3.a
ln -sf /projects/genode/genode/build/arm_v8a/core/hw/core-hw-rpi3.a.stripped /projects/genode/genode/build/arm_v8a/bin/core-hw-rpi3.a
ln -sf /projects/genode/genode/build/arm_v8a/core/hw/core-hw-rpi3.a /projects/genode/genode/build/arm_v8a/debug/core-hw-rpi3.a
/projects/genode/genode/build/arm_v8a/var/tool/ccache/genode-aarch64-gcc   /projects/genode/genode/build/arm_v8a/bin/core-hw-rpi3.o core-hw-rpi3   -o /projects/genode/genode/build/arm_v8a/bin/core-hw-rpi3
genode-aarch64-gcc: error: core-hw-rpi3: No such file or directory
make[3]: *** [<builtin>: /projects/genode/genode/build/arm_v8a/bin/core-hw-rpi3] Error 1
make[2]: *** [var/libdeps:93: core-hw-rpi3.prg] Error 2
make[1]: *** [Makefile:326: gen_deps_and_build_targets] Error 2
make[1]: Leaving directory '/projects/genode/genode/build/arm_v8a'
Error: Genode build failed
make: *** [Makefile:421: run/log] Error 252
```
After some investigation I found that this error is triggered by files:
```
build/arm_v8a/debug/core-hw-rpi3.o
build/arm_v8a/bin/core-hw-rpi3.o
```
that existed in my build directory and were left after some earlier build. Build with any of those files in build directory caused additional (failing) command being executed for `core/hw/core-hw-rpi3` program.
```
/projects/genode/genode/build/arm_v8a/var/tool/ccache/genode-aarch64-gcc   /projects/genode/genode/build/arm_v8a/bin/core-hw-rpi3.o core-hw-rpi3   -o /projects/genode/genode/build/arm_v8a/bin/core-hw-rpi3
```
After some more investigation I found a very surprising for me thing that `make` has a set default builtin pattern rules. In this case one of those rules is considered by `make` for execution because all required sources for this rule exist. Running `make` with `--no-builtin-rules` parameter disables builtin rules and causes build to succeed.

As I think that all required rules are defined by Genode build system I tend to think that disabling builtin rules in all cases should be ok. My proposal for that is in 2f8f200e27e8b09646278634ccedfde897bb920d. I did not test it extensively but I tested on some simple scenario on a newly created build directory. If you think that this corner case is marginal just ignore this patch.

tomga, 2022-08-08:
On a second thought proposed fix may not be as safe as I considered it week ago. Disabling builtin rules using `MAKEFLAGS` variable will probably propagate to subsequent make calls and in case of builds of external components that can potentially change/break them. So probably it is best to ignore this problem as a *lesser evil*.

chelmuth, 2022-08-31:
There seems to be no concrete *minor issue* left, so let's close this issue.

commit 17c13bf8 (2021-01-05): Fix minor inconsistencies in mk files

Fixes #3972
commit 98798f18 (2021-01-25): Fix minor inconsistencies in mk files

Fixes #3972
closed 2022-08-31 by chelmuth