genodelabs/genode #3973: nitpicker: defer hover changes during drag operations

nfeske, 2021-01-05:
When resizing a window with a size constraint using the motif decorator and depressing the mouse button while the pointer is outside the resize handle, the resize handle stays in hovered state.

During the analysis together with @chelmuth, we found that nitpicker updates its internal `_hover` state too eagerly, which indirectly leads to the masking of a leave event to the decorator. This issue may be related to the sporadic "jumping window" effect that sometimes occurs in Sculpt.

cproc, 2021-01-05:
@nfeske: with your latest commit I could not reproduce the jumping window or resize handle hovered state anymore.

nfeske, 2021-01-06:
@cproc, thank you for the confirmation. :-)

commit 6d106392 (2021-01-05): nitpicker: defer hover changed while dragging

When holding at least one key or button, the hovering should never
change. In the previous version, this invariant was not enforced,
but the symptoms for eventual inconsistencies were masked by two
checkes: one when generating the hover report, and one when submitting
the leave event. This patch enforces the invariant by suppressing any
change of 'User_state::_hovered' while in dragged state.

Fixes #3973
commit 72801975 (2021-01-25): nitpicker: defer hover changed while dragging

When holding at least one key or button, the hovering should never
change. In the previous version, this invariant was not enforced,
but the symptoms for eventual inconsistencies were masked by two
checkes: one when generating the hover report, and one when submitting
the leave event. This patch enforces the invariant by suppressing any
change of 'User_state::_hovered' while in dragged state.

Fixes #3973
closed 2021-01-25 by nfeske