genodelabs/genode #3975: Choosing linking to abi or real library can sometimes lead to broken builds or build errors

tomga, 2021-01-05:
If for some reason in the build directory `.abi.so` file will be created and later build will be performed where there is no symbols for that library then even though this file is stale it will be selected for linking of dependent libraries and programs.

This behavior is due to code in [generic.mk](/genodelabs/genode/blob/master/repos/base/mk/generic.mk#L164-L178) which selects between `.abi.so` and `.lib.so` only based on existence of those files in build directory.

This case can happen rarely and I think the most probable case is when switching between branches where abi for concrete library is defined and not.

I don't have a patch for it but I think that a possible route could be modifying [lib.mk](/genodelabs/genode/blob/master/repos/base/mk/lib.mk#L168-L169) in such a way to write into `LIB_TAG` file information about currently built type of library: `abi` or `so`. Information from that file could be used later for selecting target for links in dependent programs and libraries.

I spotted this behavior and report it just for your information. If you find it not important or you think that cure can be too complicated for disease than just close it and I will have no problem with that.

nfeske, 2021-01-07:
Thanks for raising this issue, which touches on a subject I have discussed previously with @chelmuth.

The magic of the distinction between ABIs and shared libraries is unfortunate. Right now, the behavior should be:

* If an symbols file for a given library listed in the LIBS variable exists, link the ABI stub, not the library.
* If a library description file for a given library listed in the LIBS variable exists, build the library before processing the target.

As shown in your report, the first point does not work in all corner cases. The second point is unfortunate in situations where the developer is not interested in building library dependencies, but wants to merely use the library in binary form and its ABI/ABI. For example, for developing a Qt5 application via Genode's build system, the Qt5 libraries naturally appear in the LIBS variable. Even though there exist symbol files for those libraries, and binary packages are readily available via the depot, the Qt5 libraries are implicitly built in the build directory, which takes ages.

I think the proper way to fix both of these shortcomings would be the introduction of an ABIS variable in addition to the existing LIBS variable. This way, a target.mk file can be explicit about using an ABI or a library. For items listed in ABIS, the build system would not try to build the corresponding library but would just expect a symbols file to be present. This way, one could develop a Qt5 application without triggering the build of the Qt5 libraries in the build directory. For the situation described in your report, the two branches would have different target.mk files, one with a LIBS declaration and one with an ABIS declaration. So the ambiguity goes away.

With this mechanism in place, however, we have a new problem: Since targets that merely use ABIs would no longer implicitly trigger the building of libraries, we need another way to build them. Right now, the build system has the `LIB=` feature, but this feature is not available when triggering the build from a run script. Strictly speaking, there are already some cases where shared libraries must be built without a depending target, in particular VFS plugins. In these cases, we have a bunch of pseudo targets (like https://github.com/genodelabs/genode/blob/master/repos/gems/src/lib/vfs/ttf/target.mk) which have the sole purpose of building a library as side effect. This of course does not scale. One could call it a hack. To provide a general solution to this problem, I'd suggest to extend the build system's front end by handling `lib/*` targets as a special case, similar to how it already handles `run/*` as a special case. So one could issue a build command as follows:

```
make init lib/vfs server/vfs ...
```

The lib/vfs argument would prompt the build-system front end to trigger the build of the library named vfs. This argument list could naturally be passed to the build command of run scripts. Of course, target.mk files within `<repo>/src/lib/` cannot work anymore. But I'd argue that those target.mk files are hacks anyway, so they better be gone. Once the build-system front-end supports the lib/* special case, all target.mk files can be removed from the src/lib/ directories.

That all said, this change is not easy. As far as I can see, it would require the following steps.
 
- Adding the lib/* feature to the build system front end. When just specifying `make lib`, all libraries should be built. This is important for building depot archives. From this point on, targets can no longer reside within src/lib/. Hence, all run scripts that contain lib/* as build argument must potentially be adapted. E.g., lib/vfs/ttf must be changed to lib/vfs_ttf because the library name is vfs_tff. Fortunately, these can be grep'ed for and they are not too many.
- As an iterative process, we can now change the target.mk files from the use of LIBS towards the use of ABIS. This should be done library after library. But there is no pressure. For each library tackled, we need to revisit the run scripts for the library listed as boot module. If listed, we need to extend the build command with a corresponding lib/<libname> argument.
- With the transition completed, be can remove the magic behind the handling of the LIBS variable. Items specified in LIBS would no longer link ABIs but always libraries.

Sorry for the wall of text. This looks like quite a lot of work. Would it be worthwhile to pursue? I tend to think so.

tomga, 2021-04-14:
@nfeske I read your proposal few times over past months and still have a feeling that some things are not exactly right in it. Like you wrote it's quite a lot of text and many issues are touched and I needed time to think them over thoroughly.

I think there are different things mixed a little and in consequence different problems are joined together which I think is not required. I'll try to split them to different issues and analyze separately.

Maybe I'll start with something that I do not like from the beginning: if I correctly understand your proposal developer will be forced to specify if he wants to link to ABI or to a LIB (if I misunderstood then probably everything written further does not have much value). I think it is wrong. As a developer I want to specify only dependencies. I don't need to specify in target configuration if I want to build full library or only ABI. Adding ABI rule to a library should not change "users" of this library. Build system should "know" depending on specified targets if dependency requires libraries to be built fully. If main target is a run script then dependent libraries probably should be built regardless of the fact if they can provide ABI or not because they are needed for target image. If I ask for building a program, by default only what is really required should be built - therefore linking to ABI should be chosen if available. This information in my opinion is not part of build description. It is a part of state of current build only.

Therefore I'd advocate that first rules you wrote describing current system should be changed to:
 * if a library does not have ABI rule then processing it should build it
 * if a library provides ABI rule then it should build only ABI unless this library is needed for an image for run script or there was an explicit requirement for full build of a library.

With those rules implemented your example problem when full build of Qt library is performed disappears.

Second issue is the one that I reported originally. In my description I also described (I think simple) solution to this problem with minimal changes to build system. It is important to have information which files in build directory can be used for linking - it can be stored in a file in build directory. I proposed using `.tag` file which is currently empty but could be easily extended. I have actually implemented it in my (unfortunately still not ready) SCons build.

Third issue is the need to be able to specify need for building full library. I complete agree with such a need. I just don't think it is important to put such targets in any configuration files. Neither `target.mk` files nor `*.run` files.

Fourth issue you mentioned is the special library targets that currently overcome the problem of specifying library dependencies from run scripts. Your proposal with `lib/*` targets probably would allow to resolve issue but again I think that the need to perform full build should be a result of knowledge of build system that this dependency comes from run script which requires libraries and not only abis.

In summary I think that:
 * specifying build dependencies in build configuration files should not be mixed with "runtime" build requirements for targets as it complicates reasoning and can lead to wrong conclusions
 * no additional build configuration changes are required and therefore no additional work for the team is needed in this area - changes in build system itself should suffice

I must admit that I don't have a full view yet as I did not inspect build of depots yet and may be mistaken in my conclusions.

nfeske, 2021-05-04:
@tomga You make very good points. In particular, I can understand your resistance against the idea to let the developer specify LIBS vs ABIS in the target.mk file.

nfeske, 2023-11-30:
Genode 23.11 has made the distinction between the use and linking of ABIs and shared libraries consistent now.

closed 2023-11-30 by nfeske