genodelabs/genode #3977: imx8_synaptics_touch_drv: high CPU load

cproc, 2021-01-06:
When running Sculpt on the i.MX8 EVK board (without touchscreen attached), 'top' reports a relatively high CPU load for the touch driver:

```
[init -> top] cpu=0.0   2     0  53.57%   0.00% thread='ep'                      label='init -> drivers -> touch_drv'
[init -> top] cpu=0.0   2     0  18.11%   0.00% thread='ep'                      label='init -> drivers -> gpio_drv'
[init -> top] cpu=0.0   0     0  15.71%   0.00% thread='irq'                     label='core'
[init -> top] cpu=0.0   2     0   7.27%   0.00% thread='gpio_drv'                label='init -> drivers -> gpio_drv'
[init -> top] cpu=0.0   0     0   2.97%   0.00% thread='entrypoint'              label='core'
[init -> top] cpu=0.0   3   819   1.24%   0.00% thread='ep'                      label='init -> timer'
[init -> top] cpu=1.0   0     0 100.00%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=2.0   0     0  95.58%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=2.0   1     0   1.38%   0.00% thread='ep'                      label='init -> leitzentrale -> nitpicker'
[init -> top] cpu=2.0   1     0   0.85%   0.00% thread='ep'                      label='init -> leitzentrale -> decorator'
[init -> top] cpu=2.0   1     0   0.56%   0.00% thread='ep'                      label='init -> leitzentrale -> gui_fb'
[init -> top] cpu=2.0   1     0   0.53%   0.00% thread='ep'                      label='init -> leitzentrale -> manager'
[init -> top] cpu=2.0   1     0   0.46%   0.00% thread='decorator'               label='init -> leitzentrale -> decorator'
[init -> top] cpu=3.0   0     0 100.00%   0.00% thread='idle'                    label='kernel'
```

skalk, 2021-01-07:
I assume it is the I2C implementation in the driver, which does active polling in its protocol implementation:

  https://github.com/genodelabs/genode/blob/master/repos/os/src/drivers/touch/synaptics_dsx/i2c.h#L69

When there is no part on the other side, it'll poll forever.

Of course, it would be nice to have a poll-free implementation on the one hand, but on the other it is somehow a general problem that we try to use several sets of device drivers (configurations) in one static driver setup here (drivers_interactiv-imx8q_evk package).

Either we split it into different packages again, or we have a measure to distinguish in between both setups (HDMI+USB HID or Touchscreen display) during runtime. Actually, I see no way to follow the last approach, apart from loading a setup-specific configuration, like its is done with device trees under Linux/*BSD.
Any opinions?

nfeske, 2021-01-07:
I think that splitting up the packages should not be the way to go. I'd rather prefer improved resilience towards unexpected conditions, like missing hardware.

How about giving up after a reasonable timeout? E.g., should the touchscreen not respond within 3 seconds, is there really any chance it will respond later?

Also, couldn't the code use the `wait_for` feature of the register framework so that the busy-ness remains limited even in the bad case? So during the 3 seconds of polling, it would not delay the boot of all the other components.

nfeske, 2022-05-25:
I sense that we won't address this issue unless the need for i.MX8 touch-screen support arises beyond demo scenarios.