genodelabs/genode #3980: Build errors due to files mentioned in dependency files

tomga, 2021-01-07:
Sometimes when working with Genode - mostly after switching branches - I experienced non-obvious problems with builds if they were performed in existing build directories.

Those build errors refer to not existing files and inform that those files are required to build some targets. They look similar to this: 

    make[3]: *** No rule to make target '/genode/repos/base-hw/src/core/spec/arm_v8/pd_session_support.cc', needed by 'pd_session_support.o'.  Stop.
    var/libdeps:76: recipe for target 'core-hw-rpi3.lib' failed
    make[2]: *** [core-hw-rpi3.lib] Error 2

The problem happens because automatically generated `.d` dependenc files may contain information about files that existed when target was previously built and now those files are missing. It can be due to moving files around in repository or some header files being added or removed.

If the nature of the issue is known than it is quick operation to fix build by simply removing proper `.d` file which internally is not required to exist. It is less quick if such error happens for the first time and missing file is some deeply included header :-)

After consulting it with @nfeske he suggested to create this issue with my idea for fixing this problem.

As written earlier build problem is because some files are mentioned as dependency and they do not exist now. There are to possibilities:
 * those files are not required for build now - in this case they can of course be safely removed from required dependencies,
 * those files are required for build now - in this case they also can be removed from dependencies and compiler will inform about missing files.

In both cases it is important to attempt to rebuild target.

Main rule of `.d` file (starting in first line) has form:

    tgt.o tgt.d: \
     dep1 \
     dep2

My idea would be to generate `.d2` file from `.d` to create something like:

    tgt.o tgt.d: \
     <MK MAGIC>
    DEPS_tgt := \
     dep1 \
     dep2

where `MK MAGIC` should do sth like below:

    if $(wildcard $(DEPS_tgt)) == $(DEPS_tgt) then $(DEPS_tgt) else <PHONY>

and including `.d2` instead of `.d` to get dependencies list.

Of course make guru should change it to proper make language (if possible) but the idea is to make no changes if all dependencies exist and force rebuild (`<PHONY>`) if any of the files are missing.

Given that this transformation from `.d` to `.d2` is just inserting some text after first line containing `tgt.o tgt.d: \` it should be easy to implement it with currently available tools.

nfeske, 2021-01-20:
Your idea is very cool. I will still need some more time to digest. In particular, I'd like to avoid the need for spilling the build directory with additional .d2 files.

nfeske, 2021-01-20:
This issue can be reproduced by the following steps:

1. Add the following line to repos/base/src/test/log/target.mk
   ```
   vpath %.cc $(PRG_DIR)/sub
   ```
2. Build the test/log target from within the build directory
   ```
   $ make -C build/x86_64 test/log
   ```
   This works as expected.

3. Move the main.cc file in the sub directory
    ```
    $ mkdir repos/base/src/test/log/sub
    $ mv repos/base/src/test/log/main.cc repos/base/src/test/log/sub/
   ```
4. Build again
   ```
   $ make -C build/x86_64 test/log
   ...
   make[2]: *** No rule to make target '/.../repos/base/src/test/log/main.cc', needed by 'main.o'.  Stop.
   ```