genodelabs/genode #3989: platform_drv(x86): support dynamic reconfiguration

alex-ab, 2021-01-13:
Up to now solely during startup the configuration is read and evaluated. With enabling dynamic re-configuration established sessions must be closed if the new configuration does not contain a valid policy for a driver anymore.

nfeske, 2021-01-25:
Very nice patch! Merged to staging.

nfeske, 2021-01-26:
When executing the base.run script on NOVA, the following problem occurs:

```
[init -> drivers] Warning: platform_drv: no route to service "Report" (label="platform_drv -> pci")
[init -> drivers -> platform_drv] Error: Report-session creation failed (label="pci", ram_quota=14336, cap_quota=3, buffer_size=4096)
[init -> drivers -> platform_drv] Error: Uncaught exception of type 'Genode::Service_denied'
[init -> drivers -> platform_drv] Warning: abort called - thread: ep
```

I think it would be sensible to keep the pci report disabled in the drivers-interactive package.

alex-ab, 2021-01-26:
4393d5d12b31d628ab8b5b34d55726db41f9223 contains a fixup. In the dynamic configuration case the pci report must be ever generated before actually a meaningful configuration can be constructed for the platform driver, because of that by default the report is generated.

alex-ab, 2021-01-26:
Sorry, 4d6437773473f4a94cbe2c3c30b03e514c821b63 contains also the case that the config is invalid.

nfeske, 2021-01-26:
Thanks for the quick response.

Instead of using an attribute, we should better follow the by-now established pattern of controlling reports via a <report> node. That is, the report should be enabled via

```
<config>
  ...
  <report pci="yes"/>
  ...
</config>
```

alex-ab, 2021-01-26:
c5a781a1ff98d982606aeaa29299f46561ad7494 contains the report node version.

nfeske, 2021-01-26:
The default value should be false to stay consistent with the pattern found everywhere else.

alex-ab, 2021-01-26:
> The default value should be false to stay consistent with the pattern found everywhere else.

As said, it is required not to be false by default in the dynamic case:

> In the dynamic configuration case the pci report must be ever generated before actually a meaningful configuration can be constructed for the platform driver, because of that by default the report is generated.

nfeske, 2021-01-26:
Please bear with me. Why can't the initial configuration contain the `<report>` node?

alex-ab, 2021-01-26:
When the configuration is dynamically generated, the component doing it must be run before the platform driver, must have reported before the platform driver is up and ask the report_rom. During developing it turned out that this was not feasible respectively racy. So not ever a valid ROM got reported by the report_rom and in turn no pci report got generated.

nfeske, 2021-01-26:
I'm still confused. Do we run into a limitation of the report_rom?

In my understanding, once the driver manager comes up, it would generate the initial config for the platform driver, enabling the report. If the platform driver came up before, it would see an invalid ROM but still registers its handler for the ROM update. Vice versa, the driver manager has subscribed to the PCI report, which initially does not exist either.

As soon as the driver manager generates the initial platform-driver config, the platform driver gets reconfigured to enable the PCI report. Now, the first meaningful PCI report appears, which prompts the driver manager to consume it and reconfigure the platform driver again.

If we follow this state-machine approach correctly on both sides, the scenario should not be prone to race conditions. It is important to follow certain rules, e.g., manually calling handle_config after installing the config-signal handler.

Can you help me pinpointing the part of this picture that I get wrong?

alex-ab, 2021-01-26:
Your picture is reasonable.

The remaining issue is that the platform driver would announce its service when seeing

```<config> <report pci="yes"/> </config>```

and all drivers try to establish sessions and would be denied and the system comes not up. The platform driver would need to consider such a configuration as not valid "enough" to announce the Platform service. With this change it should work. I will work an the change.

nfeske, 2021-01-26:
> The platform driver would need to consider such a configuration as not valid "enough" to announce the Platform service.

That is indeed an important point that I had not on my radar.

In the future, we may consider letting the platform driver defer the session creation instead of denying it in this case. So the driver would keep blocking for its platform session until a valid policy is established. But today, the synchronous way of session-creation via the root-component still stands in the way of that.

> With this change it should work. I will work an the change.

Thank you! I appreciate it.

alex-ab, 2021-01-26:
fd9480a6b0d956d974c4036f14b0d87849554064 contains the discussed changes, please review, thanks.

nfeske, 2021-01-26:
Thanks. I merged the fixup to staging.

When looking at the structure of the code, I think there are opportunities to make it more self-descriptive. E.g., it is currently not easy to connect the name of the method `update_acpi` with the code inside the method that implements actually some kind of state machine in an implicit way. However, let us leave the refactoring to another step.

commit 55cef015 (2021-01-25): platform_drv(x86): support dynamic reconfiguration

Fixes #3989
commit 6789ce8b (2021-02-23): platform_drv(x86): support dynamic reconfiguration

Fixes #3989
closed 2021-02-23 by nfeske