genodelabs/genode #3991: libc/vfs: tool/create_builddir on Genode creates invalid etc/build.conf

alex-ab, 2021-01-19:
The created ```x86_64/etc/build.conf``` by invoking ```tool/create_builddir x86_64``` on Genode looks as follow (start of file):

```
#
# Drivers ported from the OpenBSD
#
REPOSITORIES += $(GENODE_DIR)/repos/dde_bsd

#
# Drivers ported from iPXE
#                                             
REPOSITORIES += $(GENODE_DIR)/repos/dde_ipxe

rio
##
REPOSITORIES += $(GENODE_DIR)/repos/base
REPOSITORIES += $(GENODE_DIR)/repos/os
REPOSITORIES += $(GENODE_DIR)/repos/demo

...
```

During instrumenting a bit ```tool/create_builddir``` witch some debugging ```echo```s lead to a good ```build.conf```. So it seems, that a race exists between closing the file and re-opining by another component that start another append by using ```cat```.

chelmuth, 2021-01-19:
Out of curiosity: The tool was not run with `-j<num>`? I'm asking because *create_builddir* does not explicitly guard against parallel runs like follows.

```diff
--- a/tool/create_builddir
+++ b/tool/create_builddir
@@ -211,3 +211,5 @@ message:
 	@echo "Please adjust $(BUILD_DIR)/etc/build.conf according to your needs."
 
 .PHONY: $(PLATFORM)
+
+.NOTPARALLEL: $(PLATFORM)
```

alex-ab, 2021-01-19:
No, `-j` using on Genode I did not tried yet, this is for later. On issue_3991 the tool chain pkg with the tool chain is available and testable with depot_deploy.

nfeske, 2021-01-19:
This reminds me of an issue with a missing flush operation of a written-to file that I observed (and fixed) with the genodians.org scenario on NOVA. The flushing happens when closing a file descriptors. It is important that the libc-internal close_file_descriptors_on_exit function is called when the program exits, regardless of how it exits. The function is registered as atexit handler.

Maybe there exists a code path that exits the program w/o doing the atexit processing?

alex-ab, 2021-01-20:
It turned out that the seek state is inconsistent of file descriptors used in O_APPEND mode when used together with fork+wait. With the both commits 2 corner cases gets addressed and fixes for me the wrong content of the generated build.conf.

nfeske, 2021-01-20:
Thank you for investigating and fixing the problem. I'm surprised that we have not encountered it before, e.g., in the genodians.org scenario.

I merged both commits to staging.

closed 2021-01-25 by alex-ab