genodelabs/genode #3993: tool/run: search run-tool plugins in repositories

nfeske, 2021-01-19:
Until now, we need to tweak the RUN_OPT in etc/build.conf when using a board supported outside the genode repository, e.g., the boards supported in genode-world. This is described at https://genode.org/documentation/release-notes/20.05#Board_support_outside_the_Genode_main_repository.

To streamline the board support out of tree, I suggest to change the run tool to search includes in the tool/run/ sub directories of the used repositories. So the world/tool/run/boot_dir/hw would be picked up automatically.

In order to keep all boards supported by the genode repository functional in addition to the ones supported in world, the world/tool/run/boot_dir/hw snippet must be changed as follows:

```
source  [genode_dir]/tool/run/boot_dir/hw

rename bootstrap_link_address orig_bootstrap_link_address

proc bootstrap_link_address { } {

	if {[have_spec "arndale"]}   { return "0x88000000" }
	if {[have_spec "odroid_xu"]} { return "0x88000000" }
	if {[have_spec "panda"]}     { return "0x88000000" }

	return [orig_bootstrap_link_address]
}

rename base_src orig_base_src

proc base_src { } {

	if {[have_spec arndale]}   { return base-hw-arndale   }
	if {[have_spec odroid_xu]} { return base-hw-odroid_xu }
	if {[have_spec panda]}     { return base-hw-panda     }

	return [orig_base_src]
}
```

Note that the overwritten function calls the original function instead of returning an error as done in the current version (https://github.com/genodelabs/genode-world/blob/master/tool/run/boot_dir/hw).

nfeske, 2021-01-19:
Commit https://github.com/nfeske/genode/commit/be95fc50bd04fab71fea864bd95d62b80096f131 implements this suggestion.

What do @skalk and @cnuke think about it?

cnuke, 2021-01-19:
@nfeske looks good to me :+1:. I suppose the `puts stderr` call is intentionally?

nfeske, 2021-01-19:
> I suppose the puts stderr call is intentionally?

It's not. Thanks for the hint! ;-)

nfeske, 2021-01-19:
Commit https://github.com/genodelabs/genode-world/commit/e7f601f15f872a6217db1fb19da5452ed5b99b15 adjusts world in this respect.

skalk, 2021-01-19:
@nfeske I like the change! Very nive to use both "worlds" in one build directory. Originally, I'd liked to include the tool/run subdirectories of other repositories, but it seemed to be too invasive for the time back than. So I'm quite happy with that direction.

trimpim, 2021-01-20:
I like this change, It will simplify the integration for our imx8q based platform.

commit 48620c79 (2021-01-20): tool/run: search repos for run-tool plugins

This patch allows for supplementing the existing run-tool plugins with
additional plugins hosted inside a repository. E.g., it enables the run
tool to find world/tool/run/boot_dir/hw automatically.

Fixes #3993
closed 2021-01-25 by nfeske