genodelabs/genode #3995: Removal of support for Muen.sk

chelmuth, 2021-01-20:
I open this issue because I repeatedly fixed issues with *base-hw-muen* during the last years but did not see this port adopted by the community or even for cross correlating regressions/bugs in our local developments. On the other hand, the Muen sources use unfamiliar tooling and a quite complex build process (e.g., the sources must be copied to the build directory with TAR). For me personally it takes effort to learn again how it works each time a regression becomes apparent. After the regression is fixed the platform is disregarded until the another issue occurs during our nightly CI tests.

From this background I vote for removing the support for muen (or alternatively move the platform support to *genode-world*). What are your opinions (esp. @Kensan and @reet- )?

skalk, 2021-01-20:
I agree to your statement. Although, I really enjoyed the cooperation with @Kensan and @reet- some years ago, and still be thankful for their contribution! But I cannot observe continuation of this line of work . On the other side I experienced repeatedly friction too. The last time when building a docker image for Genode development including all third party software requirements. A significant time of the container composition, as well as the amount of tools necessary to add were due to dependencies of Muen. This is not about blaming the Muen build system. It is normal that you have to invest time for maintainance and integration work. But as long as nobody is benefiting from it, we should at least move it out of out test-infrastructure.

reet-, 2021-01-20:
After a long discussion with @Kensan, we decided that it would be best to just drop Muen support in Genode.

While this is certainly sad, we currently don't have the resources nor the use cases which would justify the amount of work to properly support a Muen port in genode-world.

Thanks.

nfeske, 2021-01-29:
Thank you @reet- and @Kensan for your immediate feedback.

I share Stefan's sentiment. Our collaboration - reaching back to 7 years ago - had been delightful. I truly appreciate both of you. With no concrete users of Genode on top of Muen, however, the social pressure we feel towards you remained as the sole motivation to uphold the feature. I sincerely hope that you are not too disappointed of us seeking to lift our maintenance burden.

I will prepare a commit. Should new use cases for the combination of Muen with Genode arise, this commit can be reverted as a starting point to revive the last working state from our project history and continue from there. But this effort would then be justified by a concrete use case.

reet-, 2021-01-29:
> I share Stefan's sentiment. Our collaboration - reaching back to 7 years ago - had been delightful. I truly appreciate both of you. With no concrete users of Genode on top of Muen, however, the [...]

We feel the same about our collaboration! Don't worry, we are not disappointed as your decision is totally understandable. Why should you maintain something no one (currently) uses?
 
> I will prepare a commit. Should new use cases for the combination of Muen with Genode arise, this commit can be reverted as a starting point to revive the last working state from our project history and continue from there. But this effort would then be justified by a concrete use case.

That was our way of thinking. We would revive the project if someone expresses a sincere interest.

commit bf23c3d6 (2021-01-29): Remove support for the Muen separation kernel

Fixes #3995
Fixes #3994
commit 8cc2662a (2021-02-23): Remove support for the Muen separation kernel

Fixes #3995
Fixes #3994
closed 2021-02-23 by nfeske