genodelabs/genode #3997: Crash in "usb_drv" on ThinkPad L380 Yoga

jjkarcher, 2021-01-20:
Hardware: Lenovo ThinkPad L380 Yoga

Situation:
When booting the "sculpt-20-08-b.img" image from the website using USB stick, according to the log output, the "usb_drv" component finds the various devices that are attached, but then it gives this message:

> [core] Warning: unresolvable exception 13, pd 'init -> drivers -> usb_drv', thread 'ep', cpu 0, ip=0x1005afc sp=40bfe7c0 bp=0x1410a450 no signal handler

(This was hand-typed, since I can't save the log itself.)

That is that last message from "usb_drv", and the USB stick is not recognized by Leitzentrale.

As requested on the ML, here are the outputs from:

> sudo lspci -vvvnn

[lspci.txt](https://github.com/genodelabs/genode/files/5843803/lspci.txt)

> sudo lsusb -vvv

[lsusb-nohub.txt](https://github.com/genodelabs/genode/files/5843835/lsusb-nohub.txt)

> sudo lshw  -sanitize

[lshw.txt](https://github.com/genodelabs/genode/files/5843802/lshw.txt)

I did a grep on the error message, and since it seems to occur in Nova, I think I will try to build a Sculpt/basehw 20.08 image, and see if it makes a difference.  (Note: Versions after 20.08 boot to a blank screen with only mouse pointer.)

Please let me know if I can provide more info, add debugging messages in the code, or test anything else.

Thanks!

cproc, 2021-01-21:
You can find out the source code location where the error occured as follows:

- build the USB driver with debug symbols: make drivers/usb
- in the 'debug' directory run: objdump -dSCl ./usb_drv | less
- search for the 'ip' address from the error message (1005afc)

When I do this on the 'sculpt-20.08' branch, I see the following code:

```
static bool led_match(struct input_handler *handler, struct input_dev *dev)
{
 1005ac7:       41 57                   push   %r15
/.../contrib/dde_linux-173719007927a108a9558e727b8c9b388f3159a2/src/lib/usb/include/linux/input.h:350
 1005ac9:       48 8d be 10 02 00 00    lea    0x210(%rsi),%rdi
 1005ad0:       49 bb 40 57 14 00 00    movabs $0x145740,%r11
 1005ad7:       00 00 00
 1005ada:       48 ba 90 f6 ee ff ff    movabs $0xffffffffffeef690,%rdx
 1005ae1:       ff ff ff
 1005ae4:       4c 01 d8                add    %r11,%rax
 1005ae7:       49 89 c7                mov    %rax,%r15
 1005aea:       48 01 c2                add    %rax,%rdx
 1005aed:       ff d2                   callq  *%rdx
/.../repos/dde_linux/src/drivers/usb/input/evdev.cc:490
        hid_device *hid = (hid_device *)input_get_drvdata(dev);
        hid_report *report;

        /* search report for keyboard entries */
        list_for_each_entry(report, &hid->report_enum[0].report_list, list) {
 1005aef:       4c 8b 40 58             mov    0x58(%rax),%r8
 1005af3:       48 83 c0 58             add    $0x58,%rax
 1005af7:       49 39 c0                cmp    %rax,%r8
 1005afa:       74 6c                   je     1005b68 <led_match(input_handler*, input_dev*)+0xa8>
/.../repos/dde_linux/src/drivers/usb/input/evdev.cc:492

                for (unsigned i = 0; i < report->maxfield; i++)
 1005afc:       41 8b 90 18 08 00 00    mov    0x818(%r8),%edx
```

The instruction at address 0x1005afc does not seem to be an FPU/SSE instruction, but maybe the referenced address points somewhere where for some reason a general protection fault is triggered instead of a regular page fault. To check this, you could add a debugging message above the 'for' loop like the following:

```
lx_printf("report: %p\n", report);
```

If the report address turns out to be obviously bogus, this would be a starting point to trace back to the origin of the address.

From looking at your 'lsusb' output, I wonder how your keyboard and touchpad are interfaced (I noticed only the touch screen, but maybe I overlooked something) and if the input device which triggered the error was actually the touch screen.

jjkarcher, 2021-01-21:
Thanks for the quick and informative response!

I'm using the internal keyboard and TrackPoint/TrackPad, which I assume are connected via PS/2, since they have always worked fine.

I'm having a little trouble getting "sculpt-20.08" to build (because of outdated third-party libs, if appears), but I'll add that logging and report back as soon as I get past that.

Regarding the touch screen, that's interesting.  If we resolve the crash, do you think the touch screen might actually work as an input device, based on the existing code?

jjkarcher, 2021-03-03:
Sorry for the delay in doing this.  The result is:

report: 4620383531352044

That is an odd-looking number for a pointer, and any attempt to use it (e.g., to print "report->maxfield") immediately dies.

They don't make it easy, do they?

jjkarcher, 2021-03-04:
I decided to poke around a little bit more with those structures, so I added a little more logging:

```
hid->report_enum[0]: 1800c8b0
hid->report_enum[0].numbered: 1868783959
hid->report_enum[0].report_list: 4620383531352044
hid->report_enum[1]: 1800d0c8
hid->report_enum[1].numbered: 1868783959
hid->report_enum[1].report_list: 0
hid->report_enum[2]: 1800d8e0
hid->report_enum[2].numbered: 1868783959
hid->report_enum[2].report_list: 0
hid->report_enum[3]: 1800e0f8
hid->report_enum[3].numbered: 1868783959
hid->report_enum[3].report_list: 0
```

I was following a hunch from reading the following comment in "drivers/usb_hid/include/linux/hid.h" - lines 299 - 307:

```
/*
 * HID report types --- Ouch! HID spec says 1 2 3!
 */

#define HID_INPUT_REPORT	0
#define HID_OUTPUT_REPORT	1
#define HID_FEATURE_REPORT	2

#define HID_REPORT_TYPES	3

```
It doesn't look like I've advanced the ball much, but let me know if I can add anything else.

nfeske, 2021-03-05:
@jjkarcher May I ask you to try out the most recent version of Sculpt (based on the current staging branch) on your machine?

You can find a ready-to-use image at https://genode.org/files/sculpt-21.03-test.img.gz

I'm asking since a recent fixes could potentially have an effect on the USB issue. In particular, the warnings observed in #4037 could have an IOMMU-related effect on USB controllers of older machines.

jjkarcher, 2021-03-05:
Thanks for the chance to test this!  Unfortunately, I still get the blank screen (with working mouse pointer).

nfeske, 2021-03-05:
Thanks a lot for the super quick response! The fact that the mouse pointer works is strange. Do you happen to use a USB mouse?

One possible explanation of this effect would be that the all components of the runtime subsystem (most of the leitzentrale GUI is now hosted there) starve because a high-priority driver is busy all the time. To test this hypothesis, I created a new image where I flattened most priorities. In particular, the runtime shares one priority band with the device drivers.

May you give the following image a try? https://genode.org/files/sculpt-21.03-test-2.img.gz

jjkarcher, 2021-03-05:
It's a miracle!  Not only does Leitzentrale work now, but USB storage does too!

That also means that I can now transfer log file instead of typing them by hand. :-)

USB HID devices still don't work, but I think we're getting close - here is an interesting log message:
```
[drivers -> usb_hid_drv] __kfifo_alloc not implemented
[drivers -> usb_hid_drv] BUG: failure at /plain/no/genode.git/depot/nfeske/src/usb_hid_drv/2021-03-02/src/drivers/usb_hid/dummies.c:542/__kfifo_alloc()!
```
Let me know if I can do anything else!

P.S. I am using the built-in TrackPoint and TrackPad, which I assume use PS/2, since they've always worked.

jjkarcher, 2021-03-06:
Re: the __kfifo_alloc() problem:

Just going a quick grep of the source, it looks like the Wacom driver ("depot/genodelabs/src/usb_hid_drv/2020-12-09/src/drivers/usb_hid/drivers/hid/wacom_sys.c") does use the kfifo feature, which makes sense, since the touchscreen is a Wacom unit.

BTW, @nfeske, can you send me the miracle patch, so I can build here? Thanks!

nfeske, 2021-03-07:
This new information is very helpful. Thank you @jjkarcher!

> can you send me the miracle patch, so I can build here? 

Here you go: 6d9646ac2a81e3ef4b93ae90bdcd626bb2cfa764

Given the information we just gained, I see two problems worth solving.

* We should remove the spinning in the failure case of the usb_hid_drv. Since this driver runs at a high priority, everything with a lower priority cannot run as soon as the error occurs. The problem lies here: https://github.com/nfeske/genode/blob/sculpt_21_03/repos/dde_linux/src/include/lx_emul/bug.h#L34 The while(1) should be replaced by something that infinitely blocks. (e.g., some function that calls Genode::sleep_forever, or one throws an unhandled exception) If we do this, my "magic" patch should be no longer needed.

* Adding a reasonable implementation of __kfifo_alloc

If you want to give it a try, you may consider using a debug commit similar to 7da7c0625601b3487bdcd65b9d14b62640a06e53, adding the drivers/usb_hid to the `build` list and usb_hid_drv to the `boot_modules` list. Then you can disable the --depot-auto-update in your etc/build.conf and get really quick development-build-test cycle (taking the packaging out of the loop).

jjkarcher, 2021-03-08:
Thanks for the patch and the tips - that really does speed things up!

I made a first attempt at implementing the required dummy functions by copying the function bodies directly from the Linux source into "dummies.c".  This went fairly smoothly (it only seems to need a subset of the "kfifo" functions) until I encountered what I feared from the beginning: kmalloc().  I'm afraid that's where my competence ends. :-(

I'm also out of my depth when it comes to your BUG() fix suggestion, but I'll give it a try if you can point me in the right direction.

nfeske, 2021-03-08:
> I'm also out of my depth when it comes to your BUG() fix suggestion, but I'll give it a try if you can point me in the right direction.

Sorry that I was a bit too vague. In commit https://github.com/nfeske/genode/commit/4e6170b9cd8d729f5c507e90ce07552a759556cf you can see what I meant. It replaces the `while (1) hack` with a function call to a new `lx_sleep_forever` C-ABI function. Since no suitable place for this implementation existed yet, I added the bug.cc compilation unit.

That leaves us the kfifo problem to solve. kmalloc should not be too problematic since many other drivers already use it. You can make the commonly used implementation available to the usb_hid_drv by adding the following line to the driver's lx_emul.cc file, following the pattern of, e.g., https://github.com/genodelabs/genode/blob/master/repos/dde_linux/src/drivers/usb_net/lx_emul.cc#L18

```
#include <lx_emul/impl/slab.h>
```

alex-ab, 2021-03-08:
@jjkarcher: may you please try the commit e04fa679db3b80af273d8cb15a14cec15d26e56b, which adds the kfifo implementation

alex-ab, 2021-03-08:
@jjkarcher: sorry, 96e21198b5d6ed7414157ce90ca4c63d02851cf9 contains also the updated hash for dde_linux (based upon @nfeske dde_linux: don't spin when panicking commit)

jjkarcher, 2021-03-09:
Thanks a million, @nfeske and @alex-ab!  Those patches worked like a charm.

Now I get an BUG() error for unimplemented devres_open_group() in "wacom_sys.c" (again the culprit).

Doing a grep, it looks like there is an implementation in "contrib/dde_linux-ddde9dc3e9076830babf9402445ed6b8a3022362/src/drivers/framebuffer/imx8/drivers/base/devres.c".  Assuming that this version will work, what is the correct way to integrate that into "usb_hid_drv"?

alex-ab, 2021-03-10:
> Doing a grep, it looks like there is an implementation in "contrib/dde_linux-ddde9dc3e9076830babf9402445ed6b8a3022362/src/drivers/framebuffer/imx8/drivers/base/devres.c". Assuming that this version will work, what is the correct way to integrate that into "usb_hid_drv"?

The linux files which are actually unpacked and used by usb_hid are specified by repos/dde_linux/usb_hid.list. The file you found is part of the imx8 framebuffer port (repos/dde_linux/imx8_fb.list).

You may now try to re-add more linux sources and see where it leads you (missing symbols/structs etc). This may become hairy, but probably is the only way to get the driver fully working. The other option is to return for the unsupported functions some error code in repos/dde_linux/src/drivers/usb_hid/dummies.c and see how the driver can life with the situation, e.g. for devres_open_group replace TRACE_AND_STOP just by TRACE.

alex-ab, 2021-03-16:
@jjkarcher: May you please give 42fd2e5c5f51c5844d054760ad15c457c6b28970 a spin, in order to check that the fault goes away (however with no working wacom driver). I'm asking, because we want to release the next Sculpt revision soon and rather should disable the driver temporarily until a acknowledged fix is available.

jjkarcher, 2021-03-16:
I can confirm that this patch does suppress the wacom driver, and thus the BUG() errors.

In doing so, it creates a mini-bug that makes me very happy:  If you touch the screen, the mouse pointer disappears, following which the TrackPoint/TrackPad still work, but the buttons do not.

This part makes me even happier: If you bring the pen (!) near the screen and/or click the screen, the mouse pointer disappears, but the TrackPoint/TrackPad and buttons still work afterward.  The coordinates are the problem, since you can move the cursor over the entire screen by moving the pen over about a 3cm x 2cm square in the upper left corner, and pressing activates the button!

(When the Wacom driver was running, the touchscreen and pen did nothing.)

I found the following warnings in the log, generated by "lx_emul.cc", to which I added the parameters:
```
[drivers -> usb_hid_drv] Warning: find_next_zero_bit_le() : Offset greater max size: size = 768; offset = 256; max_size = 64
[drivers -> usb_hid_drv] Warning: find_next_zero_bit_le() : Offset greater max size: size = 768; offset = 256; max_size = 64
[drivers -> usb_hid_drv] Warning: find_next_zero_bit_le() : Offset greater max size: size = 768; offset = 256; max_size = 64
[drivers -> usb_hid_drv] Warning: find_next_zero_bit_le() : Offset greater max size: size = 768; offset = 256; max_size = 64
[drivers -> usb_hid_drv] dev_info: input: USB HID v1.11 Device [HID 056a:5158] on usb-usbbus-/input0

[drivers -> usb_hid_drv] Warning: find_next_zero_bit_le() : Offset greater max size: size = 768; offset = 256; max_size = 64
[drivers -> usb_hid_drv] Warning: find_next_zero_bit_le() : Offset greater max size: size = 768; offset = 256; max_size = 64
[drivers -> usb_hid_drv] Warning: find_next_zero_bit_le() : Offset greater max size: size = 768; offset = 256; max_size = 64
[drivers -> usb_hid_drv] Warning: find_next_zero_bit_le() : Offset greater max size: size = 768; offset = 256; max_size = 64
[drivers -> usb_hid_drv] Warning: find_next_zero_bit_le() : Offset greater max size: size = 768; offset = 256; max_size = 64
[drivers -> usb_hid_drv] dev_info: input: USB HID v1.11 Mouse [HID 056a:5158] on usb-usbbus-/input1
```
Just using grep, I couldn't find the file calling find_next_zero_bit_le().  Is there a clever way to track this down?

Regarding the pen, I infer that it is being reported as the "Mouse" on "input1".  (In case you can't tell, this is making my day! ;^) )

jjkarcher, 2021-03-16:
Minor point: I should have tested better before writing my last comment, because I was wrong about the touchscreen - it works exactly the same as the pen!  I can use the entire screen from the top-left rectangle, and touching works as a button press also!

(And the mouse buttons still work afterward, so I don't know what I did wrong with that observation.)

Thanks for all the quick help!

I will read up on coordinate mapping next...

alex-ab, 2021-03-17:
> Just using grep, I couldn't find the file calling find_next_zero_bit_le(). Is there a clever way to track this down?

I get hits using grep in the contrib/dde_linux-<hash>. You can also try to take the [backtrace option](https://www.genodians.org/nfeske/2021-02-11-pine-fun-debugging), as described by the article by @nfeske.

alex-ab, 2021-03-17:
> I will read up on coordinate mapping next...

Presumably you have "just" to configure your screen width/height for [usb_hid configuration for Sculpt](https://github.com/genodelabs/genode/blob/master/repos/gems/recipes/raw/drivers_managed-pc/drivers.config#L155) read in by the [usb_hid driver](https://github.com/genodelabs/genode/blob/master/repos/dde_linux/src/drivers/usb_hid/main.cc#L197). The values should end up and to be used in [evdev.cc](https://github.com/genodelabs/genode/blob/master/repos/dde_linux/src/drivers/usb_hid/evdev.cc#L60)

nfeske, 2021-03-23:
I merged https://github.com/alex-ab/genode/commit/42fd2e5c5f51c5844d054760ad15c457c6b28970 as an interim solution for the upcoming Sculpt OS release to staging.

jjkarcher, 2021-03-24:
I have poked around a little bit, and it looks like the touchscreen and pen sometimes generate "unconventional" codes to the HID driver, which can cause the latter to seize up.  Judging from the comments in the Linux driver source, this is a common problem.  I guess that's the reason for the Wacom driver in the first place. ;^)

Question: Since the crash is gone, do you want to close this ticket, and open a new one for making the Wacom driver work?  Or is it better to just continue here?

nfeske, 2021-03-25:
> Since the crash is gone, do you want to close this ticket, and open a new one for making the Wacom driver work?

Thanks for being so considerate. Now that you are asking, I would indeed prefer closing this issue since none of the Genode core developers is actively pursuing the Wacom-driver work. On the other hand, we can of course not demand from you to work on it. So there is the risk of this issue becoming stale or putting some kind of pressure on either side.

If you are fine to close the issue, please go ahead. Should you be able to bring the Wacom driver to work, let's celebrate that with a dedicated issue. :-)

jjkarcher, 2021-03-25:
Sounds good.  I will move my Wacom driver questions over to the ML.  Thanks!

closed 2021-03-25 by jjkarcher