genodelabs/genode #3998: Remove need to customize tool/run/boot_dir/hw

nfeske, 2021-01-21:
In the context of issue #3993, the mechanism for customizing the tool/run/boot_dir/hw as currently needed when hosting a board support in an external repository (like genode-world) got slightly improved. However, the need for customization is unfortunate in the first place. I propose introducing a more formal solution of the two current customizations to eliminate the need to host potentially ambiguous tool/run/boot_dir/hw files in multiple repositories.

Only two functions are subjected to customization:

* `bootstrap_link_address` returns the link address of the final system image, depending on the used board. It is a physical address that must point to somewhere in physical memory. Since the RAM layout differs between boards, the value is board-specific.
* `base_src` is a support function for the package management. It returns the name of the source archive for the base-hw for a given board.

With the following conventions, we can make those functions generic across boards.

* Each base-src archive for given board is called `base-hw-[board]`. With this convention - which is already in place informally - the `base_src` can be simplified to a simple string concatenation.

* We create a new directory repos/base-hw/board/ as a designated place for board-property information. It has a sub directory for each board, named after the board. E.g., repos/base-hw/board/pbxa9/. Within the board-specific directory, there is a file called `image_link_address` that contains the value currently returned by the `bootstrap_link_address` function.
  With this directory structure in place, the `bootstrap_link_address` function becomes able to read the return value from the file that matches `[board]`.

@skalk do you find this idea appealing? If yes, I'd go ahead with implementing it. It would address concerns raised by @tomga regarding the composeability of board-support repositories, and it would ease the process of adding new boards (no need to understand the customization mechanism of the run tool plugins).

nfeske, 2021-01-21:
Commit https://github.com/nfeske/genode/commit/2b1e4d5e4037add7028d197678b66c5a24ece060 implements the idea.

skalk, 2021-01-21:
@nfeske, yes I like the idea. I would suggest to use the same pattern of base_src for other depot packages resp. for other kernels too. The only question that arose to me is, what about non-existent board packages? By now, the function exited the run-script without an error. But if we always return a package-name you'll be informed by error that the specific package couldn't be found.
It is not a problem for me personally, but users got informed with a specific hint beforehand. Now, they'll see a more common error message.

nfeske, 2021-01-21:
> Now, they'll see a more common error message.

I just gave this a try by adding the following line at the beginning of the log.run script:
```
import_from_depot [depot_user]/src/[base_src]
```

The resulting error looks like this:
```
make run/log KERNEL=hw BOARD=pine_a64lts
including /.../tool/run/boot_dir/hw
including /.../repos/base/run/log.run
Error: recipe for 'src/base-hw-pine_a64lts' not found - unable to guess version
Makefile:329: recipe for target 'run/log' failed
```

It does not look too bad to me.

nfeske, 2021-01-21:
Or did I misunderstood? You mean the exit 0 was deliberate? I have to admit that have not considered that.

I wonder, what is the situation where the successful exit in this case is desirable?

nfeske, 2021-01-21:
Let us try the CI chewing on it.

skalk, 2021-01-21:
It is not that important, I just wanted to hint that until now, an ingenous user what have seen:

  ``Test requires base-hw kernel archive, which is missing for this build configuration``

without additional make-errors. And now the message becomes a bit more cryptic. That is all.

m-stein, 2021-01-25:
Sorry for writing on this that late. A minor suggestion from my side: We already use ``spec`` as subdir name whenever we have hardware specifics in base-hw. What about using the scheme ``base-hw/spec/pbxa9`` instead of ``base-hw/board/pbxa9`` in order to stay consistent?

skalk, 2021-01-25:
@m-stein: the `spec` subdirectories gather all pathes that are derived via `SPEC` variables. But boards are no longer a `SPEC` variable. Only kernels and architectures like x86_64 etc. are defined as `SPEC` variables.

commit 916fa7f1 (2021-01-21): base-hw: introduce board-property directories

This patch moves the bootstrap-link-address information from the
tool/run/boot_dir/hw file to board-specific property files that can be
accessed by using the board as key. This eliminates the need to
customize boot_dir/hw when hosting board-support in an external
repository.

Fixes #3998
closed 2021-01-25 by nfeske