genodelabs/genode #4000: run/bash: Error: deadlock ahead

rite, 2021-01-26:
When running `KERNEL=linux BOARD=linux make -C build/x86_64 run/bash`, I get the following log output after issuing a command like `ls` within "genode's bash":

```
Genode 20.11-102-ga981fb864c
...
[init -> /bin/bash]   0x1000000 .. 0x10ffffff: linker area
[init -> /bin/bash]   0x40000000 .. 0x4fffffff: stack area
[init -> /bin/bash]   0x50000000 .. 0x601c7fff: ld.lib.so
[init -> /bin/bash]   0x10e0b000 .. 0x10ffffff: libc.lib.so
[init -> /bin/bash]   0x10d6f000 .. 0x10e0afff: vfs.lib.so
[init -> /bin/bash]   0x10d35000 .. 0x10d6efff: libm.lib.so
[init -> /bin/bash]   0x10d25000 .. 0x10d34fff: posix.lib.so
[init -> /bin/bash] Error: deadlock ahead, mutex=0x10ff9610, return ip=0x50053716
```

...and bash deadlocks. Is anyone else experiencing this problem?

nfeske, 2021-01-26:
Thanks for reporting. I immediately tried to reproduce the problem but it works here.

Just to double-check that there is no possible inconsistency with the depot, have you enabled the `--depot-auto-update` RUN_OPT in your build.conf?

While comparing your output to mine, @chelmuth noticed that ld.lib.so is slightly larger (2 pages) in my case.

rite, 2021-01-26:
> I immediately tried to reproduce the problem but it works here.

Thanks for checking so quickly! It doesn't surprise me you couldn't reproduce - if you could, you most likely would have noticed the problem long before me :)

> Just to double-check that there is no possible inconsistency with the depot, have you enabled the `--depot-auto-update` RUN_OPT in your build.conf?

I enabled  `--depot-auto-update`  just now and re-run. No depot archives have been recreated, and still the same problem. Also after completely deleting the `api/bin/pkg/raw/src`-directories in my depot, combined with `make cleanall`.

> While comparing your output to mine, @chelmuth noticed that ld.lib.so is slightly larger (2 pages) in my case.

The size of my `ld.lib.so` didn't change after the rebuild:

```
[init -> /bin/bash]   0x1000000 .. 0x10ffffff: linker area
[init -> /bin/bash]   0x40000000 .. 0x4fffffff: stack area
[init -> /bin/bash]   0x50000000 .. 0x601c7fff: ld.lib.so
[init -> /bin/bash]   0x10e0b000 .. 0x10ffffff: libc.lib.so
[init -> /bin/bash]   0x10d6f000 .. 0x10e0afff: vfs.lib.so
[init -> /bin/bash]   0x10d35000 .. 0x10d6efff: libm.lib.so
[init -> /bin/bash]   0x10d25000 .. 0x10d34fff: posix.lib.so
[init -> /bin/bash] Error: deadlock ahead, mutex=0x10ff9610, return ip=0x50053716
```

I'm using Arch Linux as build and debug environment. I can check if I'm able to successfully execute `run/bash` of previous Genode versions on my system.

chelmuth, 2021-01-26:
For clarification, the output for *ld.lib.so* is currently not entirely correct, as it also includes the size of the stack area.
```
[init -> /bin/bash]   0x50000000 .. 0x601c7fff: ld.lib.so
```
should actually be logged as
```
[init -> /bin/bash]   0x50000000 .. 0x501c7fff: ld.lib.so
```
But nevertheless, the question is: Why is your binary 2 pages smaller than ours.

@ssumpf do you have an idea how to fix this? The reason is that `Linker::dump_link_map()` just adds `Object::size()` to `Link_map::addr` but the the first ELF segment of *ld-linux.lib.so* is the stack area at `0x40000000`.

nfeske, 2021-01-26:
> But nevertheless, the question is: Why is your binary 2 pages smaller than ours.

This mystery is solved. I happened to have the following line in my etc/tools.conf:

```
CC_OPT += -fno-omit-frame-pointer
```

With this option disabled, I get the same size value as @rite.

rite, 2021-01-26:
Not sure if it helps to narrow down the problem but I just executed `run/bash` on Genode 20.02:

```
Genode 20.02 <local changes>
...
[init -> /bin/bash]   0x40000000 .. 0x4fffffff: stack area
[init -> /bin/bash]   0x50000000 .. 0x601cefff: ld.lib.so
[init -> /bin/bash]   0xae13000 .. 0xaffffff: libc.lib.so
[init -> /bin/bash]   0xad84000 .. 0xae12fff: vfs.lib.so
[init -> /bin/bash]   0xad4a000 .. 0xad83fff: libm.lib.so
[init -> /bin/bash]   0xad3a000 .. 0xad49fff: posix.lib.so
Warning: blocking canceled in entrypoint constructor
Warning: invalid signal-context capability
[init -> /bin/bash] Error: libc suspend() called from non-user context (0xaf5f61d) - aborting
[init -> /bin/bash] Error: libc suspend() called from non-user context (0xaf5f61d) - aborting
[init] child "/bin/bash" exited with exit value 1
```

The commands I issued within bash are:

```
bash-4.4# ls
bin  dev  home  lib  share  tmp
bash-4.4# cd home
bash-4.4# ls
```

What might be interesting perhaps, is that the depot tool creates a new version for the `src/libc`-archive, which indicates that my toolchain creates different code... This is not the case for the current Genode version thought.

nfeske, 2021-01-26:
> Not sure if it helps to narrow down the problem but I just executed run/bash on Genode 20.02:

I'm afraid that this version is too old for drawing a meaningful connection to now. The turned the guts of the libc around by 180 degrees in summer.

That leaves GDB as a tool to investigate. Can you try the following?

1. enable `CC_OPT += -fno-omit-frame-pointer` in your etc/tools.conf to help GDB to produce meaningful backtraces.
2. build noux-pkg/bash in the build directory to produce the unstripped binary of the bash executable.
3. add `bash.tar` to the `boot_modules` in the bash.run script to ensure that the bash binary corresponds with the version built in the build directory (using the `-fno-omit-frame-pointer` option)
3. create a symlink in <build-dir>/debug/bash pointing to the unstripped bash binary (`bash -> ../noux-pkg/bash/bash`).
4. create a symlink in <build-dir>/debug/ld.lib.so pointing to ld-linux.lib.so in the same directory (so GDB will pick up the Linux version of the dynamic linker when asking for ld.lib.so).
4. execute the bash.run script to the point where the deadlock-ahead message appears
5. use a different terminal to cd into the <build-dir>/debug directory
6. determine the PID of the bash running on Genode (e.g., `ps aux | grep "Genode.*bash"`)
7. attach GDB to the bash process while pointing GDB to the right binary: `gdb -p <PID> bash`
8. print the backtraces of the various threads using the GDB commands `info threads`, `thread <N>`, `bt`.

rite, 2021-01-26:
Thanks for the excellent step-by-step instructions, @nfeske.

### Threads
```
(gdb) info threads
  Id   Target Id             Frame 
* 1    LWP 25490 "ld.lib.so" pseudo_end () at /home/rite/projects/genode/repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
  2    LWP 25491 "ld.lib.so" pseudo_end () at /home/rite/projects/genode/repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
  3    LWP 25492 "ld.lib.so" pseudo_end () at /home/rite/projects/genode/repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
  4    LWP 25493 "ld.lib.so" pseudo_end () at /home/rite/projects/genode/repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
  5    LWP 25494 "ld.lib.so" pseudo_end () at /home/rite/projects/genode/repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
```

### Backtrace of thread 1
```
#0  pseudo_end () at /home/rite/projects/genode/repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
#1  0x00000000500511ad in lx_recvmsg (flags=0, msg=0x400ffa70, sockfd=...) at /home/rite/projects/genode/repos/base-linux/src/lib/syscall/linux_syscalls.h:208
#2  Genode::ipc_call (dst=..., snd_msgbuf=..., rcv_msgbuf=...) at /home/rite/projects/genode/repos/base-linux/src/lib/base/ipc.cc:342
#3  0x000000005004bef4 in Genode::Capability<(anonymous namespace)::Constructor>::_call<(anonymous namespace)::Constructor::Rpc_construct> (
    args=<synthetic pointer>..., this=0x400ffc38) at /home/rite/projects/genode/repos/base/include/base/rpc_client.h:141
#4  Genode::Capability<(anonymous namespace)::Constructor>::call<(anonymous namespace)::Constructor::Rpc_construct> (this=0x400ffc38)
    at /home/rite/projects/genode/repos/base/include/base/capability.h:166
#5  (anonymous namespace)::invoke_constructor_at_entrypoint (cap=...) at /home/rite/projects/genode/repos/base/src/lib/base/entrypoint.cc:325
#6  Genode::Entrypoint::Entrypoint (this=0x500fdea0 <Genode::bootstrap_component()::startup+352>, env=...)
    at /home/rite/projects/genode/repos/base/src/lib/base/entrypoint.cc:349
#7  0x0000000050043298 in Genode::Startup::Startup (this=0x500fdd40 <Genode::bootstrap_component()::startup>)
    at /home/rite/projects/genode/repos/base/src/lib/base/component.cc:239
#8  Genode::bootstrap_component () at /home/rite/projects/genode/repos/base/src/lib/base/component.cc:254
#9  0x00000000500b4b40 in _main () at /home/rite/projects/genode/repos/base/src/lib/startup/_main.cc:75
#10 0x0000000000000000 in ?? ()
```

### Backtrace of thread 2
```
#0  pseudo_end () at /home/rite/projects/genode/repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
#1  0x0000000050052338 in lx_futex (val=0, op=0, uaddr=<optimized out>) at /home/rite/projects/genode/repos/base-linux/src/lib/syscall/linux_syscalls.h:441
#2  thread_stop_myself (myself=<optimized out>) at /home/rite/projects/genode/repos/base-linux/src/include/base/internal/lock_helper.h:62
#3  Genode::Lock::lock (
    this=this@entry=0x10ff9690 <unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+16864>, myself=...) at /home/rite/projects/genode/repos/base/src/lib/base/lock.cc:128
#4  0x000000005005285f in Genode::Mutex::acquire (
    this=this@entry=0x10ff9690 <unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+16864>)
    at /home/rite/projects/genode/repos/base/src/lib/base/mutex.cc:25
#5  0x0000000050057497 in Genode::Mutex::Guard::Guard (mutex=..., this=<synthetic pointer>) at /home/rite/projects/genode/repos/base/include/base/mutex.h:62
#6  Genode::Registry_base::_insert (
    this=0x10ff9690 <unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+16864>, 
    element=...) at /home/rite/projects/genode/repos/base/src/lib/base/registry.cc:60
#7  0x0000000010f57061 in Genode::Registry<Libc::Monitor::Job>::Element::Element (obj=..., registry=..., this=0x401fd230)
    at /home/rite/projects/genode/repos/base/include/base/registry.h:108
#8  Libc::Monitor::Pool::monitor (job=..., 
    this=0x10ff9688 <unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+16856>)
    at /home/rite/projects/genode/repos/libports/src/lib/libc/internal/monitor.h:130
#9  Libc::Kernel::_monitor (
    this=0x10ff54b0 <unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>, fn=..., 
    timeout_ms=0) at /home/rite/projects/genode/repos/libports/src/lib/libc/internal/kernel.h:553
#10 0x0000000010f89bb8 in Libc::Monitor::monitor<Libc::Vfs_plugin::stat(char const*, stat*)::<lambda()> > (timeout_ms=0, fn=..., this=<optimized out>)
    at /home/rite/projects/genode/repos/libports/src/lib/libc/internal/monitor.h:82
#11 Libc::Vfs_plugin::stat (this=<optimized out>, path=<optimized out>, buf=<optimized out>)
    at /home/rite/projects/genode/repos/libports/src/lib/libc/vfs_plugin.cc:795
#12 0x0000000010f37c00 in stat (path=path@entry=0x10eb057 ".", buf=buf@entry=0x401fd980) at /home/rite/projects/genode/repos/os/include/os/path.h:239
#13 0x00000000010af502 in getcwd (buf=buf@entry=0x401fe090 "", size=size@entry=512)
    at /home/rite/projects/genode/contrib/bash-02edac1a780fc13b59c3c8ad64cc6d390d1e9cc5/src/noux-pkg/bash/lib/sh/getcwd.c:141
#14 0x0000000010f408e9 in Libc::Child_config::<lambda()>::operator()(void) const (__closure=0x401fe7f0)
[... cut ...]
    at /home/rite/projects/genode/repos/base/src/lib/base/entrypoint.cc:127
#9  0x000000005009f0c4 in Genode::Entrypoint::Signal_proxy_thread::entry (this=0x10ffd060 <fork_kernel_routine()::fork_ep+3776>)
    at /home/rite/projects/genode/repos/base/include/base/entrypoint.h:81
#10 Genode::Thread::_thread_start () at /home/rite/projects/genode/repos/base-linux/src/lib/base/thread_linux.cc:80
#11 0x00000000500b4f2b in thread_start () at /home/rite/projects/genode/repos/base-linux/src/lib/syscall/spec/x86_64/lx_clone.S:62

nfeske, 2021-01-26:
Thanks for insightful backtrace. Thread 2 is the important one.

During fork, which is executed within the libc monitor (as it should), the config for the new child is generated. The code calls `getcwd`, which unfortunately can have the side effect of I/O operations like `stat`. Such operations, in turn, assume that they originate from the application layer, and hence try to enter the monitor of the libc kernel. This ultimately results in the nested attempt to acquire the libc-monitor's mutex.

To make matters worse, the bash overrides the libc' `getcwd` function by bringing an own implementation. So the flow of control actually leaves the libc and moves right into the bash code.

To solve this issue, we have to find a way to side-step the call of `getcwd` while generating the config, maybe by executing it before fork enters the monitor and passing the information along the call chain. Please give me a bit of time to digest the problem.

I'm puzzled that you are the first triggering it.

nfeske, 2021-01-26:
Upon closer inspection, the libc-internal `getcwd` is safe because we implement `__getcwd` without I/O side effects (https://github.com/genodelabs/genode/blob/master/repos/libports/src/lib/libc/file_operations.cc#L735).

But the bash overriding getcwd prevails. Interestingly, in my version of bash, the overriding does not happen:

```
build/x86_64$ nm debug/bash |grep getcwd
0000000001120e20 D bash_getcwd_errstr
                 U getcwd
```

The symbol is missing, which is good. Could you please execute this command with your bash binary. I expect that in your case, `getcwd` is not undefined.

rite, 2021-01-26:
> The symbol is missing, which is good. Could you please execute this command with your bash binary. I expect that in your case, `getcwd` is not undefined.

Indeed:
```
$ nm debug/bash |grep getcwd
0000000001120e60 D bash_getcwd_errstr
00000000010af480 T getcwd
```

nfeske, 2021-01-26:
That means, the configure script of bash when cross-compiling for Genode somehow behaves differently on Arch than on Debian. In my case, the configure script is happy with my getcwd:
```
build/x86_64$ grep GETCWD noux-pkg/bash/config.h 
#define HAVE_GETCWD 1
/* #undef GETCWD_BROKEN */
```

To continue the investigation, I'm afraid that there is no way around digging into noux-pkg/bash/config.log. If you want to repeatedly and interactively issue the configure script for bash (I know you want, won't you?), you may find the generated shell snippet at noux-pkg/bash/env.sh handy. By including it via `source env.sh`, it sets all environment variables as needed for the Genode build.

```
build/x86_64$ cd noux-pkg/bash
build/x86_64/noux-pkg/bash$ source env.sh
<gnu_build>/home/rite/projects/genode/contrib/bash-02edac1a780fc13b59c3c8ad64cc6d390d1e9cc5/src/noux-pkg/bash/configure 
...
```

rite, 2021-01-27:
It isn't happy with my getcwd:
```
#define HAVE_GETCWD 1
#define GETCWD_BROKEN 1
```

So I guess I indeed "want to repeatedly and interactively issue the configure script for bash" ;) Thanks for the pointers @nfeske!

rite, 2021-02-24:
I can't reproduce the error anymore. I'm not sure but I guess it might be related to a Arch Linux update. Nevertheless, this issue can be closed.

closed 2021-02-24 by rite

nfeske, 2021-02-24:
Thanks for not letting the issue become stale. I appreciate it.