genodelabs/genode #4004: Build-system support for ccache

nfeske, 2021-01-29:
Since ccache is - strictly speaking - orthogonal to the build system, configuring the build system for the use of ccache was left to each developer. Setting up ccache is not straight-forward though. One must manually create hooks (symlinks shadowing the compiler executables), tweaking the PATH environment variable, and customizing the CROSS_DEV_PREFIX in etc/tools.conf. In short, only seasoned developers jump through those hoops. Many others miss out on the joys of ccache.

The build-system front end should better make ccache easily available using a simple option in the etc/build.conf file.

nfeske, 2021-01-29:
Commit https://github.com/genodelabs/genode/commit/05350b898346794e664c424a335154907d8d466d implements this feature.

cnuke, 2021-02-06:
Please merge 54da9b4 to staging. This moves the `check_tool` function definition to global scope (it used to be guarded by `STATIC_ANALYZE`.

On a side node, creating the symlinks does not work when using a recent GNU make version 4.3 - it works fine with 4.1. I have not checked why that is though.

nfeske, 2021-02-06:
Thanks @cnuke. I merged the fixup to staging. I'd very much appreciate your further investigation about the symlink mystery.

nfeske, 2021-02-06:
> I did not test it thoroughly but it seems to work. I know that run tool will not use ccache but it doesn't seem to compile directly much. Anything more?

@tomga the approach works for the Genode build system but breaks when invoking a 3rd-party build system and funneling CROSS_DEV_PREFIX (or CUSTOM_CC, CUSTOM_CXX) to the corresponding variables of the 3rd-party build system (e.g., GNU build system as called for the noux-pkg targets, or the Fiasco.OC build system). The whitespace is a recipe for a quoting disaster to happen. The symlink approach sidesteps these problems.

tomga, 2021-02-06:
> the approach works for the Genode build system but breaks when invoking a 3rd-party build system

@nfeske, thanks. I expected that I'm missing some important use case.

ssumpf, 2021-02-17:
@nfeske: 835eae2 fixes the nested build problem we debugged this morning.

nfeske, 2021-02-17:
Thanks a lot!

tomga, 2021-02-17:
@ssumpf, @nfeske would you be willing to describe what case does it fix? Yesterday I implemented similar solution in my experiment to use Scons to build Genode and I'd like to know where should I expect a change and why.

nfeske, 2021-02-17:
The Genode build system is called in a nested way when executing a run script. Outer build system -> run tool -> inner build system.

The problem was that the inner build-system inherited the CUSTOM_CXX variable as redirected by the outer build system and tried to redirect it another time. It mistook the already redirected CUSTOM_CXX (pointing to the symlink inside var/tool/) as tool-chain location.

The fix ensures that CUSTOM_CXX is not inherited from outside. So the inner build system performs like the outer one.

The question of why different make versions differ in behavior is not clear. Presumably, the handling of environment variables has changed in a subtle way.

@tomga Is this explanation helpful for you?

tomga, 2021-02-17:
@nfeske thanks. It is clear now. Somehow I expected a more subtle problem, possibly with some rare cases where `CUSTOM_CC` is modified in the library/program build. For example lines below can cause some problems but did not investigated it.

```
tool/autopilot:395:		exec echo "CUSTOM_CC  = ccache \$(CROSS_DEV_PREFIX)gcc" >> $tools_conf
repos/libports/lib/mk/luacxx.mk:10:CUSTOM_CC = $(CXX)
```

Regarding that build system nesting you mention, it is something I had in my "queue" of open problems how to avoid in Scons based builds. I planned to ask on the mailing list for an opinion how to retrieve list of build targets from run script. That would allow to first perform build in one step and later execute run tool. Unfortunately this inner build system can be started more than once and I have no good solution for it yet.

nfeske, 2021-02-17:
You may consider modifying the run tool, or execute the run script by some kind of dummy run tool that custom (or empty) implementation of the run tool's procedures. For example, you could override the run tool's `build` procedure to merely collect the arguments but not trigger any build. Once the run script calls `build_boot_image`, your version of this function would merely output the collected build arguments and exit. So the run script would be evaluated twice. Once for the information gathering, and once for the actual integration and test steps. That is just a vague idea though.

tomga, 2021-02-17:
> Once the run script calls `build_boot_image`, your version of this function would merely output the collected build arguments and exit

Thanks for this suggestion but unfortunately it will not work - at least without modifications - because it is the `build_boot_image` function calls `run_boot_dir` which is implemented differently in different boot directories. And those different implementations call `build` with own targets. Maybe if I could depend on that all those different implementations call build exactly one time I could change `build` procedure to exit if it is called from inside of `build_boot_image`. I'll check all implementations of `run_boot_dir` later to see how it is now.

closed 2021-02-23 by nfeske

nfeske, 2021-03-05:
Commits https://github.com/genodelabs/genode/commit/5a0af777afdf13c1713bc06d894c3d9e22a1d110 https://github.com/genodelabs/genode/commit/5ab982ce8ccefa0eb0f71789ed9d550b1223a31b and https://github.com/genodelabs/genode/commit/9e8b37d6597a69f7251ff67b43aebf6826871852 extend the ccache support to the depot tools and the --depot-auto-update feature of the run tool.

commit 05350b89 (2021-01-29): Add ccache support to build system

This patch simplifies the use of ccache with the build system. Up until
now, each developer had to set up the ccache hooks manually, adjust the
PATH variable, and customize the etc/tools.conf in each build directory.
With the patch, ccache can be enabled by un-commenting a single line in
the etc/build.conf file.

Fixes #4004
commit cbae9bc1 (2021-02-23): Add ccache support to build system

This patch simplifies the use of ccache with the build system. Up until
now, each developer had to set up the ccache hooks manually, adjust the
PATH variable, and customize the etc/tools.conf in each build directory.
With the patch, ccache can be enabled by un-commenting a single line in
the etc/build.conf file.

Fixes #4004