genodelabs/genode #4007: usb_hid_drv: stability improvements

cproc, 2021-02-01:
Fix some potential page faults and memory leaks.

closed 2021-02-23 by nfeske

ssumpf, 2023-06-21:
@cproc: Can you remember why the `mutex_destroy` was necessary? I ask because I am not sure if I need this in the new version. What would happen in the current driver without commit https://github.com/genodelabs/genode/commit/caeba13dd37cbf3ee47e4a59fe3fe667af4745e9? Thanks.

chelmuth, 2023-06-21:
@ssumpf do you use *dde_linux/src/lib/lx_emul/shadow/drivers/input/evdev.c*? In this case, the patch should not be required.

ssumpf, 2023-06-21:
> @ssumpf do you use _dde_linux/src/lib/lx_emul/shadow/drivers/input/evdev.c_? In this case, the patch should not be required.

Thanks for the hint, I just started with `evdev`. Will have a look.

cproc, 2023-06-21:
> @cproc: Can you remember why the `mutex_destroy` was necessary? I ask because I am not sure if I need this in the new version. What would happen in the current driver without commit [caeba13](https://github.com/genodelabs/genode/commit/caeba13dd37cbf3ee47e4a59fe3fe667af4745e9)? Thanks.

I believe it fixed memory leaks on device reconnection because of the waiters list allocation:

https://github.com/genodelabs/genode/blob/79e262921e554baf565156b504ead70782913deb/repos/dde_linux/src/include/legacy/lx_emul/impl/mutex.h#L50-L55

commit caeba13d (2021-02-02): usb_hid_drv: stability improvements

Fixes #4007
commit 33406940 (2021-02-23): usb_hid_drv: stability improvements

Fixes #4007