genodelabs/genode #4010: fec_nic_drv: strange behavior during DHCP

m-stein, 2021-02-02:
While debugging the recently failed ping test on imx6q_sabrelite, I found that the fec_nic_drv shows quite some strange behavior even before the Uplink-session commits.

The problem on current staging is that the DHCP client of the NIC router uplink sometimes can't finish (for like 5 minutes at least). Sometimes, however it finishes very fast, and sometimes it finishes after a while (seconds to minutes). When looking into the packets at the driver, I see that the driver sometimes forwards the same server response twice and sometimes doesn't receive a response that was definitely send by the DHCP server. Even in the good case it seems to be normal that the router has to repeat DHCP multiple times.

The situations doesn't seem to differ a lot with the state directly before
f6d195a9de nic drivers: provide optional Uplink-client mode.
For instance, when I added the packet log (that wasn't a problem on staging) to the driver, DHCP ended up in endless retries with strange responses. The same when I merely removed the second, unnecessary ping from the scenario.

I got the feeling that, even if there should be bugs in the recent Uplink adaption, the driver itself already was in such an unstable condition that it isn't sensible to debug the Uplink adaptions until the underlying FEC problems are fixed.

Here are some of my logs filtered for only the driver output and DHCP packets (``----`` marks a new DHCP client attempt, these are all cases where DHCP worked out in the end):

Staging with the driver in NIC-server mode and a NIC router
```
[init -> drivers -> nic_drv] --- freescale ethernet driver ---
[init -> drivers -> nic_drv] netdev_info: registered PHC device 0
[init -> drivers -> nic_drv] dev_info: attached PHY driver [Generic PHY] (mii_bus:phy_addr=fec-1:06, irq=POLL)
[init -> drivers -> nic_drv] netdev_info: Link is Up - 1Gbps/Full - flow control rx/tx
[init -> nic_router] [uplink] NIC sessions: 1
-------------------------------------
[init -> nic_router] [uplink] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
...
-------------------------------------
[init -> nic_router] [uplink] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
[init -> drivers -> nic_drv] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
[init -> nic_router] [uplink] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
-------------------------------------
[init -> nic_router] [uplink] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> nic_router] [uplink] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
[init -> drivers -> nic_drv] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
...
-------------------------------------
[init -> nic_router] [uplink] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
[init -> drivers -> nic_drv] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
[init -> nic_router] [uplink] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
[init -> nic_router] [uplink] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
-------------------------------------
[init -> nic_router] [uplink] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
[init -> nic_router] [uplink] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
[init -> nic_router] [uplink] prune re-request timeout of DHCP client
[init -> nic_router] [uplink] dynamic IP config: interface 10.0.0.55/24, gateway 10.0.0.1 P2P 0, DNS server 10.0.0.2
```

Staging with driver in NIC-server mode and a NIC bridge
```
[init -> drivers -> nic_drv] --- freescale ethernet driver ---
[init -> drivers -> nic_drv] netdev_info: registered PHC device 0
[init -> drivers -> nic_drv] dev_info: attached PHY driver [Generic PHY] (mii_bus:phy_addr=fec-1:06, irq=POLL)
[init -> drivers -> nic_drv] netdev_info: Link is Up - 1Gbps/Full - flow control rx/tx
---------------------------------
[init -> ping_1] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
---------------------------------
[init -> ping_1] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
---------------------------------
[init -> ping_1] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
...
---------------------------------
[init -> ping_1] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] snd ETH de:ad:be:ef:00:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP de:ad:be:ef:00:00 > 0.0.0.0 cmd 1
[init -> drivers -> nic_drv] rcv ETH 00:fd:45:fc:6f:80 > de:ad:be:ef:00:00 IPV4 10.0.0.2 > 10.0.0.55 UDP 67 > 68 DHCP de:ad:be:ef:00:00 > 10.0.0.2 cmd 2
[... 56 lines cut ...]
[init -> ping_1] drop packet: DHCP client expects DHCP reply
[init -> drivers -> nic_drv] rcv ETH 00:fd:45:fc:6f:80 > ff:ff:ff:ff:ff:ff IPV4 10.0.0.2 > 255.255.255.255 UDP 67 > 68 DHCP 02:02:02:02:03:00 > 10.0.0.2 cmd 2
[init -> ping_1] rcv ETH 00:fd:45:fc:6f:80 > ff:ff:ff:ff:ff:ff IPV4 10.0.0.2 > 255.255.255.255 UDP 67 > 68 DHCP 02:02:02:02:03:00 > 10.0.0.2 cmd 2
[init -> drivers -> nic_drv] rcv ETH 00:fd:45:fc:6f:80 > ff:ff:ff:ff:ff:ff IPV4 10.0.0.2 > 255.255.255.255 UDP 67 > 68 DHCP 02:02:02:02:03:01 > 10.0.0.2 cmd 2
-------------------------------
[init -> ping_1] snd ETH 02:02:02:02:03:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP 02:02:02:02:03:00 > 0.0.0.0 cmd 1
[init -> ping_1] bad IP protocol
[init -> ping_1] rcv ETH 00:fd:45:fc:6f:80 > ff:ff:ff:ff:ff:ff IPV4 10.0.0.2 > 255.255.255.255 UDP 67 > 68 DHCP 02:02:02:02:03:01 > 10.0.0.2 cmd 2
[init -> ping_1] drop packet: DHCP client expects DHCP targeting the router
[init -> ping_1] rcv ETH 02:02:02:02:03:00 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP 02:02:02:02:03:00 > 0.0.0.0 cmd 1
[... cut ...]
[init -> ping_1] rcv ETH 00:fd:45:fc:6f:80 > ff:ff:ff:ff:ff:ff IPV4 10.0.0.2 > 255.255.255.255 UDP 67 > 68 DHCP 02:02:02:02:03:00 > 10.0.0.2 cmd 2
[init -> ping_1] IP config: interface 10.0.0.166/24, gateway 10.0.0.1, DNS server 10.0.0.2
[init -> ping_1] bad IP protocol
...
[init -> drivers -> nic_drv] rcv ETH 00:fd:45:fc:6f:80 > ff:ff:ff:ff:ff:ff IPV4 10.0.0.2 > 255.255.255.255 UDP 67 > 68 DHCP 02:02:02:02:03:01 > 10.0.0.2 cmd 2
...
[init -> ping_1] rcv ETH 00:fd:45:fc:6f:80 > ff:ff:ff:ff:ff:ff IPV4 10.0.0.2 > 255.255.255.255 UDP 67 > 68 DHCP 02:02:02:02:03:01 > 10.0.0.2 cmd 2
```

m-stein, 2021-02-02:
The problem seems to affect also other tests on imx6q_sabrelite like netperf_lwip.

m-stein, 2021-04-30:
@nfeske It seems that all the recent network-test fails on imx6 are related to this issue. Somehow, the problem is only triggered by base-hw. The driver forwards DHCP responses twice which confuses the router:
```
[init -> nic_router] [uplink] snd ... DHCP DISCOVER de:ad:be:ef:00:00 > 0.0.0.0
[init -> nic_router] [uplink] rcv ... DHCP OFFER de:ad:be:ef:00:00 > 10.0.0.2
[init -> nic_router] [uplink] snd ... DHCP REQUEST de:ad:be:ef:00:00 > 0.0.0.0
[init -> nic_router] [uplink] rcv ... DHCP OFFER de:ad:be:ef:00:00 > 10.0.0.2
[init -> nic_router] [uplink] drop packet (DHCP client expects an acknowledgement)
```

On foc and sel4 the packets are not duplicated:
```
[init -> nic_router] [uplink] snd ... DHCP DISCOVER de:ad:be:ef:00:00 > 0.0.0.0
[init -> nic_router] [uplink] rcv ... DHCP OFFER de:ad:be:ef:00:00 > 10.0.0.2
[init -> nic_router] [uplink] snd ... DHCP REQUEST de:ad:be:ef:00:00 > 0.0.0.0
[init -> nic_router] [uplink] rcv ... DHCP ACK de:ad:be:ef:00:00 > 10.0.0.2
```

m-stein, 2021-05-26:
Should I prioritize the FEC NIC driver in order to potentially fix the nightly tests for the upcoming release?

nfeske, 2021-05-26:
@m-stein Thanks for coming back to this issue. Indeed, I would appreciate your investigation very much.

m-stein, 2021-05-27:
My findings for today:

The strange MAC address (0xdeadbeef0000) is not a problem. The problem also occurs when switching the driver to NIC-server mode but can become non-effective due to a different timing (the double packet receive occurs nonetheless). The problem seems not to be related to the receive path but the transmission.

The bootloader runs a very normal looking DHCP handshake with the server. However, the very first packet that is sent by Genode over the same controller doesn't reach the server. Or better it does, but only together with the second packet that is sent (in my case a second DHCP discover after 10 seconds). So, two identical packets arrive at the server after 10 seconds.

Nonetheless, right after sending it, the first DHCP discover goes through all the layers of the driver down to being copied into the DMA. The packet data copied to the DMA is the same as when the packet was generated in the router. AFAICT, nothing is different from the way the seconds DHCP discover is handled.

This leaves me with the assumption, that the FEC is, at some point, configured differently and always buffers one sent packet until the next. This would also correlate with the spurious-ARP quickfix in #4020.

m-stein, 2021-05-28:
Findings of the day:

* The problem also occurs if the first packet is sent way after the initialization of the controller (so, it's not that the controller would have to have more time before functioning correctly).
* None of the symptoms occur on base-foc although using the same driver and the driver seems to traverse the same code paths as with base-hw when sending.
* The problem can be quickfixed by putting an ``msleep(1);`` before the last register write ``writel(0, txq->bd.reg_desc_active);`` but after the statement ``status |= (BD_ENET_TX_READY | BD_ENET_TX_TC); bdp->cbd_sc = cpu_to_fec16(status);`` in ``fec_enet_txq_submit_skb`` (contrib dde linux).
* However, the problem can't be fixed by putting any or all of these instructions ``mb(); rmb(); wmb(); smp_mb(); smp_rmb(); smp_wmb(); dma_rmb(); dma_wmb();`` before the mentioned register write.
* The ``wmb`` instruction that is often called in ``fec_enet_txq_submit_skb`` seems to be correctly implemented by our armv7 lx emul (asm "dsb st", "memory"). However, there is another variant in linux mainline that does a so called "ARM heavy memory barrier" that conditionally does an outer cache sync and a "SoC memory barrier". I didn't dive deeper into this.
* As ``fec_enet_txq_submit_skb`` writes the packet to DMA before triggering transmission with the mentioned register write, @chelmuth suggested to replace ``case UNCACHED: T::Tex::set(v, 1);`` with ``case UNCACHED: T::Tex::set(v, _device_tex());`` in ``base-hw/src/include/hw/spec/arm/page_table.h``. This fixes the problem but we don't know whether this is due to changed timing or due to DMA being mepped differently.
* Mainline Linux sources change quite some stuff in the FEC NIC driver compared to our contrib version and they also introduce new code that explicitely is for quirks of imx6q. I didn't look too much into it, but the new code can't be simply compiled without changing our glue code.

nfeske, 2021-05-28:
Thank you for interesting summary.

m-stein, 2021-05-31:
Findings of the day:
* The problem __can__ be fixed by reading the ``bdp->cbd_sc`` (mentioned in my last post) at any point after it was written and before the write to ``txq->bd.reg_desc_active`` like this:
   ```
   static unsigned long volatile tmp;
   tmp = *(unsigned long volatile *)&bdp->cbd_sc;
   ```
   So, it seems to be related to out-of-order execution of the write. As this is what the ``wmb()`` call in the code is for, @skalk suggested that our ``wmb()`` implementation might be ineffective.
* However, none of the flavors ``sy|st|ish|ishst|nsh|nshst|osh|oshst`` of the ARM instructions ``dsb``, ``dmb``, or ``isb`` alone or all in combination fix the problem.
* declaring ``bdp`` to reference a volatile type (``struct bufdesc volatile *bdp;``) doesn't fix the problem.

I was told that an update of the FEC NIC port to newest Linux mainline with the new DDE approach is preferable. Therefore, I would opt for quick-fixing the problem inside the current contrib code with the above mentioned artificial read of ``bdp->cbd_sc`` until we find the time to update the port.

Commit 9b17b170dda744eef6096a0e9cad3cdaa71e74c6 is a suggestion for a quick fix.

skalk, 2021-06-04:
It is strange indeed. I suspect our DDE Linux emulation layer does not contain all memory barriers resp. synchronization points necessary. In fact, I could see that `readl` or `writel` is implemented quite simple in our emulation environment, whereby Linux does more stuff including `wmb()` and `rmb()`. When only changing those two functions, the issue still remained. But I can imagine there are other simplifications in our emulation layer that lead to a less strong memory ordering in the driver code. To me it seems reasonable to take the somehow strange quick fix for the time-being until the driver gets updated to use the new DDE Linux approach, which keeps more original kernel semantic. @chelmuth do you agree?

chelmuth, 2021-06-04:
Are you referring to 9b17b170dda744eef6096a0e9cad3cdaa71e74c6? I have no objections.

skalk, 2021-06-15:
@chelmuth Yes, indeed.

m-stein, 2021-06-28:
Commit ff128df1319e651d6339746119bfa14e6f5c77fa breaks the fec nic on sel4. More specificly speaking the read of the bdp member that it introduces somehow causes that packets are not sent by hardware anymore. I see that the nic router keeps sending dhcp discover packets and nothing happens. Changing the point where the read is issued or how the read is issued doesn't help. Only when I remove the read op, the dhcp handshake finishes without problems.  It seems that foc imx6q_sabrelite isn't negatively affected by the fix. I cannot explain why the fix affects sel4 but I would currently refrain from going deeper into the topic. We may revert ff128df1319e651d6339746119bfa14e6f5c77fa or leave it.

commit ce81d6ca (2021-06-16): dde_linux/fec: fix unsynchronized TX DMA write

Unfortunately, our current implementation of 'wmb()' doesn't seem to do what we
want it to do. On base-hw + imx6q_sabrelite, the write of bdp->cbd_sc seems to
get re-ordered after the write to txq->bd.reg_desc_active in the transmission
path of the contrib code. Due to this, the transmission of the packet is only
triggered the next time a packet is sent. However, we only quick-fix it by
enforcing the execution of the write with a volatile global read as we will
soon update the FEC NIC port with a new DDE approach anyway.

Fixes #4010
commit ff128df1 (2021-06-25): dde_linux/fec: fix unsynchronized TX DMA write

Unfortunately, our current implementation of 'wmb()' doesn't seem to do what we
want it to do. On base-hw + imx6q_sabrelite, the write of bdp->cbd_sc seems to
get re-ordered after the write to txq->bd.reg_desc_active in the transmission
path of the contrib code. Due to this, the transmission of the packet is only
triggered the next time a packet is sent. However, we only quick-fix it by
enforcing the execution of the write with a volatile global read as we will
soon update the FEC NIC port with a new DDE approach anyway.

Fixes #4010
closed 2022-05-31 by m-stein