genodelabs/genode #4012: riscv: Update to privileged spec v1.10 and replace Spike emulator

ssumpf, 2021-02-09:
* Update base-hw RISC-V support to privileged ISA specification v1.10
* Replace https://github.com/ssumpf/riscv-isa-sim and Berkeley Boot Loader (https://github.com/skalk/bbl-lite) with the Qemu RISC-V virt platform

ssumpf, 2021-02-10:
@chelmuth: 8bfde36 is the first proposed version of my RISC-V update, could you take a look?

ssumpf, 2021-02-10:
@chelmuth: 85e2412 fixes the ld segmentation fault (trace policies) and floating point link errors (e.g, nitpicker) we have experienced for a long time.

chelmuth, 2021-02-10:
Thanks for this update. Besides the usual nitpicking (see my comments to your patch) the switch to a recent ISA spec and Qemu is a great feature. What confused me a bit is the intermixed use of *riscv_qemu* and *risc_virt* for the Qemu board. If both are synonymous, could *riscv_virt* be replaced by *riscv_qemu* throughout the commit?

ssumpf, 2021-02-11:
@chelmuth, @m-stein: c8a5421 incorporates your suggestions.

chelmuth, 2021-02-11:
I'm fine with the changes, so, now it's time to update the CI tools (tester, builder, limited package build?).

ssumpf, 2021-02-11:
> I'm fine with the changes, so, now it's time to update the CI tools (tester, builder, limited package build?).

@chelmuth: I would say so.

ssumpf, 2021-02-11:
@nfeske: 73c4c43 uses `.rept` for the seven `nop` instructions.

skalk, 2021-02-11:
@ssumpf I've added my comments in your previous commit. Sorry I did not encountered the fixups you already did. So please ignore any duplicate comments.

chelmuth, 2021-02-11:
@ssumpf please also consider @skalk's comments abouts board vs. spec in the https://github.com/ssumpf/genode-riscv repository.

ssumpf, 2021-02-11:
@skalk: e0433e1 implements your suggestions

ssumpf, 2021-02-11:
> @ssumpf please also consider @skalk's comments abouts board vs. spec in the https://github.com/ssumpf/genode-riscv repository.

@chelmuth: @skalk changes have also been applied to https://github.com/ssumpf/genode-riscv.

chelmuth, 2021-02-11:
Merged, tested, fixup'd and pushed the current state to genode staging. @ssumpf there's also a fixup for build error of genode-riscv at https://github.com/chelmuth/genode-riscv/commit/325b59a6efb6d6a84eb41fafad8c4ed6dc3a162b

ssumpf, 2021-02-12:
> Merged, tested, fixup'd and pushed the current state to genode staging. @ssumpf there's also a fixup for build error of genode-riscv at [chelmuth/genode-riscv@325b59a](https://github.com/chelmuth/genode-riscv/commit/325b59a6efb6d6a84eb41fafad8c4ed6dc3a162b)

@chelmuth: Thanks, I have updated "genode-riscv" accordingly.

nfeske, 2021-02-26:
Fixed in Genode 21.02.

closed 2021-02-26 by nfeske