genodelabs/genode #4013: core: Error: freeing non-empty slab block

alex-ab, 2021-02-09:


chelmuth, 2021-02-10:
I wonder which exception is the reason behind this. The allocator implementation itself avoids exceptions.

nfeske, 2021-02-10:
I guess, the `Allocator_avl_base::_alloc_block_metadata` method calls `_md_alloc->alloc(sizeof(Block), &b))`, which can trigger an `Out_of_ram` or `Out_of_caps` exception.

@alex-ab to give a definite answer to @chelmuth's question, could you try replacing the catch-all by catching these both exceptions individually instead and print a message for each? Just as a test.

alex-ab, 2021-02-10:
With debug commit e54f65c6c29605bdff2810cf3f1f43764fc02e05 I see around ~20 Out of caps messages during running ```make KERNEL=hw core``` and no other exception/message.

nfeske, 2021-02-10:
That makes sense. Thanks @alex-ab for clearing up the question, and of course for fixing the bug! ;-)

commit 984c5155 (2021-02-10): allocator_avl: avoid memory leakage

Fixes #4013
closed 2021-02-23 by nfeske