genodelabs/genode #4014: allocator avl: warnings about dangling allocations

alex-ab, 2021-02-10:
The warning in the allocator avl during destruction is meant to hint for potential trouble ahead. In the course of running the tool chain on Genode more or less on every fork/wait such messages show up, which spill up the log.

After investigation it turned out, that the warnings in most of the cases are not critical. Not critical because the dangling pointers are actually part of the metadata allocator of the avl implementation and get destructed shortly after anyway. 

It seems, that @m-stein already investigated and addressed it with 71512cd043a97338e47e6275d01cddcb2c0a93c0 in one way to avoid, more or less, non critical false warnings.

In general, we should remove any uncertainty about false or true warnings, to make this tool usable.

alex-ab, 2021-02-10:
0c05fc8517ec4d1b4354c6ed8a3c59c789b89823 and  cb9ddc9d1edfd5f0da62c889ef475da6f12344fe are proof of concept commits, to show that by reordering the destruction a bit, the false warnings go away (for my use-case to 0 warnings). Admittedly, they introduce overhead in form of extra code, (public) symbols and compute time. I can't say I'm very happy about it.

I would be glad about different/better opinions about the topic.

alex-ab, 2021-02-11:
> @alex-ab Since I am fine with the two commits, should I go forward with merging them to staging as is, or do you wish to refine them in some way? I just ask because you spoke of them being a mere proof of concept.

I was not sure whether the naming of the methods are good enough. Functionality wise the commits are working. So, from my side you can add to staging.

nfeske, 2021-02-11:
> I was not sure whether the naming of the methods are good enough.

No complaints. :-)

I merged both commits to staging.

nfeske, 2021-02-23:
The fixes has arrived in master. Can this issue be closed?

alex-ab, 2021-02-23:
Yes.

closed 2021-02-23 by alex-ab