genodelabs/genode #4015: libc: setup environ of forked process

alex-ab, 2021-02-11:


alex-ab, 2021-02-11:
6ad853652ef2c18dd7322e025b9072533ffc97d5 reads in the environ for a clone as done in posix/construct.cc and qt5_component.cc, which helps that collect2 finds the linker.

nfeske, 2021-02-12:
Thanks for reporting.

Your proposed fix is not a good solution though. First, it merely masks the actual problem, which is that `environ` loses its value during fork despite of being part of the application state. Second, by conveniently reusing `populate_args_and_env`, the memory allocated for `argv` leaks. Granted, the leak is limited to one allocation at start time but it looks sloppy.

Commit 0198034c2ab4be63c69bd65a25ddc334fc9f9999 contains the fix the underlying problem along with a test.

commit 0198034c (2021-02-12): libc: clone 'environ' pointer value after fork

The 'environ' pointer is a global variable that must be considered as
part of the application state. It must be copied from the parent to the
child process during fork. Otherwise, a child returning from fork is
unable to access environment variables before invoking execve. The
actual environment variables and their values are already captured
because they reside at the application heap, which is cloned from the
parent. So the copied 'environ' pointer refers to valid data.

Fixes #4015
closed 2021-02-23 by nfeske