genodelabs/genode #4016: platform_drv + hw + usb_net: irq session creation failed

m-stein, 2021-02-12:
When running ``make run/usb_net KERNEL=hw BOARD=pc`` (or another usb_net scenario on hw) on x86_64, the platform_drv fails to create an IRQ session for IRQ number 0:
```
kernel initialized
ROM modules:
 ROM: [0000000000877000,000000000089f2f8) acpi_drv
 ROM: [0000000000610000,0000000000611245) config
 ROM: [000000000000c000,000000000000d000) core_log
 ROM: [00000000008b0000,0000000000911320) init
 ROM: [00000000009c6000,0000000000abdf60) ld.lib.so
 ROM: [000000000069a000,00000000008769d8) libc.lib.so
 ROM: [0000000000410000,00000000004d20c8) nic_router
 ROM: [00000000003b2000,000000000040fec8) platform_drv
 ROM: [000000000000a000,000000000000b000) platform_info
 ROM: [0000000000525000,0000000000545968) report_rom
 ROM: [00000000008a0000,00000000008af928) test-lwip_httpsrv
 ROM: [0000000000912000,00000000009279d8) timer
 ROM: [0000000000612000,0000000000699f30) usb_net_drv
 ROM: [0000000000928000,00000000009c52d8) vfs.lib.so
 ROM: [00000000004d3000,0000000000524338) vfs_lwip.lib.so
 ROM: [0000000000546000,000000000060f2a0) x86_pc_usb_host_drv

Genode 20.11-179-geaec145f72
16240 MiB RAM and 64533 caps assigned to init
[init -> usb_net_drv] --- USB net driver ---
[init -> nic_router] [uplink] dynamic IP config: none
[init -> nic_router] [uplink] NIC sessions: 0
[init -> usb_net_drv] Using configured mac: 02:00:00:00:01:01
[init -> nic_router] [downlink] static IP config: interface 10.0.3.1/24, gateway 0.0.0.0 P2P 0
[init -> nic_router] [downlink] NIC sessions: 0
[init -> acpi_drv] Found MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
[init -> acpi_drv] Found MCFG
[init -> acpi_drv] MCFG BASE 0xf8000000 seg 0x0 bus 0x0-0x3f
[init -> acpi_drv] Found DMAR
[init -> acpi_drv] 39 bit DMA physical addressable , IRQ remapping supported
[init -> acpi_drv] DMA remapping structure type=0
[init -> acpi_drv] DMA remapping structure type=0
[init -> acpi_drv] DMA remapping structure type=1
[init -> acpi_drv] DMA remapping structure type=1
[init -> acpi_drv] XSDT OEM 'ALASKA', table id 'A M I', revision 17244169, creator 'AMI ' (65555)
[init -> platform_drv] ECAM/MMCONF range 00000000:00000000.0-0000003f:0000001f.7 - addr [00000000f8000000,00000000fc000000)
[init -> platform_drv] Root bridge: 00000000:00000000.0
[init -> acpi_drv] SMBIOS table (entry point: 0x1504c0 structures: 0xec390)
[init -> platform_drv] 00000000:00000001.0 - bridge 01:00.0
[init -> platform_drv] 00000000:0000001c.0 - bridge 02:00.0
[init -> platform_drv] usb_drv -> : assignment of PCI device 00:14.0 succeeded
[init -> usb_drv] ioremap: mapped phys 0xf7e20000 (size 65536) to virt 0x20000
[init -> usb_drv] ioremap: return sub range phys 0xf7e20000 (size 65536) to virt 0x20000
[init -> usb_drv] dev_info: xHCI Host Controller
[init -> nic_router] [downlink] NIC sessions: 1
[init -> usb_drv] dev_info: new USB bus registered, assigned bus number 1
[init -> test-lwip_httpsrv] lwIP Nic interface down
[init -> usb_drv] dev_info: hcc params 0x200077c1 hci version 0x100 quirks 0x00009810
[init -> platform_drv] 00000000:00000014.0 adjust IRQ as reported by ACPI: 10 -> 21
[init -> platform_drv] Error: IRQ-session creation failed (ram_quota=6144, cap_quota=3, irq_number=0, irq_trigger=0, irq_polarity=0, device_config_phys=0xf80a0000)
[init -> platform_drv] 00000000:00000014.0 uses IRQ, vector 0x15, supports: MSI
[init -> usb_drv] dev_info: USB hub found
[init -> usb_drv] dev_info: 15 ports detected
[init -> usb_drv] dev_info: xHCI Host Controller
[init -> usb_drv] dev_info: new USB bus registered, assigned bus number 2
[init -> usb_drv] dev_info: USB hub found
[init -> usb_drv] dev_info: 6 ports detected
[init -> platform_drv] usb_drv -> : assignment of PCI device 00:1d.0 succeeded
[init -> usb_drv] ioremap: mapped phys 0xf7e37000 (size 1024) to virt 0x13000
[init -> usb_drv] ioremap: return sub range phys 0xf7e37000 (size 1024) to virt 0x13000
[init -> usb_drv] dev_info: EHCI Host Controller
[init -> usb_drv] dev_info: new USB bus registered, assigned bus number 3
[init -> test-lwip_httpsrv] Create new socket ...
[init -> usb_drv] dev_warn: Enabling legacy PCI PM
[init -> platform_drv] 00000000:0000001d.0 adjust IRQ as reported by ACPI: 6 -> 23
[init -> platform_drv] 00000000:0000001d.0 uses IRQ, vector 0x17
[init -> usb_drv] dev_info: irq 130, io mem 0xf7e37000
[init -> test-lwip_httpsrv] Warning: read blocked until lwIP interface is ready
[init -> test-lwip_httpsrv] Warning: read blocked until lwIP interface is ready
[init -> usb_drv] dev_info: USB 2.0 started, EHCI 1.00
[init -> usb_drv] dev_info: USB hub found
[init -> usb_drv] dev_info: 3 ports detected
[init -> usb_drv] dev_info: new full-speed USB device number 2 using xhci_hcd
[init -> usb_drv] dev_info: new full-speed USB device number 3 using xhci_hcd
[init -> usb_drv] dev_info: new high-speed USB device number 4 using xhci_hcd
[init -> test-lwip_httpsrv] lwIP Nic interface up address=10.0.3.2 netmask=255.255.255.0 gateway=10.0.3.1
[init -> test-lwip_httpsrv] Now, I will bind ...
[init -> test-lwip_httpsrv] Now, I will listen ...
[init -> test-lwip_httpsrv] Start the server loop ...
[init -> usb_drv] dev_info: new high-speed USB device number 5 using xhci_hcd
[init -> usb_drv] dev_info: USB hub found
[init -> usb_drv] dev_info: 4 ports detected
[init -> usb_drv] dev_info: new high-speed USB device number 2 using ehci-pci
[init -> usb_drv] dev_info: USB hub found
[init -> usb_drv] dev_info: 8 ports detected
[init -> usb_drv] dev_info: new full-speed USB device number 6 using xhci_hcd
[init -> usb_drv] dev_info: USB hub found
[init -> usb_drv] dev_info: 4 ports detected
[init -> usb_drv] dev_info: new full-speed USB device number 7 using xhci_hcd
[init -> usb_net_drv] netif_info: open: enable queueing (rx 4, tx 4) mtu 1500 ASIX framing
[init -> usb_net_drv] netif_info: register '
[init -> usb_net_drv] ' at usb-usbbus-, ASIX AX88179 USB 3.0 Gigabit Ethernet, 00:24:9b:04:59:c8
[init -> usb_net_drv] MAC address 00:24:9b:04:59:c8
[init -> usb_net_drv] netdev_info: ax88179 - Link status is: 1
[init -> nic_router] [uplink] NIC sessions: 1
[init -> nic_router] [uplink] dynamic IP config: interface 10.0.0.78/24, gateway 10.0.0.1 P2P 0, DNS server 10.0.0.2
```
This does not seem to impede the usb_net_drv from working as the tests succeed anyway. The same setup only with nova doesn't show this error while the platform driver still seems to request the same IRQs.

alex-ab, 2021-02-12:
The platform driver tries to allocate first an MSI, which base-hw does not support and denies. Afterwards it tries to allocate an normal IRQ, which succeeds. base-nova supports MSI, because of that the first attempt succeeds and no denied session creation is shown.

chelmuth, 2021-02-12:
The diagnostic error message is just an artifact of the platform_drv probing for an MSI IRQ session.

m-stein, 2021-02-12:
@chelmuth @alex-ab Thanks for the quick feedback and explanation on this! What do you think? To me, errors that do not express something that should be fixed either in the code or the integration are frustrating. What about detecting that the error is only diagnostic and printing at at least as mere warning?

alex-ab, 2021-02-12:
The error message steams from the generic base/connection.h class, and it is considered normally as fatal. So, it is not just about exchanging an error() with warning() message.

chelmuth, 2021-02-12:
The probing could be replaced by retrieving the kernel information from the platform_info ROM and requesting MSI on supported kernels only.

m-stein, 2021-02-15:
We have a similar issue with the platform_info ROM when running, for instance, netperf_lxip_bridge on pistachio:
```
[init -> nic_drv -> acpi_drv] [31mError: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="platform_info")[0m[0m
[init -> nic_drv -> acpi_drv] [31mError: Could not open ROM session for "platform_info"[0m[0m
[init -> nic_drv -> nic_drv] --- iPXE NIC driver started ---[0m[0m
[init -> nic_router] [uplink] dynamic IP config: none[0m[0m
[init -> nic_router] [uplink] NIC sessions: 0[0m[0m
[init -> nic_drv -> nic_drv] -- init iPXE NIC[0m[0m
[init -> nic_router] [server] static IP config: interface 10.0.3.1/24, gateway 0.0.0.0 P2P 0[0m[0m
[init -> nic_router] [server] NIC sessions: 0[0m[0m
[init -> nic_drv -> platform_drv] [31mError: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="platform_info")[0m[0m
[init -> nic_drv -> platform_drv] [31mError: Could not open ROM session for "platform_info"[0m[0m
[init] child "nic_router" announces service "Nic"[0m[0m
[init] child "nic_router" announces service "Uplink"[0m[0m
[init -> nic_drv -> acpi_drv] Found MCFG[0m[0m
[init -> nic_drv -> acpi_drv] MCFG BASE 0xe0000000 seg 0x0 bus 0x0-0x10[0m[0m
[init -> nic_drv -> acpi_drv] Found MADT[0m[0m
[init -> nic_router] [31mError: Session_env: less CAP quota consumed than expected[0m[0m
[init -> nic_drv -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 5[0m[0m
[init -> nic_drv -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13[0m[0m
[34mWarning: invalid signal-context capability[0m
[init -> nic_router] [server] NIC sessions: 1[0m[0m
[init -> nic_drv -> acpi_drv] RSDT OEM 'LENOVO', table id 'TC-2R   ', revision 100925648, creator ' LTP' (0)[0m[0m
[init -> nic_bridge] --- NIC bridge started (mac=02:02:02:02:02:01) ---[0m[0m
[init] child "nic_bridge" announces service "Nic"[0m[0m
[init -> nic_drv -> acpi_drv] [31mError: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="platform_info")[0m[0m
[init -> nic_drv -> acpi_drv] [31mError: Could not open ROM session for "platform_info"[0m[0m
[init -> nic_drv -> acpi_drv] SMBIOS table (entry point: 0x145e30 structures: 0xef540)[0m[0m
[init -> nic_drv -> platform_drv] ECAM/MMCONF range 00000000:00000000.0-00000010:0000001f.7 - addr [e0000000,e1100000)[0m[0m
[init -> nic_drv -> platform_drv] [34mWarning: MADT IRQ 9-> GSI 9 flags 13 ignored[0m[0m
[init -> nic_drv -> platform_drv] [34mWarning: MADT IRQ 0-> GSI 2 flags 5 ignored[0m[0m
[init -> nic_drv -> platform_drv] Root bridge: 00000000:00000000.0[0m[0m
[init -> nic_drv -> platform_drv] 00000000:0000001c.0 - bridge 05:00.0[0m[0m
[init -> nic_drv -> platform_drv] 00000000:0000001e.0 - bridge 11:00.0[0m[0m
[init -> nic_drv -> platform_drv] [31mError: ECAM/MMCONF MMIO access out of bounds - ACPI table information is wrong![0m[0m
[init -> nic_drv -> platform_drv] [31mError:  adjust size from 0x1100000->0x10000000[0m[0m
```
Despite these errors, the test finished successful in the past. I had an offline discussion with @alex-ab. We could provide an empty platform_info on all platforms that currently don't provide such a ROM file in order to fix the errors. Or this dummy platform_info could contain only the kernel name in order to fix also the original issue.

What do you think?

m-stein, 2021-02-15:
When running netperf_lxip_bridge on pistachio myself (macho) I also get another error that I can't explain right away:
```
[init -> nic_drv -> acpi_drv] Error: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="platform_info")
[init -> nic_drv -> platform_drv] Error: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="platform_info")
[init -> nic_drv -> acpi_drv] Error: Could not open ROM session for "platform_info"
[init -> nic_drv -> platform_drv] Error: Could not open ROM session for "platform_info"
Error: I/O memory [000e0000,00100000) not available
Error: Local MMIO mapping failed!
[init -> nic_drv -> acpi_drv] Error: IO_MEM-session creation failed (cap_quota=3, ram_quota=6144, base=0xe0000, size=0x20000, wc=no)
```
I have ``--autopilot`` set. It occurs on staging, master and 20.11. The test is stuck after that.

chelmuth, 2021-02-15:
> We could provide an empty platform_info on all platforms that currently don't provide such a ROM file in order to fix the errors. Or this dummy platform_info could contain only the kernel name in order to fix also the original issue.
> 
> What do you think?

I like the approach to provide known info on all base platforms, but would not call it *dummy* ROM.

chelmuth, 2021-02-15:
> Error: I/O memory [000e0000,00100000) not available
> Error: Local MMIO mapping failed!
> [init -> nic_drv -> acpi_drv] Error: IO_MEM-session creation failed (cap_quota=3, ram_quota=6144, base=0xe0000, size=0x20000, wc=no)
> ```

I have no idea about the reason for this error. Isn't this the BIOS area? Does *acpi_drv* explicitly request this area?

alex-ab, 2021-05-25:
I added to my issue_4016 branch two cosmetic commits to avoid the misleading red messages.

commit a6c31c3b (2021-05-26): netperf_*_usb: fix cap quota for sel4+hw x86

Fixes #4016