genodelabs/genode #4019: base-hw: streamline packaging

nfeske, 2021-02-16:
The current depot recipes for the base-hw kernel leave room for improvements:

- Most src archives contain a large superset of the source codes required for the respective boards. E.g., x86-specific code is packaged into the archives for ARM boards. This is not strictly a problem, but it is not clean either.
- Hosting board support in an external repository is not easy. E.g. genode-world hosts this rather convoluted include snippet: https://github.com/genodelabs/genode-world/blob/master/recipes/src/base-hw_content.inc It would be nice to provide infrastructure that alleviates the need for such crutches.

nfeske, 2021-02-16:
Commit https://github.com/genodelabs/genode/commit/002032e99a7f20de0179dafece61ddfd995c6fb6 improves the mechanism. I tested the build for all packages provided by the base-hw repository and my (not yet published) allwinner repository. Genode-world will need adjustments.

@skalk Do you agree with this change?

I merged it to staging already just to see how all the boards fare.

skalk, 2021-02-16:
@nfeske: I'm fine with the changes, and appreciate the cleanup work!

nfeske, 2021-02-16:
Thanks for the prompt feedback!

ssumpf, 2021-02-17:
@nfeske: 6494235 adjusts the path for the startup library + uses `select_from_repositories` for crt0.s of the same library.

nfeske, 2021-02-17:
Thank you. I merged your select_from_repositories change to the separate commit https://github.com/genodelabs/genode/commit/635985f542cdb2b4273d2956ac99650866da2766

commit 002032e9 (2021-02-16): base-hw: streamline extraction of depot archives

This patch simplifies the packaging of the base-hw kernel for a given
board, in particular when the board support is hosted as an external
repository such as genode-world. Regardless of where the board-support
is hosted, the content.mk file becomes as simple as:

  include $(GENODE_DIR)/repos/base-hw/recipes/src/base-hw_content.inc

The board name is automatically inferred from the path of the src
recipe. The architecture is determined from board/<name>/arch files,
following the pattern of the image_link_address. The attempt to build a
base-hw- [... cut]
commit db7c042e (2021-02-19): base-hw: streamline extraction of depot archives

This patch simplifies the packaging of the base-hw kernel for a given
board, in particular when the board support is hosted as an external
repository such as genode-world. Regardless of where the board-support
is hosted, the content.mk file becomes as simple as:

  include $(GENODE_DIR)/repos/base-hw/recipes/src/base-hw_content.inc

The board name is automatically inferred from the path of the src
recipe. The architecture is determined from board/<name>/arch files,
following the pattern of the image_link_address. The attempt to build a
base-hw- [... cut]
closed 2021-02-23 by nfeske