genodelabs/genode #4024: libc: pthread TLS optimizations

cproc, 2021-02-19:
Improve pthread TLS storage access time by:
- determining the current pthread object with a libc-local stack pointer calculation
- storing TLS data in the pthread object instead of an array of linked lists

cproc, 2021-02-19:
Commit ee78c36 implements these ideas. It also fixes the current problem that TLS data is not deleted when a pthread gets destroyed. Also, TLS key destructor functions given to `pthread_key_create()` are now called on thread destruction. The base library had to be modified to support the stack-based pthread object lookup.

A simple measurement loop (in temporary commit 01cf130) of repeated calls of `pthread_setspecific()` and `pthread_getspecific()` showed the following results on base-linux (time stamp difference between 'start' and 'finish'):

- without optimizations:

```
2.356912 [init -> test -> test-pthread] start
7.490337 [init -> test -> test-pthread] finish
7.514385 [init -> test -> test-pthread] start
12.508088 [init -> test -> test-pthread] finish
```

- with optimizations:

```
2.358089 [init -> test -> test-pthread] start
3.408050 [init -> test -> test-pthread] finish
3.431449 [init -> test -> test-pthread] start
4.487403 [init -> test -> test-pthread] finish
```

nfeske, 2021-02-19:
That looks great! I also like the patch and the fact that you measured the effect.

One detail may be considered as an improvement though: You are introducing a bunch of class variables. Have you considered the alternative to wrap this state into a dedicated object that is constructed at a central place and passed as reference to the everyone who needs this state? In general, I try to avoid global variables (such as class variables) as much as possible.

cproc, 2021-02-19:
@nfeske: I'm not exactly sure about the central place to construct such an object, but would the solution in commit 3518e8e fit your idea? It constructs an 'Tls_info' object as static function variable. There's a slight performance decrease after this change:

```
2.393972 [init -> test -> test-pthread] start
3.706415 [init -> test -> test-pthread] finish
3.719721 [init -> test -> test-pthread] start
5.039049 [init -> test -> test-pthread] finish
```

nfeske, 2021-02-19:
@cproc Thank you for your quick response. Admittedly, I find the new version not easier to read than your original one. So maybe, I should put my purism regarding global variables aside here. ;-)

nfeske, 2021-02-19:
I merged your original commit along with the small whitespace tweak. I'm very curious about the tests. Another fixup with the comment of `stack_address` would be appreciated.

commit 06486e52 (2021-02-19): fixup "libc: pthread TLS optimizations" (whitespace)

Fixes #4024
commit c3f966ba (2021-02-19): libc: pthread TLS optimizations

Fixes #4024
closed 2021-02-23 by nfeske