genodelabs/genode #4025: vmm: add virtio block device model

skalk, 2021-02-19:
To use the ARM vmm in the field a block device model is still missing. Providing a Virtio block device would be the natural fit.

skalk, 2021-02-19:
@ssumpf you may review commit cdc860b because it restructures the Virtio base device model you've written initially.

skalk, 2021-02-19:
@chelmuth, @nfeske : apart from @ssumpf review and of course you own one, in my eye you can put it to staging to fasten the test cycle before the release. Of course, according to your own schedule and as occasion demands.

nfeske, 2021-02-19:
I merged your beautiful commits to staging.

Regarding the review, I'd like to rely on @ssumpf since I'm unfamiliar with VirtIO.

ssumpf, 2021-02-19:
@skalk: I am fine the changes, the virtio device looks way better now ;)

commit 8866336a (2021-02-19): vmm: add virtio block device model

* Add new virtio device model
* Extend test run-script with vfat block test image
* Add vmm depot src recipe
* Use packages in test run-script

Fix #4025
closed 2021-02-23 by nfeske