genodelabs/genode #4027: build system: Improve link times for core

ssumpf, 2021-02-19:
Especially on base-hw RISC-V we experience long link times for core and it's debugging version. After some investigation we found that linking the base and core libraries into a large relocatable object file which in turn was linked to core was causing the long link times. A better solution seems to be to create another library archive from base and core libraries which is later linked to core.

ssumpf, 2021-02-19:
@nfeske: 9bb6207 contains a first version of the link time improvement for core.

nfeske, 2021-02-19:
Thank you for investigating. This looks very promising!

nfeske, 2021-02-19:
Upon closer inspection, @chelmuth and me think that the thin archives won't work as they - like you suspected - miss the actual data.

A reasonable way seems to assemble a new archive from the object files gathered via `ar -t` for the various .a files of the link items. I think you mentioned this idea today. It seems legit.

chelmuth, 2021-02-19:
Another option I discovered just now [1] is to use MRI scripts [2] like follows.

```bash
> cat foo.mri
create sum.a
addlib 1.a
addlib 2.a
save
end
> ar -M < foo.mri
> ar t sum.a 
vm_session.o
trace.o
thread_myself.o
thread_bootstrap.o
...
```

[1] https://stackoverflow.com/a/23621751/453786
[2] https://sourceware.org/binutils/docs/binutils/ar-scripts.html

ssumpf, 2021-02-22:
@chelmuth: Your suggestion does the job! Thanks. I did a little research on MRI. There is nothing much to find but "Magnetic resonance imaging". It seems to have been a tool chain that was there before the GNU tool chain (https://sourceware.org/binutils/docs/binutils/ar-scripts.html) or as @cnuke mentioned today: "We might have to ask the gray beards".

ssumpf, 2021-02-22:
@nfeske: I have revised the last commit using @chelmuth suggestion with 706b7ff.  It has been tested on all platforms and all kernels by hand with packages (`run/demo`) and simple `run/log`.

nfeske, 2021-02-23:
Very cool! I merged your commit to staging.

nfeske, 2021-02-25:
Fixed in master.

ssumpf, 2021-03-08:
@nfeske: f68e655 introduced a regression where the debug versions of core and bootstrap are not build any more. 530d76e addresses this issue. Please consider for staging.

commit 2fefda02 (2021-03-08): run boot_dir hw: build debug ELF binaries again

- build debug version of core and bootstrap found under
  _var/run/<runscript>.core_ (.bootstrap). This is a regression when
  switching from object to archive library linkage of core.

fixes #4027
closed 2021-03-12 by nfeske