genodelabs/genode #4028: prepare_port: simple cache for downloaded files

tomga, 2021-02-19:
When working with ports it is not uncommon (especially with `dde_linux`) that port hash is changed due to some minor change (e.g. adding a patch), `prepare_port` has to be called and it downloads same files that were downloaded previously when preparing previous version of port. In case of `dde_linux` about 200MB of data is downloaded and in case of my internet connection (no cable connection) it takes quite some time. Given that those files are already downloaded it is a waste of time and bandwidth.

tomga, 2021-02-19:
e2f5b896bd668091a5b6dae92590b620e22567ab implements simple cache for downloaded by `prepare_port` files.

nfeske, 2021-02-22:
That is a very nice suggestion! Especially your remark about dde_linux is spot on. The patch also looks very good to me. I especially like the use of the hash as file name.

I'd like to merge the patch as soon the dust of the current release (scheduled for Thursday) has settled, presumably mid of next week.

tomga, 2021-02-22:
@nfeske I did not expect that patch will be accepted without further discussion :-) I remember you did not like the idea when I asked for this feature during Hack'n'Hike almost two years ago. I remember you raised concerns that it will not work for git repositories but maybe it was more because I wanted something but did not implement it myself. Nevertheless it's me who is not completely happy with current patch :-)

I'd like to be able to pass a cache location with environment variable to be able to share cache for multiple working directories. Probably part that exports `CONTRIB_CACHE` value will have to be changed. I'll check and provide an updated version.

Also as I expected that issue about handling of git repositories will come back I also looked at the way they are processed and I think that this idea could also be extended to them. Would you be interested in it too?

nfeske, 2021-02-22:
> I remember you did not like the idea when I asked for this feature during Hack'n'Hike almost two years ago.

It is always good to have a look at a concrete patch to form an opinion. :-)

I also have a few remaining doubts, particular about the handling of corrupted downloads. Such corner cases usually take a bit of practical use to cover them.

> Also as I expected that issue about handling of git repositories will come back I also looked at the way they are processed and I think that this idea could also be extended to them. Would you be interested in it too?

For me, the most interesting use case is the one you mentioned (dde_linux) as well as virtualbox (takes ages to download, changes of the port file happen much more frequently than the switch to a different archive). I agree that git could in principle be covered using the commit hash as part of the name but I don't see the use case as strong as for the archives. In my opinion, it is not really worth addressing right now.

tomga, 2021-02-24:
> I also have a few remaining doubts, particular about the handling of corrupted downloads.

I tried to keep in mind possible corruption scenarios. I ensured that checksum is verified after retrieval from cache (for case of corrupted cache) and to put something into cache only after checksum verification (I assumed that make breaks processing on first failed command in case there is a list of commands).

Currently I see only one problem - if cache is corrupted for some reason the situation will not heal itself without manual intervention like removal of cache. I saw this case earlier too but I thought it will make patch more complicated to handle this case.

In [branch issue_4028](/tomga/genode/commits/issue_4028) I added 4a6baeb17cb0b183c89f76ac52948d40c0885109 that change to things:

 * fix for one scenario that could lead to corrupted file in cache; it was possible by filling filesystem totally during making a copy into cache. I made a change to copy to temporary file first and later rename to final file.
 * allowed to provide contrib cache location to be able to share it for more than one repository and as this name started to be externally visible in environment I changed name from `CONTRIB_CACHE` to `GENODE_CONTRIB_CACHE`.

nfeske, 2021-02-26:
All what you write sounds fantastic! Your thoughtfulness, e.g., the naming of the name of the environment variable and playing through scenarios for file corruption, is very much appreciated.

> Currently I see only one problem - if cache is corrupted for some reason the situation will not heal itself without manual intervention like removal of cache.

I think that the internal integrity of the cache should be safe to assume. This is not different from the content of the existing contrib sub directories, or the content of the depot.

I have no reservations against merging your patch. I nevertheless want to first give it a thorough manual try before merging it to staging.

nfeske, 2021-03-03:
I added your commits to staging and supplemented the fixup https://github.com/genodelabs/genode/commit/b85359a3386ff52d708ee209ca74e9526efe0e3a.

Thanks @tomga for this cool feature and the lean implementation!

commit efdc6155 (2021-03-03): prepare_port: simple cache for downloads

Fixes #4028
commit de625829 (2021-03-12): prepare_port: simple cache for downloads

Fixes #4028
closed 2021-03-12 by nfeske