genodelabs/genode #4029: init: support to route PD service to client

alex-ab, 2021-02-24:
When configuring a component to use a PD service provided by another client, e.g.:

```
<start name="cpu_balancer">
   <provides>
     <service name="PD"/>
     <service name="CPU"/>
   </provides>
   ...
</start>

<start name="cpu_burner">
  ...
  <route>
    <service name="PD">  <child name="cpu_balancer"/> </service>
    <service name="CPU"> <child name="cpu_balancer"/> </service>
    ...
  </route>
</start>
```

leads to following exception in the init component:

```
[init] child "cpu_balancer"
[init]   RAM quota:  1800K
[init]   cap quota:  66
[init]   ELF binary: cpu_balancer
[init]   priority:   0
[init]   provides service PD
[init]   provides service CPU
[init] child "client"
[init]   RAM quota:  1800K
[init]   cap quota:  66
[init]   ELF binary: cpu_burner
[init]   priority:   1
[init] Error: PD session (label="cpu_burner", ram_quota=196608, cap_quota=13, diag=0) unavailable
[init] Error: Uncaught exception of type 'Genode::Service_denied'
[init] Warning: abort called - thread: ep
```

alex-ab, 2021-02-24:
In the [child.cc at line 819](https://github.com/genodelabs/genode/blob/master/repos/base/src/lib/base/child.cc#L819) the pd cap is invalid, since the child did not start yet. The commits bb40fda3cd07f46fd2a13513c3de66ca158c63c8 and d66b3f8c666772359c960eca03576df786bf7b48 showcase a working solution, but probably can be solved differently.

nfeske, 2021-02-24:
@alex-ab as follow-up to our offline discussion, here you can find a bare-bones example of starting a child. The `local_*_service` stuff can be removed. The parent services are extended on demand.

https://github.com/genodelabs/genode/blob/master/repos/os/src/server/loader/child.h

alex-ab, 2021-02-24:
The offline discussed easier fix b71c18c6a6235bfb97f864b4c09fe3e9582d0c8a result also in an exception:

```
[init] child "cpu_balancer"
[init]   RAM quota:  1800K
[init]   cap quota:  66
[init]   ELF binary: cpu_balancer
[init]   priority:   0
[init]   provides service PD
[init]   provides service CPU
[init] child "cpu_burner"
[init]   RAM quota:  1800K
[init]   cap quota:  66
[init]   ELF binary: cpu_burner
[init]   priority:   1
[init] Warning: 0x170b20 825
[init] Error: Uncaught exception of type 'Genode::Pd_session::Invalid_session'
[init] Warning: abort called - thread: ep
```

alex-ab, 2021-02-24:
> @alex-ab as follow-up to our offline discussion, here you can find a bare-bones example of starting a child. The `local_*_service` stuff can be removed. The parent services are extended on demand.
> 
> https://github.com/genodelabs/genode/blob/master/repos/os/src/server/loader/child.h

Thanks, I will try in the next days.

alex-ab, 2021-02-24:
In case, on the issue_4029 branch the run/cpu_balancer test case triggers the issue.

nfeske, 2021-02-24:
Thanks for doing this experiment.

alex-ab, 2021-02-25:
As discussed offline, [issue_4029_balancer_pd](https://github.com/alex-ab/genode/commits/issue_4029_balancer_pd) contains just the cpu_balancer with PD service support.

nfeske, 2021-02-25:
Sorry @alex-ab, I commented on the wrong issue. Moved/updated comment follows:

Commit https://github.com/nfeske/genode/commit/188be14b3e079ab66c45406d13ae0d5f28066b6d implements the proxy idea that we discussed. I named the proxy component "shim".

Commit https://github.com/nfeske/genode/commit/d5b5f4d8205dd0c19faca76c5926ca8fc0e13ea8 adjusts the cpu_balancer.run script.

With the two commits, I don't see the error anymore but I'm not 100% sure if the output is the expected one. When enabling the verbose mode, something useful seems to happen ("useful" as far as burning CPU cycles may get).

With https://github.com/nfeske/genode/commit/b78de8583c623fabad5fb0ed8d5d3100faf614d0, the run script finishes.

alex-ab, 2021-02-25:
@nfeske: Cool. Very appreciated. I will check/use asap.

nfeske, 2021-03-04:
Fixup https://github.com/genodelabs/genode/commit/607aeda99f5a85eca7d15853553865e53c4c227b removes the shim from the common routes.

nfeske, 2021-03-12:
Fixed in master.

closed 2021-03-12 by nfeske