genodelabs/genode #4030: test-libc_fifo_pipe fails on nova/x86_32

chelmuth, 2021-02-24:
Since the introduction of #3883 the test-libc_fifo_pipe depot-autopilot test fails nightly on 32-bit x86 hardware. This does not happen on Qemu or 64-bit x86 hardware. The log looks like follows.

```
[2021-02-24 03:48:36] [init -> depot_autopilot] --- Run "test-libc_fifo_pipe" (max 120 sec) ---
[2021-02-24 03:48:36] [init -> depot_autopilot] 
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.183 [init -> fifo-pipe-test] test started with 15 iterations
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.192 [init -> fifo-pipe-test] --- test iteration 0 started ---
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.201 [init -> fifo-pipe-test] re-starting echo
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.210 [init -> fifo-pipe-test] starting thread to send data to pipe
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.223 [init -> fifo-pipe-test] starting thread to receive data from pipe
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.241 [init -> fifo-pipe-test] joining sender thread 
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.291 [init -> init -> echo] echo started
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.598 [init -> fifo-pipe-test] sent 315392 bytes
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.671 [init -> fifo-pipe-test] joined sender thread
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.683 [init -> fifo-pipe-test] joining receiver thread 
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.698 [init -> init -> echo] piped 315392 bytes
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.709 [init -> fifo-pipe-test] received a total of 315392 bytes
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.719 [init -> fifo-pipe-test] joined receiver thread
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.727 [init -> fifo-pipe-test] --- test iteration 1 started ---
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.737 [init -> fifo-pipe-test] re-starting echo
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.744 [init -> fifo-pipe-test] starting thread to send data to pipe
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.757 [init -> fifo-pipe-test] starting thread to receive data from pipe
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.783 [init -> fifo-pipe-test] joining sender thread 
[2021-02-24 03:48:36] [init -> depot_autopilot] 0.831 [init -> init -> echo] echo started
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.133 [init -> fifo-pipe-test] sent 315392 bytes
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.211 [init -> fifo-pipe-test] joined sender thread
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.222 [init -> fifo-pipe-test] joining receiver thread 
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.238 [init -> init -> echo] piped 315392 bytes
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.250 [init -> fifo-pipe-test] received a total of 315392 bytes
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.260 [init -> fifo-pipe-test] joined receiver thread
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.268 [init -> fifo-pipe-test] --- test iteration 2 started ---
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.277 [init -> fifo-pipe-test] re-starting echo
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.286 [init -> fifo-pipe-test] starting thread to send data to pipe
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.298 [init -> fifo-pipe-test] starting thread to receive data from pipe
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.325 [init -> fifo-pipe-test] joining sender thread 
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.372 [init -> init -> echo] echo started
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.677 [init -> fifo-pipe-test] sent 315392 bytes
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.754 [init -> fifo-pipe-test] joined sender thread
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.762 [init -> fifo-pipe-test] joining receiver thread 
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.779 [init -> init -> echo] piped 315392 bytes
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.792 [init -> fifo-pipe-test] received a total of 315392 bytes
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.801 [init -> fifo-pipe-test] joined receiver thread
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.810 [init -> fifo-pipe-test] --- test iteration 3 started ---
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.819 [init -> fifo-pipe-test] re-starting echo
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.828 [init -> fifo-pipe-test] starting thread to send data to pipe
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.840 [init -> fifo-pipe-test] starting thread to receive data from pipe
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.865 [init -> fifo-pipe-test] joining sender thread 
[2021-02-24 03:48:37] [init -> depot_autopilot] 1.913 [init -> init -> echo] echo started
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.212 [init -> fifo-pipe-test] sent 315392 bytes
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.294 [init -> fifo-pipe-test] joined sender thread
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.305 [init -> fifo-pipe-test] joining receiver thread 
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.319 [init -> init -> echo] piped 315392 bytes
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.331 [init -> fifo-pipe-test] received a total of 315392 bytes
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.341 [init -> fifo-pipe-test] joined receiver thread
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.349 [init -> fifo-pipe-test] --- test iteration 4 started ---
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.359 [init -> fifo-pipe-test] re-starting echo
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.367 [init -> fifo-pipe-test] starting thread to send data to pipe
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.383 [init -> fifo-pipe-test] starting thread to receive data from pipe
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.405 [init -> fifo-pipe-test] joining sender thread 
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.454 [init -> init -> echo] echo started
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.755 [init -> fifo-pipe-test] sent 315392 bytes
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.843 [init -> fifo-pipe-test] joined sender thread
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.853 [init -> fifo-pipe-test] joining receiver thread 
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.865 [init -> init -> echo] piped 315392 bytes
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.875 [init -> fifo-pipe-test] received a total of 315392 bytes
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.884 [init -> fifo-pipe-test] joined receiver thread
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.892 [init -> fifo-pipe-test] --- test iteration 5 started ---
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.902 [init -> fifo-pipe-test] re-starting echo
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.910 [init -> fifo-pipe-test] starting thread to send data to pipe
[2021-02-24 03:48:38] [init -> depot_autopilot] 2.920 [init -> fifo-pipe-test] starting thread to receive data from pipe
[... 91 lines cut ...]
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.281 [init -> fifo-pipe-test] joined receiver thread
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.291 [init -> fifo-pipe-test] --- test iteration 13 started ---
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.300 [init -> fifo-pipe-test] re-starting echo
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.309 [init -> fifo-pipe-test] starting thread to send data to pipe
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.321 [init -> fifo-pipe-test] starting thread to receive data from pipe
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.346 [init -> fifo-pipe-test] joining sender thread 
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.395 [init -> init -> echo] echo started
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.694 [init -> fifo-pipe-test] sent 315392 bytes
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.785 [init -> fifo-pipe-test] joined sender thread
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.794 [init -> fifo-pipe-test] joining receiver thread 
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.807 [init -> init -> echo] piped 315392 bytes
[2021-02-24 03:48:43] [init -> depot_autopilot] 7.815 [init -> fifo-pipe-test] received a total of 315392 bytes
[... cut ...]
[2021-02-24 03:48:43] page fault, pd='init -> dynamic -> test-libc_fifo_pipe -> vfs' thread='ep' cpu=0 ip=0x10f75498 address=0x128848 stack pointer=0x401fe4d0 qualifiers=0x6 irUWp reason=1
[2021-02-24 03:50:36] [init -> depot_autopilot] 
[2021-02-24 03:50:36] [init -> depot_autopilot]  test-libc_fifo_pipe             failed   119.985  timeout 120 sec
```

As the page fault happens in *test-libc_fifo_pipe -> vfs* I'm quite sure the issue is associated to the new VFS plugin. Please note, the used test machine is a bit dated, i.e., *slow*.

chelmuth, 2021-02-24:
For the record, there are following options to run the test. Don't forget to enable `RUN_OPT += --depot-auto-update` in *etc/build.conf*.

```
make -C build/x86_32 KERNEL=nova run/depot_autopilot TEST_SRCS='' TEST_PKGS=test-libc_fifo_pipe
make -C build/x86_32 KERNEL=nova run/test PKG=test-libc_fifo_pipe
```

chelmuth, 2021-03-03:
Just checked our statistics again and discovered the test also fails on sel4/x86_32 but less regularly.

@sidhussmann do you have any news about this issue?

sidhussmann, 2021-03-04:
@chelmuth to reproduce, do I need actual x86_32 hardware or can I use x86_64 hardware in 32 bit mode?

chelmuth, 2021-03-04:
Up to now, the test failed only on an old *Intel Core2 Duo CPU E7300*. After your question I ran `make -C build/x86_32 KERNEL=nova run/depot_autopilot TEST_SRCS='' TEST_PKGS=test-libc_fifo_pipe TEST_REPEAT=until_failed` on *Intel Core i7-4770* and experienced the following error after 4 iterations with current staging.

```
[init -> depot_autopilot] 1.410 [init -> fifo-pipe-test] joining sender thread 
no RM attachment (WRITE pf_addr=0x128848 pf_ip=0x10f75498 from pager_object: pd='init -> dynamic -> test-libc_fifo_pipe -> vfs' thread='ep') 
page fault, pd='init -> dynamic -> test-libc_fifo_pipe -> vfs' thread='ep' cpu=0 ip=0x10f75498 address=0x128848 stack pointer=0x401fe4d0 qualifiers=0x6 irUWp reason=1
```

chelmuth, 2021-03-04:
After about twenty iterations, the same test hardware also fails in x86_64 mode.

sidhussmann, 2021-03-06:
@chelmuth thank you for making it easier for me to tackle this problem. I will take a closer look next week. 

On a side note, the related issue is #3583 and not #3883, as written in the issue description.

sidhussmann, 2021-03-11:
@chelmuth commit 5a60468dfb1a389668d7b127e04176a12e218155 fixed the page fault for me.

chelmuth, 2021-03-11:
Thanks for looking into this, I merged the fix to staging.

nfeske, 2021-03-12:
Thanks @sidhussmann for attending this issue. The fix reached the master branch just now.

closed 2021-03-12 by nfeske