genodelabs/genode #4031: VirtualBox 6

nfeske, 2021-02-24:
VirtualBox 6 is already in the works by @chelmuth and me since several months. So the time is ripe to open an official issue about it. Let us collect information about the currently supported feature set and the remaining limitations here.

nfeske, 2021-02-24:
In commit 6168456c63686dbcdf72cb3fa3f9490085d28410, which just entered the staging branch, there is still a bunch of commented-out code present. These leftovers should better be removed before mastering.

m-stein, 2021-02-24:
Just noticed it in my "git log" and am very happy :)
> just entered the staging branch

chelmuth, 2021-02-24:
> In commit [6168456](https://github.com/genodelabs/genode/commit/6168456c63686dbcdf72cb3fa3f9490085d28410), which just entered the staging branch, there is still a bunch of commented-out code present. These leftovers should better be removed before mastering.

I addressed this remark in 78b175a3f0543d6a9e602838ecb4187b054fdfa8.

chelmuth, 2021-11-04:
We released VirtualBox 6 with Sculpt 21.10. Further work will use dedicated issues.

closed 2021-11-04 by chelmuth