genodelabs/genode #4034: base-hw: support packaging of boards with multiple CPU architectures

nfeske, 2021-02-26:
The virt_qemu board is available for multiple architectures such as arm_v7a and arm_v8a. This situation can also happen for other boards, like Rpi3.

Currently, the base-hw/recipes/src/base-hw_content.inc packaging utility supports only one architecture per board. The source archive created for such a board should better contain the sources for all supported architectures, not just one.

Thanks @skalk for raising this issue.

nfeske, 2022-06-17:
I think we went into a wrong direction here. The added degree of freedom to allow one type of board to have different CPU architectures creates uncertainty that I would prefer to avoid. Ultimately, the degree of freedom is not enough either because one may wish to customize other board properties (link address, qemu args) differently per architecture, too (see https://github.com/genodelabs/genode-riscv/pull/3#issuecomment-1158072494). That is a rabbit hole I did not foresee.

Please let us revert this change and instead handle the virt_qemu case as distinct boards.

* base-hw/boards/virt_qemu_arm_v7a/
* base-hw/boards/virt_qemu_arm_v8a/
* base-riscv/boards/virt_qemu_riscv/
* base-hw/boards/virt_qemu_x86_64/

We'd gain a perfect correspondence between board names and directory/file names, avoid the complexity of special syntax in the board/*/arch files, and regain the intuitive (limited) relationship between board and architecture. In the real world, each board has a specific CPU after all. Let us establish the notion of the "board" as the single anchor for everything hardware-related with regard to both supported peripherals and the CPU.

Granted, this scheme comes with a certain amount of duplication. But I argue that it is rather marginal when put in relation to the benefit of a simpler scheme. The worst case is probably the (future?) board/virt_qemu_*/devices definition, which would need multiple copies. But I think that's acceptable because the number of architectures will remain rather bounded. The regular parts of the source tree (C++ code, recipes, mk files) outside the boards/ directory can handle the duplication between the boards by using includes.

m-stein, 2022-06-17:
@nfeske To me, your suggestion sounds perfectly reasonable. I'd appreciate the clarity that is gained through the virt_qemu_<arch> names.

skalk, 2022-06-17:
I do not know where the different link address needs stem from with regard to riscv. My assumption was that the machine layout is identical including memory and virtio peripherals, but only the CPU differs. That is why I was thinking it would be benificial to share the virt_qemu board description, related raw and pkg packages and run-script snippets.
I was also not assuming that we want to support different QEMU versions in parallel, now that we've reached to use one and the same version. Therefore, I did not see the need for different qemu args.

However, I won't stay in the way, when it turns out that my assumptions are not correct.

ssumpf, 2022-08-11:
@nfeske: beafcbb and https://github.com/ssumpf/genode-riscv/commit/0031d64c14a1b50852409ab9f06d64b69e6777da (RISC-V) are my first take on the `virt_qemu` board renaming. What do you think?

nfeske, 2022-08-11:
This is exactly what I had in mind. Thanks a lot @ssumpf for applying the scheme. Is the commit ready for staging from your point of view?

ssumpf, 2022-08-11:
@nfeske: From my point of view yes.

nfeske, 2022-08-11:
Just merged the commit to staging. I'd appreciate you keeping an eye on the automated build/test results.

ssumpf, 2022-08-11:
> Just merged the commit to staging. I'd appreciate you keeping an eye on the automated build/test results.

Ack

nfeske, 2022-08-17:
Merged to master.

commit e06d38c6 (2021-02-26): base-hw: support packaging of multi-arch boards

With this patch, the board/<name>/arch property file can have multiple
lines where each line denotes an instruction-set architecture supported
for the board.

Issue #4019
Fixes #4034
commit c68443e2 (2021-03-12): base-hw: support packaging of multi-arch boards

With this patch, the board/<name>/arch property file can have multiple
lines where each line denotes an instruction-set architecture supported
for the board.

Issue #4019
Fixes #4034
closed 2022-08-17 by nfeske