genodelabs/genode #4035: sculpt: keyboardio Atreus keyboard does not work

nfeske, 2021-03-02:
With the switch from the monolithic USB driver to the USB-host driver + USB HID driver, my keyboardio Atreus keyboard no longer works. I tracked this down to the following problems.

* The USB device has 5 interfaces, three of which are HID interfaces. Unfortunately, the interface 0 is not a HID interface, which prompts the USB host driver to report a non-HID "class" attribute for the device. The reporting of the class of interface 0 as the device class for the entire device cuts too short and should be changed. However, to keep the immediate fix as non-invasive as possible, I refined the heuristics behind the "device class" value, setting it to HID whenever at least one HID interface is present.

* The USB HID driver wrongly probes all interfaces of the keyboard, also the non-HID interfaces. This produces the following error message during the initialization.
   ```
   dev_err: couldn't find an input interrupt endpoint
   ```
  This problem can be fixed by checking the HID device class as a precondition of the probing the interface at the HID driver code.

* The interface 4 offers some kind of boot-time keyboard. The submission on an interrupt URB results in the immediate error ENOSPC, which is a bandwidth reservation error. Since submitting URBs is an asynchronous operation at the USB session interface, which can never return this condition as immediate response to the submission, the error is not detected / handled at the submission phase at the HID driver. However, it shows up at the HID driver as response to the interrupt request, delivered asynchronously. However, the asynchronous interrupt handler does not anticipate ENOSPC, to the effect that this error re-occurs in a loop. It disappears once when interacting with the keyboard. In the case at hand, the interface is actually not needed. To settle the loop, I let the USB host driver reflect ENOSPC as a STALL. Since this is not expected to be correct in all cases, I added a diagnostic message hinting at this corner case.

With those changes, the keyboard becomes usable with the new version of Sculpt.

nfeske, 2021-03-02:
With the commits above, I can use the keyboard in Sculpt OS. @ssumpf, @skalk Are you fine with the changes?

skalk, 2021-03-02:
Oh dear, what a chain of shortcomings! I'm fine with the implementation, although I fear that the interim fix will stay for longer and might produce issues in other settings, e.g., webcam with HID functionality, or audio. But I do not have the insights in those scenarios like @ssumpf and @cnuke. On the other hand, I can perfectly understand that you need a solution right now. So green lights from my side.

nfeske, 2021-03-02:
I already had a chat with @cnuke and @ssumpf today about the further steps. It is clear to everyone of us that further work is needed.

Actually, I already took a few steps in this direction today but stopped because I did not want to introduce major changes in-between the Genode 21.02 release and the upcoming Sculpt release. Once Sculpt is out, I'll go forward and revisit the reporting/config concept.

skalk, 2021-03-02:
I see. Thank you for the explanation! I should have known it better that you do not favor an interim solution ;-). I also like the code update of the usb host report function. It looks much better now, and the ugly snprintf went away.

ssumpf, 2021-03-03:
I have major changes to the USB session interface in order to adjust to the lessons we have learned on my TODO list.

commit 0b90af18 (2021-03-02): usb_host: refine class heuristics on device report

This patch is an interim fix for using HID devices that offer a HID
interface as not the first interface. It also supplements the
interface classes as supplemental information to the USB-devices
report.

Fixes #4035
commit e1bb0e8e (2021-03-12): usb_host: refine class heuristics on device report

This patch is an interim fix for using HID devices that offer a HID
interface as not the first interface. It also supplements the
interface classes as supplemental information to the USB-devices
report.

Fixes #4035
closed 2021-03-12 by nfeske