genodelabs/genode #4036: NIC router: fix uncaught bit alloc exception

m-stein, 2021-03-03:
While updating his Sculpt and trying to apply it's usual network setup with a manually configured NIC router, @nfeske observed managed to bring the his NIC-router instance down with a uncaught exception of type Bit_allocator_dynamic::Out_of_indices. In this incident, re-configuration with several domains, a dynamically IP-configured VM and possibly jumping of the VM between domains through changing policies were involved. The specific pattern that lead to the issue isn't known.

m-stein, 2021-03-03:
Tracking all possible paths that could trigger the mentioned exception, I could find two places where the exception should be caught but wasn't by now. The following patch fixes this and adds unconditional log output, so, we will be able to debug the problem a bit more in detail next time:
023dc00a02 nic_router: fix uncaught bit alloc exception

m-stein, 2021-03-03:
This updated version more correctly marks a domain invalid if the DHCP server creation failed because of the exception:
a68fa655b0 nic_router: fix uncaught bit alloc exception

nfeske, 2021-03-03:
Merged to staging.

m-stein, 2021-03-04:
I removed the fixed label as there is new information regarding this issue.

I could reproduce different uncaught IP-allocation-related exceptions with these tests:
9d9e3ba2ac nic_router: tests to trigger uncaught exceptions

Note that test 4 doesn't trigger any exception although it does something similar but with domains switching.

m-stein, 2021-06-22:
The issue was closed automatically by 25a212a. I reopened it as 9d9e3ba can be used to systematically reproduce uncaught exceptions. I also wanted to add a note that the issue was originally **observed with Sculpt 21.03**.

m-stein, 2022-05-31:
Commit https://github.com/genodelabs/genode/commit/9d9e3ba2ac22a4c98ace2fe5f78c7a042fb47f66 shows no uncaught exceptions on current staging anymore. I'll therefore close the issue.

commit c9ac3602 (2021-03-03): nic_router: fix uncaught bit alloc exception

* catch every possible path of the Bit_allocator_dynamic::Out_of_indices
  exception
* add unconditional log output in the new "catch" directives, so, we will be
  able to debug the problem a bit more in detail next time

Fixes #4036
commit 25a212aa (2021-03-12): nic_router: fix uncaught bit alloc exception

* catch every possible path of the Bit_allocator_dynamic::Out_of_indices
  exception
* add unconditional log output in the new "catch" directives, so, we will be
  able to debug the problem a bit more in detail next time

Fixes #4036
closed 2022-05-31 by m-stein