genodelabs/genode #4037: platform driver: use the same label for all device PDs

nfeske, 2021-03-03:
When booting Sculpt with Genode 21.02, a number of warnings appear:

```
[init] Warning: re-attempted PD session request 2 times (args: ram_quota=8K, cap_quota=0)
[init] Warning: re-attempted PD session request 4 times (args: ram_quota=16K, cap_quota=0)
[init] Warning: re-attempted PD session request 8 times (args: ram_quota=32K, cap_quota=0)
[init] Warning: re-attempted PD session request 16 times (args: ram_quota=52K, cap_quota=12)
[init] Warning: re-attempted PD session request 2 times (args: ram_quota=8K, cap_quota=0)
[init] Warning: re-attempted PD session request 4 times (args: ram_quota=16K, cap_quota=0)
[init] Warning: re-attempted PD session request 8 times (args: ram_quota=32K, cap_quota=0)
[init] Warning: re-attempted PD session request 16 times (args: ram_quota=52K, cap_quota=12)
[init] Warning: re-attempted PD session request 2 times (args: ram_quota=8K, cap_quota=0)
[init] Warning: re-attempted PD session request 4 times (args: ram_quota=16K, cap_quota=0)
[init] Warning: re-attempted PD session request 8 times (args: ram_quota=32K, cap_quota=0)
[init] Warning: re-attempted PD session request 16 times (args: ram_quota=52K, cap_quota=12)
[init] Warning: re-attempted PD session request 2 times (args: ram_quota=8K, cap_quota=0)
[init] Warning: re-attempted PD session request 4 times (args: ram_quota=16K, cap_quota=0)
[init] Warning: re-attempted PD session request 8 times (args: ram_quota=32K, cap_quota=0)
[init] Warning: re-attempted PD session request 16 times (args: ram_quota=52K, cap_quota=12)
[init] Warning: re-attempted PD session request 2 times (args: phys_start=0x0, phys_size=0x100000000, managing_system=yes, diag=0, label="drivers -> ", ram_quota=11469, cap_quota=0)
[init] Warning: re-attempted PD session request 4 times (args: phys_start=0x0, phys_size=0x100000000, managing_system=yes, diag=0, label="drivers -> ", ram_quota=19661, cap_quota=0)
[init] Warning: re-attempted PD session request 8 times (args: phys_start=0x0, phys_size=0x100000000, managing_system=yes, diag=0, label="drivers -> ", ram_quota=36045, cap_quota=0)
[init] Warning: re-attempted PD session request 2 times (args: phys_start=0x0, phys_size=0x100000000, managing_system=yes, diag=0, label="drivers -> ", ram_quota=11469, cap_quota=0)
[init] Warning: re-attempted PD session request 4 times (args: phys_start=0x0, phys_size=0x100000000, managing_system=yes, diag=0, label="drivers -> ", ram_quota=19661, cap_quota=0)
[init] Warning: re-attempted PD session request 8 times (args: phys_start=0x0, phys_size=0x100000000, managing_system=yes, diag=0, label="drivers -> ", ram_quota=36045, cap_quota=0)
```

I tracked down those warnings to the PD sessions created by the platform driver as device PDs. The platform driver labels those sessions after its clients (the client's label). Since this label is then step-by-step prefixed by the parent hierarchy towards core, the resulting session args exceed the maximum length of 160 bytes when the top-level init tries to pass it to core. In this condition, the `Rpc_in_buffer::string` mechanism kicks in, zapping the string to an empty one. This, in turn, results in a cap/ram quota set to zero, which then triggers the re-attempt with successively increased quotas a seen in the log.

The easiest way to avoid those warnings would be the use of a static label "device PD" instead of the client-provided label. So the resulting label at the top-level init is bounded by the hierarchy level of the platform driver.

@alex-ab Do you deem the imprinting of the driver's label into the device PD's label as important, or can we drop it?

nfeske, 2021-03-03:
Commit https://github.com/nfeske/genode/commit/b2a74821b77228e9758b6b46365fb3492d4100dc fixes the issue.

nfeske, 2021-03-03:
Commit https://github.com/nfeske/genode/commit/2b78f952caa6bd23ffc3c9f83dd29c5d55259200 may help us to diagnose such problems in the future.

alex-ab, 2021-03-04:
> @alex-ab Do you deem the imprinting of the driver's label into the device PD's label as important, or can we drop it?

It is not important, wee may drop the labeling here.

nfeske, 2021-03-04:
Thanks for the feedback. Merged to staging.

commit 3c6f8973 (2021-03-04): platform driver: use same label for all device PDs

Fixes #4037
commit ef8af741 (2021-03-04): base: detect truncated session args

This patch reflects the truncation of session arguments (e.g., by very
long session labels) by denying the session request.

Fixes #4037
closed 2021-03-12 by nfeske