genodelabs/genode #4039: init: Uncaught exception of type 'Genode::Bit_array_base::Invalid_clear'

trimpim, 2021-03-05:


trimpim, 2021-03-05:
f2d054103d77482d20f41718b8458deea6673114 reproduces the uncaught exception described in https://lists.genode.org/pipermail/users/2021-February/007540.html within a few minutes.

Currently I only have tested this on base-linux.

nfeske, 2021-03-08:
@trimpim thank you very much for your work to make the issue reproducible! I'll investigate this issue today.

nfeske, 2021-03-08:
I can reproduce this problem after reducing the timeout at https://github.com/trimpim/genode/commit/f2d054103d77482d20f41718b8458deea6673114#diff-a5e200edbb5c25790208fed2dda90ff0976b762865ce26aa8523574e892c8b67R46 to 1000.

nfeske, 2021-03-08:
Thanks to @trimpim's excellent test and the great help of @chelmuth, I was able to encircle the problem today.

Commit https://github.com/nfeske/genode/commit/a8692620ae6bc1a7a025b1c6cb1038a41498cb1c should fix the issue. I still need to think of a good way to add a regression test for this situation to the test-init package but you may give the fix a try already.

nfeske, 2021-03-09:
I wonder, can you observe the fault in the scenario you shared with me in https://github.com/trimpim/genode/commit/f2d054103d77482d20f41718b8458deea6673114?

nfeske, 2021-03-09:
@trimpim please don't invest too much time for testing. Let's first wait for the CI results on my side.

nfeske, 2021-03-09:
> I can see this issue in Sculpt on NOVA. Investigating...

Fixed by https://github.com/genodelabs/genode/commit/7ba05d04b349f2d6760809e701dc4a97f7a14f25

@trimpim can you give this fixup a try?

trimpim, 2021-03-10:
@nfeske with both patches applied, our scenario now runs for more than 250 configuration changes whereas before it stopped after a dozen or so.

commit 0ba859c3 (2021-03-08): base: fix child destruction while close requested

This patch fixes a corner case where a child is destructed while a
asynchronous close request to a sibling server is still pending.

The child immediately discarded the session ID as the end of the
close-session processing, assuming that this ID is never to be needed
again. The session-state continues to exist to handle asynchrous close
protocol with the server.

However, if the child is destructed at this point (before the server
responded to the session request), the destruction of the child would
not cover the discharging of the session sta [... cut]
closed 2021-03-12 by nfeske