genodelabs/genode #4040: boot_dir/nova: enable Intel HWP

cnuke, 2021-03-05:
While debugging some other issue I noticed that my target system drives the CPU only with its base frequency, leaving existing performance reserves untapped (cf. #3871). During an offline discussion it was decided to enable the feature unconditionally to reap the benefits when running Sculpt as well as having more test coverage on how different systems might behave.

cnuke, 2021-03-05:
Commit a710caf sets the required bender option unconditionally when using NOVA.

nfeske, 2021-03-05:
Thanks @cnuke for the patch. I merged it to staging right away.

nfeske, 2021-03-10:
Fixup https://github.com/genodelabs/genode/commit/c69d2a0c39847f841bcd8cf86f32d44213b22ffa removes the HWP message from the build log, which is otherwise printed on each build.

commit f2b73701 (2021-03-05): boot_dir/nova: enable intel_hwp

Instruct bender to enable Intel HWP on supported CPUs unconditionally
to gain more insight by using the feature most prominently in Sculpt.

Fixes #4040.
closed 2021-03-12 by nfeske