genodelabs/genode #4051: Genode:String byte-wise copy

sergey-platonov, 2021-03-17:
I have noticed that `Genode::String` creation uses the byte-wise copy [here](https://github.com/genodelabs/genode/blob/master/repos/base/include/util/string.h#L635):
```
  void out_string(char const *str, size_t n)
  {
      while (n-- > 0 && _capacity_left() && *str)
          _append(*str++);
  }
```

Maybe it will be more optimal in the sense of performance to use a `memcpy` here? We will need to traverse the string to find the termination zero, but then we will be able to copy the string as a whole. Well, on certain platforms, of course, but still. It seems, that it will be better at least in the sense of data locality. Will you consider a pull request with the following changes? 

```
  void out_string(char const *str, size_t n)
  {
      n = min(CAPACITY - _num_chars - 1 - n, n);
      size_t len = 0;
      char const *s = str;
      for (; s && *s && len < n; ++s, ++len);
      n = min(n, len);
      memcpy(&_buf[_num_chars], str, n);
      _num_chars += n;
  }
```

nfeske, 2021-03-17:
Thank you for the suggestion. However, I would not consider this change at this time, for the following reasons:

* I have no evidence that the performance of the `out_string` function presents a bottleneck in practice. Without evidence, I would argue that it isn't worth optimizing.
  In situations where the efficiency of string processing becomes crucial, Genode's string API is not the answer. At least we don't try to pitch it as an application-level API. One should better rely on the beaten tracks like using the C++ standard library or some other library that is specialized for this topic. Please find my reasoning explained in the following article https://genodians.org/nfeske/2019-01-22-conscious-c++ (Section "Narrowing the scope to the operating-system level").

* In terms of performance, it is not intuitively clear to me how the suggested alternative is faster than the original one because it iterates over the string twice: the for-loop for determining the `len`, and the `memcpy`.

* Most importantly, I'm uneasy with the increase of complexity of the logic. Pointer arithmetic and integer overflows are risky. Hence we try hard to avoid it in our code. E.g., the first line computes `CAPACITY - _num_chars - 1 - n`. From looking at the code, it is not immediately clear to me that an underflow has no undesired effects. In contrast, the original code is quite easy to assess.

I hope you are not disappointed by my dismissive response.

BTW, for improving the performance of Genode's string/memcpy functions, let me recommend giving the compiler's optimization level 3 a try. By default, Genode's components are compiled with optimization level 2. To change the optimization level for your component, try adding the following line to your target.mk file:
```
CC_OLEVEL := -O3
```

In some cases (depending on the architecture and the particular component), such a simple change can make a huge difference without touching any code.

sergey-platonov, 2021-03-17:
Thank you for the answer, Norman.

I was thinking to set up performance tests to actually measure different scenarios in case your only concern would be an actual performance impact. But, since it's not the most important argument, let's leave it like that.

> I hope you are not disappointed by my dismissive response.

Of course not, I wanted to discuss it and hear your opinion on the matter.

closed 2021-03-17 by sergey-platonov