genodelabs/genode #4056: lwip: Zero TCP Window when transferring large data

throwException, 2021-03-23:
**Scenario**

Use lwip to send 1 MiB to a remote TCP server with a latency of about 10ms and receive the same data back.

**Expected result**

The data is received back.

**Actual result**

The script stalls.

**Detail**

Wireshark shows a TCP window size of zero from the test client.

**Steps to reproduce**

1. Checkout the test code from: https://github.com/throwException/genode/tree/test-lwip_large_data_echo_tcp

2. Adapt the runscript at `repos/libports/run/lwip_large_data_echo_tcp.run` to reflect your local tap device, IP network and the server's IP address.

3. Start netcat on the remote server: 3. `ncat -l -k -p 5000 -c cat`

4. Start Wireshark and capture your local tap device

5. Run the `lwip_large_data_echo_tcp` runscript.

chelmuth, 2021-03-25:
Thanks for the excellent bug report! Do you possibly have any hints what might cause this behavior?

throwException, 2021-03-25:
> Thanks for the excellent bug report! Do you possibly have any hints what might cause this behavior?

From what I understand of TCP zero window, I would suspect that the receive buffer is too small.

I tried tweaking lwip options at `contrib/lwip-6e0661b21fde397041389d5d8db906b5a6543700/src/lib/lwip/src/include/lwip/opt.h` but couldn't improve anything.

chelmuth, 2021-07-22:
I adapted the test to current staging (with Uplink-only linux_nic_drv) and also enabled lwip TCP-window debugging messages. The now log now reflects that the congestion window is very dynamic and, therefore, the effective TCP window is changed often. In the end, lwip receives a window update to 0 from the server side on ACK.

```
[init -> test-large_data_echo_tcp] tcp_receive: congestion avoidance cwnd 65364
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 4992, cwnd 65364, wnd 4992, effwnd 5792, seq 782902, ack 778558
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 4992, cwnd 65364, wnd 4992, effwnd 5792, seq 782902, ack 778558
[init -> test-large_data_echo_tcp] tcp_receive: window update 768
[init -> test-large_data_echo_tcp] tcp_receive: congestion avoidance cwnd 65364
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 768, cwnd 65364, wnd 768, effwnd 1448, seq 782902, ack 782902
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 768, cwnd 65364, wnd 768, effwnd 1448, seq 782902, ack 782902
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 768, cwnd 65364, wnd 768, effwnd 1448, seq 782902, ack 782902
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 768, cwnd 65364, wnd 768, effwnd 1448, seq 782902, ack 782902
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 768, cwnd 65364, wnd 768, effwnd 1448, seq 782902, ack 782902
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 768, cwnd 65364, wnd 768, effwnd 768, seq 782902, ack 782902
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 768, cwnd 65364, wnd 768, effwnd 768, seq 782902, ack 782902, i 0
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 768, cwnd 65364, wnd 768, effwnd 1448, seq 783670, ack 782902
[init -> test-large_data_echo_tcp] tcp_receive: window update 0
[init -> test-large_data_echo_tcp] tcp_receive: congestion avoidance cwnd 65364
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
[init -> test-large_data_echo_tcp] tcp_output: snd_wnd 0, cwnd 65364, wnd 0, effwnd 680, seq 783670, ack 783670
```

throwException, 2024-04-13:
> This issue seems stale (or not?).

@chelmuth It's stale from our point of view.

closed 2024-04-15 by chelmuth