genodelabs/genode #4057: ahci_drv, part_block do not support reconfiguration of policies

ehmry, 2021-03-25:
`ahci_drv` and `part_block` both generate reports that might be used to generate client policies. Currently it is not possible to reconfigure either after their reports are processed because their configuration ROMs are never updated.

ehmry, 2021-03-25:
Easy fix at 5833181782a158bb67c3b569d872fbbdf75ee51a

chelmuth, 2021-03-26:
Your fix suggests the components are dynamically reconfigurable, which is only true for new session requests. Invalidating old policies (and thereby access permissions) is not considered, but should be integral part of the dynamics.

ehmry, 2021-04-06:
Yes, the short-term problem solved here is that it may seem intuitive that new policies could be added, but really the components need to be restarted for that to work.

closed 2021-04-20 by ehmry