genodelabs/genode #4061: 'egl_api' port preparation error

cproc, 2021-03-30:
When I try to prepare the 'egl_api' port on the current staging branch, an error occurs:

```
egl_api  download https://raw.githubusercontent.com/KhronosGroup/EGL-Registry/5804dc63f9ca70d5e9a6e31addec6936e2950ee4/api/EGL/egl.h
cp: cannot create regular file '/.../contrib/cache/ac71f423a7e7f1d39fe980aa8bf3c60eccf31c323d8ac5f734895783cdad6e99_include/EGL/egl.h.tmp': No such file or directory
make[2]: *** [/.../tool/ports/mk/install.mk:208: egl.h.file] Error 1
make[1]: *** [/.../tool/ports/mk/prepare_single_port.mk:43: _install_in_port_dir] Error 2
make: *** [/.../tool/ports/prepare_port:47: egl_api] Error 2
```

This error seems to be related to issue #4028, although for some reason the nightly 'check-ports' test said

```
OK    ... source egl_api
```

I also wonder about the dependency of the 'egl_api' port when doing the following steps in an unprepared staging tree:

- create build directories for x86_64 and arm_v7a
- enable all repositories in build.conf
- run 'make' in x86_64 build directory (only for the missing ports message)
- prepare all reported missing ports
- run 'make' in arm_v7a build directory

Now a message about missing 'egl_api' port appears (it didn't appear when building for x86_64):

```
Error: Ports not prepared or outdated:
  egl_api

You can prepare respectively update them as follows:
  /.../tool/ports/prepare_port egl_api
```

When grepping for 'egl_api' in the 'repos' tree, only the following two files are found:

```
libports/recipes/api/egl_api/content.mk:PORT_DIR := $(call port_dir,$(REP_DIR)/ports/egl_api)
libports/lib/import/import-egl_api.mk:INC_DIR += $(call select_from_ports,egl_api)/include
```

From this output it is not clear to me which application or library actually depends on 'egl_api' and why this dependency apparently only exists on ARM.

nfeske, 2021-03-31:
Thanks @tomga for the super quick response. I merged your fix to staging just now.

> From this output it is not clear to me which application or library actually depends on 'egl_api' and why this dependency apparently only exists on ARM.

To investigate those inter-dependencies, a look into the generated `<builddir>/var/libdeps` file for both architectures may be insightful.

tomga, 2021-03-31:
> To investigate those inter-dependencies, a look into the generated <builddir>/var/libdeps file for both architectures may be insightful.

I already mentioned it yesterday and actually I tried to find some important difference but they are quite big (more than 5500 lines) and the only difference I spotted regarding `egl_api` was that in one case it was added to `MISSING_PORTS` variable and did not in the other. That's why I asked for an answer for this puzzle if someone knows the answer.

cproc, 2021-03-31:
Mystery solved: as mentioned by @tomga,

```
LIBS = $(sort $(ALL_LIB_MK_FILES:.mk=))
```

in `repos/base/src/lib/target.mk` adds `egl_api` to the `LIBS` variable.

Then, in `repos/base/mk/dep_prg.mk`

```
 include $(foreach LIB,$(LIBS),$(call select_from_repositories,lib/import/import-$(LIB).mk))
```

includes the `import-egl_api.mk` file, which adds the `egl_api` port reference to the `INC_DIR` variable:

```
INC_DIR += $(call select_from_ports,egl_api)/include
```

And finally, in `repos/dde_rump/lib/import/import-rump.mk`

```
ifeq ($(filter-out $(SPECS),arm),)
        # rump include shadows some parts of 'machine' on ARM only,
        # Therefore, it must be included before RUMP_BASE/include/machine
        INC_DIR := $(RUMP_PORT_DIR)/src/sys/rump/include $(INC_DIR)
        INC_DIR += $(RUMP_PORT_DIR)/src/sys/arch/arm/include
endif
```

evaluates the `INC_DIR` variable on ARM only and detects the missing port.

tomga, 2021-03-31:
@cproc thanks for explanation. I completely missed fact that INC_DIR is not evaluated until really used.

commit 8fdff6d5 (2021-03-31): prepare_port: cache handling for files with paths

This patch is a follow-up fix for issue #4028 that fixes #4061.
commit 0a849a16 (2021-04-20): prepare_port: cache handling for files with paths

This patch is a follow-up fix for issue #4028 that fixes #4061.
closed 2021-04-20 by nfeske