genodelabs/genode #4069: allow event_filter to mute superfluous unknown key events from quirky keyboards

jschlatow, 2021-04-08:
When I was trying sculpt on my current laptop, I noticed that the GUI became unresponsive to mouse clicks once I typed in the wifi password. Since my external keyboard already required special treatment with Linux, I started investigating...

First of all, the keyboard is a Cherry MX Board 3.0S. Strangely, it generates extra scancodes with every key press/release. Both, in Linux and Sculpt, these are mapped to the key code 240, i.e. KEY_UNKNOWN. This would not be much of an issue though if these were well-behaved. Unfortunately, the KEY_UNKNOWN events are interleaved with the normal key events and leave the KEY_UNKNOWN in a pressed state. Let's say I press the `j` key, I get a sequence of key events like:
```
PRESS KEY_J
RELEASE KEY_UNKNOWN
RELEASE KEY_J
PRESS KEY_UNKNOWN
```

This becomes an issue because it a) hinders the detection of key sequences and b) keep KEY_UNKNOWN in a pressed state. The latter explains why the GUI becomes unresponsive to mouse clicks: Looking at nitpickers keystate report after the first key stroke on my external keyboard, it will report mouse clicks as combined key pressed of KEY_UNKNOWN and BTN_LEFT, which is a perfectly correct interpretation of the presented key events.

I believe the best solution is to mute KEY_UNKNOWN events using event_filter's remap policy. Naively, I tried remapping KEY_UNKNOWN to KEY_RESERVED (which I basically did on my Linux system). Yet, it requires a few patches to event_filter to make this policy work...

jschlatow, 2021-04-08:
Commits 8f451f1 and 6bf7e9c contain the functional changes. For debugging purposes, I also extended event_dump with a passthrough functionality in 5776354 to be able to place it between the usb_hid_drv and event_filter for instance. Since multiple key codes will print as KEY_UNKNOWN, I also modified the `Input::Event::print` method in 93be54e to log the key code as well. Note, that the key code is only printed for KEY_UNKNOWN presses so that existing test cases are not affected by this change. I also stumbled upon a minor inconsistency in the drivers_managed-pc raw recipe (see cd9d6c2).

By the way, I was wondering why KEY_UNKNOWN is defined as 0x1fe in [keycodes.h](../blob/master/repos/os/include/input/keycodes.h). The usb_hid driver maps all unknown scancodes to code 240, hence I expected to find a matching definition in keycodes.h.
I was also pondering about whether it would be worth distinguishing KEY_UNKNOWN and undefined key codes (e.g. KEY_UNDEF) but could not make my mind about that. Any opinions?

As I'm writing this, another idea just pops up into my mind: Alternative to having an ambiguous definition of KEY_UNKNOWN, we could define placeholder symbols for all currently undefined key codes, e.g. KEY_239 for key code 239. This way, we would circumvent the ambiguity of KEY_UNKNOWN and also enable remapping of any key code.

nfeske, 2021-04-09:
What do you think about making the dropping of specific key codes explicit in the event-filter configuration instead of attaching builtin semantics to specific codes? For example, by adding a `<ignore-key>` node type to the remap filter,  one could write:

```
<remap>
   <ignore-key name="KEY_UNKNOWN"/>
   ...
 </remap>
```

But the mechanism may be useful for other keys as well. E.g., some music fans might want to apply it to KEY_VOLUMEDOWN.

Regarding the event-dump component, are you aware of nitpicker's `<report keystate="yes"/>` feature? This was added for debugging issues like this one.

In general, I think we should not give too much weight to keys of code KEY_UNKNOWN. They are annoying but not interesting after all. So the extension of the `print` code as suggested in 93be54eb2ead2bd5b71021f0d890f710a283061d should probably better be kept as a debug commit, shouldn't it?

jschlatow, 2021-04-26:
@nfeske Please have a look at my updated branch [4069_allow_event_filter_to_mute_key_events](https://github.com/genodelabs/genode/compare/master...jschlatow:4069_allow_event_filter_to_mute_key_events).

I added a `<ignore-key>` node to the remap filter and a `<log>` filter node as you mentioned in an offline discussion.

Most of the issues with my keyboard were solved by a factory reset, but two quirks remain:
- The first press on the CAPSLOCK key releases a key press of KEY_POWER. The key release never follows. All following presses of CAPSLOCK work fine.
- The first press on NUMLOCK emits a whole bunch of additional key presses that are never released: KEY_A, KEY_D, KEY_R, KEY_Y, KEY_SPACE, KEY_F3, ...

Unfortunately, I cannot reproduce this behaviour on my Linux system.

Both issues leave nitpicker in a state with pressed keys and renders the system unusable.
I can fix the capslock problem with the `<ignore-key>` feature, since I don't use KEY_POWER.
However, in the numlock case, ignoring is clearly not an option. As the additional key presses are enclosed by the press and release event of numlock, this could potentially be dealt with by a new combo filter. Alternatively, a reset functionality for nitpicker (i.e. flushing the state of pressed keys) could be helpful as a workaround to get it into a valid state again. What do you think?

jschlatow, 2021-04-27:
@nfeske Thanks for the review. I pushed two fixup commits.

nfeske, 2021-04-27:
While I'm at merging your commits, should I also add ee8b051539bfd7097617590bb05f3b3b5bf3b379 to staging?

jschlatow, 2021-04-27:
> While I'm at merging your commits, should I also add [ee8b051](https://github.com/genodelabs/genode/commit/ee8b051539bfd7097617590bb05f3b3b5bf3b379) to staging?

Yes, that commit is still valid, though only for consistency as mentioned in the commit message.

jschlatow, 2021-04-27:
Thanks for merging.

> I wonder, to perfectly wrap up this nice package, how do you think of exercising the new features in the automated event_filter test?

I'll have a try.

What's your opinion on the numlock matter [mentioned above](https://github.com/genodelabs/genode/issues/4069#issuecomment-826951843)?

nfeske, 2021-04-27:
> What's your opinion on the numlock matter mentioned above?

Thanks for the reminder.

I'm divided. On the one hand, I see the need. On the other hand, it pains me to inflate the complexity of nitpicker on the account of the brokeness of a particular keyboard. I also currently lack an obvious idea of a suitable interface for invoking such a reset.

I'm also hesitant to jump at it because the erratic behavior of the keyboard may not only trip out nitpicker but also the event filter which sits in-between the driver and nitpicker. So the reset may tape over the problem for nitpicker while an inconsistent key count is still observed by the event filter. In particular, the event filter allows for configuration only if the key count is zero (follow the code of `all_sessions_idle()`. Once the strange sequence occurs, the filter can no longer be reconfigured, which would just be observed as another bug like "unable to switch keyboard layout on Cherry MX 3.0".

Your statement that you were not able to reproduce the issue on Linux made me curios. May there exist a quirk that is applied on Linux but not in our usb_hid driver? To me, it seems most sensible to address the strange behavior as close to the device as possible (if possible) instead of fixing the broken event stream after the fact. Hence, I'd appreciate if you could take a closer look contrasting Linux' HID driver with ours. Both should be very similar because our's is ported from Linux.

jschlatow, 2021-04-27:
Thanks for the feedback. I agree that adding such complexity only to deal with a broken device should be avoided.

I'll have a closer look at the root cause. I'm curious anyway ;-)

jschlatow, 2021-04-27:
Commit 11d50bd adds the corresponding test cases.

nfeske, 2021-04-28:
That's excellent. Thanks for the effort! I merged your test-case commit to staging.

closed 2021-05-05 by nfeske