genodelabs/genode #4070: lx_fs: add watch

trimpim, 2021-04-09:


nfeske, 2021-04-19:
@trimpim What a splendid commit series! I very much appreciate your great care for the details and the focus on testing. I will merge your commits right after the next master update.

The watch feature is really cool to have. I was actually tempted to add it at one point to make the work with goa more flexible (e.g., for live editing of menu-view dialogs). So I'm delighted about you providing the feature.

trimpim, 2021-04-19:
@nfeske I have found an error during stress testing the feature, and am working on a fixup.

nfeske, 2021-04-20:
@trimpim Thanks for the heads-up. Please give me a short notice when you are comfortable with me picking up the commits.

trimpim, 2021-04-21:
@nfeske fd1ba22d18d54885df7e24bfb7ec0ff390ddf931 fixes an error in the watch implementation.

However, during testing I have encountered a problem in the interaction of `lx_fs` and `fs_rom` in the case when a watched file is changed frequently from the Linux side.

In our scenario we have two components watching the same file via a `fs_rom`. From time to time it happens, that one of the components no longer gets notified about changes.

Adding some debug print to both `lx_fs` and `fs_rom` I see the following pattern:

```
[init -> lx_fs]  Notifier::entry()::igonre_fn  watch cap count=2
[init -> lx_fs]  Watch_node::watch_respons()
[init -> lx_fs]  Session_component::_process_packet_op()
[init -> lx_fs]  Watch_node::watch_respons()
[init -> lx_fs]  Session_component::_process_packet_op()
[init -> fs_rom] update() 
[init -> fs_rom] _try_read_dataspace()
[init -> lx_fs]  Watch_node::~Watch_node()
[init -> lx_fs]  Watch_node::Watch_node()
[init -> fs_rom] update() 
[init -> fs_rom] _try_read_dataspace()
[init -> lx_fs]  Watch_node::~Watch_node()
[init -> lx_fs]  Notifier::entry()::igonre_fn  watch cap count=1    <- from here on only ever one of the components gets notified
[init -> lx_fs]  Watch_node::Watch_node()
[init -> fs_rom] _try_read_dataspace()
[init -> lx_fs]  Watch_node::~Watch_node()
[init -> lx_fs]  Watch_node::watch_respons()
[init -> lx_fs]  Session_component::_process_packet_op()
[init -> lx_fs]  Watch_node::Watch_node()
```
I imagine, that this could also happen, if only one component would be watching the file, although with a lower probability.

I'll try to create a simple test scenario to reproduce it, as this is what I have found in our relative large scenario with a lot of things going on.

In the meanwhile have you any idea how it would be possible to work around the problem or even a potential solution strategy?

nfeske, 2021-04-23:
I'm unfortunately not able to dive right into the guts of this topic right now. It is too far off my current area of focus.

For further debugging, I would try the usual strategies: E.g., increasing the stress. Does the corner case occur more quickly with 10 watchers instead of 2? Or, cross-correlation. Does the corner case occur when swapping-out components (e.g., lx_fs by vfs) but mimic the rest of the scenario as is.

I'm wondering how should we best proceed? Do you see the remaining problems more as a limitation that could be documented? Or is it a showstopper?

My line of thinking is: Before your work, watching lx_fs was not possible at all. With your work, the watching works but apparently not 100% reliably. Overall, I perceive your commit series as a very welcome improvement and the corner case you describe is not a regression. When being up-front about the current limitation (documenting watching as experimental), I am not against merging the state as is. But this rather lax judgement comes from the perspective of my primary use cases, e.g., Sculpt, where lx_fs does not play crucial role anyway.

What do you think?

m-stein, 2021-04-23:
@trimpim I was very happy to see your commits regarding watch in lx_fs because my CBE manager (#4032) run script uses lx_fs and it would be nice to see file size updates when expanding the back end image. Unfortunately, when cherry-picking f1da7f97c4 9015761835 ab3db4d0d7 7180511f91 fd1ba22d18 to my topic branch, I got an "deadlock ahead" from the lx_fs. So, I created a new staging-based branch 4032_A_graphical_control_interface_for_the_CBE_lx_fs_deadlock (maybe your commits need a newer base) but it gives me the same:
```
[init -> back_end_fs] Error: deadlock ahead, mutex=0x7ffff7fe6578, return ip=0x414381
```

m-stein, 2021-04-23:
This is the backtrace of the deadlock in lx_fs:
```
#0  Genode::Mutex::acquire (this=this@entry=0x7ffff7fe6578)
    at /home/lypo/genodelabs/genode/repos/base/src/lib/base/mutex.cc:25
#1  0x0000000000414df1 in Genode::Mutex::Guard::Guard (mutex=..., this=<synthetic pointer>)
    at /home/lypo/genodelabs/genode/repos/base/include/base/mutex.h:62
#2  Genode::Id_space<File_system::Node>::Element::~Element (this=0x7ffff7f519a0, __in_chrg=<optimized out>)
    at /home/lypo/genodelabs/genode/repos/base/include/base/id_space.h:109
#3  File_system::Open_node<Lx_fs::Node>::~Open_node (this=0x7ffff7f519a0, __in_chrg=<optimized out>)
    at /home/lypo/genodelabs/genode/repos/os/src/server/lx_fs/open_node.h:64
#4  Genode::destroy<File_system::Open_node<Lx_fs::Node>, Genode::Heap&> (obj=0x7ffff7f519a0, dealloc=...)
    at /home/lypo/genodelabs/genode/repos/base/include/base/allocator.h:307
#5  Lx_fs::Session_component::close(Genode::Id_space<File_system::Node>::Id)::{lambda(File_system::Open_node<Lx_fs::Node>&)#1}::operator()(File_system::Open_node<Lx_fs::Node>&) const (__closure=<optimized out>, open_node=...)
    at /home/lypo/genodelabs/genode/repos/os/src/server/lx_fs/main.cc:414
#6  Genode::Id_space<File_system::Node>::Element::_for_each<File_system::Open_node<Lx_fs::Node>, Lx_fs::Session_component::~Session_component()::{lambda(File_system::Open_node<Lx_fs::Node>&)#1}>(Lx_fs::Session_component::~Session_component()::{lambda(File_system::Open_node<Lx_fs::Node>&)#1} const&) const (this=<optimized out>, fn=...)
    at /home/lypo/genodelabs/genode/repos/base/include/base/id_space.h:71
#7  0x00000000004258cc in Genode::Id_space<File_system::Node>::for_each<File_system::Open_node<Lx_fs::Node>, Lx_fs::Session_component::~Session_component()::{lambda(File_system::Open_node<Lx_fs::Node>&)#1}>(Lx_fs::Session_component::~Session_component()::{lambda(File_system::Open_node<Lx_fs::Node>&)#1} const&) const (fn=..., this=0x7ffff7fe6570)
    at /home/lypo/genodelabs/genode/repos/base/include/util/avl_tree.h:227
#8  Lx_fs::Session_component::~Session_component (this=0x7ffff7fe4018, __in_chrg=<optimized out>)
    at /home/lypo/genodelabs/genode/repos/os/src/server/lx_fs/main.cc:298
#9  0x0000000000409226 in Genode::destroy<Lx_fs::Session_component, Genode::Allocator*> (obj=0x7ffff7fe4018, 
    dealloc=<optimized out>) at /home/lypo/genodelabs/genode/repos/base/include/base/allocator.h:301
#10 Lx_fs::Root::_destroy_session (this=<optimized out>, session=0x7ffff7fe4018)
    at /home/lypo/genodelabs/genode/repos/os/src/server/lx_fs/main.cc:659
#11 0x000000000041a9f0 in Genode::Root_component<Lx_fs::Session_component, Genode::Multiple_clients>::close (
    this=0x4a8070 <Component::construct(Genode::Env&)::inst+80>, session_cap=...)
    at /home/lypo/genodelabs/genode/repos/base/include/base/native_capability.h:76
#12 0x0000000000419f9b in Genode::Meta::call_member<Genode::Meta::Empty, Genode::Typed_root<Lx_fs::Session_component>, Genode::Meta::Pod_tuple<Genode::Capability<Genode::Session>, Genode::Meta::Empty> > (
    func=&virtual Genode::Root::close(Genode::Capability<Genode::Session>), args=..., server=...)
    at /home/lypo/genodelabs/genode/repos/base/include/base/native_capability.h:62
#13 Genode::Root::Rpc_close::serve<Genode::Typed_root<Lx_fs::Session_component>, Genode::Meta::Empty> (args=..., 
    server=...) at /home/lypo/genodelabs/genode/repos/base/include/root/root.h:70
#14 Genode::Rpc_dispatcher<Genode::Typed_root<Lx_fs::Session_component>, Genode::Typed_root<Lx_fs::Session_component> >::_do_serve<Genode::Root::Rpc_close> (args=..., this=0x4a80f8 <Component::construct(Genode::Env&)::inst+216>)
    at /home/lypo/genodelabs/genode/repos/base/include/base/rpc_server.h:146
#15 Genode::Rpc_dispatcher<Genode::Typed_root<Lx_fs::Session_component>, Genode::Typed_root<Lx_fs::Session_component> >::_do_dispatch<Genode::Meta::Type_list<Genode::Root::Rpc_close> > (out=..., in=..., opcode=..., 
    this=0x4a80f8 <Component::construct(Genode::Env&)::inst+216>)
    at /home/lypo/genodelabs/genode/repos/base/include/base/rpc_server.h:180
#16 Genode::Rpc_dispatcher<Genode::Typed_root<Lx_fs::Session_component>, Genode::Typed_root<Lx_fs::Session_component> >::_do_dispatch<Genode::Meta::Type_list<Genode::Root::Rpc_upgrade, Genode::Root::Rpc_close> > (out=..., in=..., opcode=..., 
    this=0x4a80f8 <Component::construct(Genode::Env&)::inst+216>)
    at /home/lypo/genodelabs/genode/repos/base/include/base/rpc_server.h:201
#17 Genode::Rpc_dispatcher<Genode::Typed_root<Lx_fs::Session_component>, Genode::Typed_root<Lx_fs::Session_component> >::_do_dispatch<Genode::Meta::Type_list<Genode::Root::Rpc_session, Genode::Root::Rpc_upgrade, Genode::Root::Rpc_close> > (
---Type <return> to continue, or q <return> to quit---
    this=0x4a80f8 <Component::construct(Genode::Env&)::inst+216>, opcode=..., in=..., out=...)
    at /home/lypo/genodelabs/genode/repos/base/include/base/rpc_server.h:201
#18 0x000000000045546a in Genode::Rpc_entrypoint::<lambda(Genode::Rpc_object_base*)>::operator() (
    obj=0x4a8070 <Component::construct(Genode::Env&)::inst+80>, __closure=<synthetic pointer>)
    at /home/lypo/genodelabs/genode/repos/base/include/base/mutex.h:64
#19 Genode::Object_pool<Genode::Rpc_object_base>::apply<Genode::Rpc_entrypoint::entry()::<lambda(Genode::Rpc_object_base*)> > (func=..., capid=<optimized out>, this=0x4aab28 <Genode::bootstrap_component()::startup+520>)
    at /home/lypo/genodelabs/genode/repos/base/include/base/object_pool.h:147
#20 Genode::Rpc_entrypoint::entry() () at /home/lypo/genodelabs/genode/repos/base/src/lib/base/rpc_dispatch_loop.cc:68
#21 0x0000000000465a05 in thread_start (arg=0x57ee70)
    at /home/lypo/genodelabs/genode/repos/base-linux/src/lib/lx_hybrid/lx_hybrid.cc:400
#22 0x00007ffff75b26db in start_thread (arg=0x7ffff75aa700) at pthread_create.c:463
#23 0x00007ffff78eb71f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

m-stein, 2021-04-23:
It seems that the problem occurs when closing the session to the lx_fs, which is done in my scenario by destructing the client component (fs_query) via the sandbox lib.

trimpim, 2021-04-26:
@nfeske before the findings of @m-stein I would have agreed with you on integrating the feature. However @m-steins finding is a show stopper, and I'm working on it.

trimpim, 2021-04-26:
@m-stein fe8ba2f fixes the error when a client disconnects. Can you give it a try?

m-stein, 2021-04-26:
@trimpim Thanks for the fast fix. It works just fine with the CBE manager!

m-stein, 2021-04-26:
Just as a reminder: The lx_fs commits change the integration of lx_fs in a way that it now requires a Timer connection (wasn't the case beforehand). So, all scenarios with lx_fs have to be adapted accordingly.

m-stein, 2021-04-26:
@trimpim I try to grasp what are the reasons that led to using a Timer session in lx_fs. I understand that you want to limit rate and throughput with which the notify queue is processed. What I struggle with is drawing the connection between the signals that you submit in ``_process_notify`` and the packet ack queue that, according to your doc, will overflow if you don't. What are these signals (there seem to be two types "notify" and "ignore" as I learn from ``Lx_fs::Notifier::entry()``) and how do they affect the ack queue? Sorry, I'm not that deep into the lx_fs topic, but could you try explaining it to me?

trimpim, 2021-04-27:
@m-stein the signals that are sent in `_process_notify()` are captured by `_signal_handler` in `watch.h` this indirectly calls `Session_component::_process_packet_op()`. This via `notify_listenders()` of the node calls `Listener::notify()` in `repos/os/include/filesystem/listener.h`. Here the packets are sent to the client if the ACK queue isn't full.

The Linux kernel provides a lot of different notifications within the `inotify` framework. `IN_MODIFY` is delivered when the file is modified in place, `IN_IGNORED` is delivered when a file is (re)moved (also when vim uses backup files).

I will have a look at the run scripts that use `lx_fs` and check that a timer is provided.

nfeske, 2021-05-05:
I added the commit series from the trimpim/1120-lx_fs-watch branch to staging now.

While looking through the commits, I wondered if f9d44a30ea55704822bebd3150212e30f3d93189 could be squashed into 358e1c92edcb66211974262cf6cf9883b4157ed9?

trimpim, 2021-05-05:
@nfeske I'm currently running the hopefully last stress test on a further fixup commit to the series. I would agree to squash the commits, just kept them separate for easier review.

I can also squash the commits before I push, if you prefer.

nfeske, 2021-05-05:
> I can also squash the commits before I push, if you prefer.

At the commit series is already on staging, I prefer additional fixups to avoid force-pushing staging too often. Please just tell me your squashing preferences :-)

trimpim, 2021-05-06:
@nfeske my stress test passed, so with a1347b2 `lx_fs` now also survives very fast changes on a watched file.

I would squash fd1ba22d18d54885df7e24bfb7ec0ff390ddf931, fe8ba2f71e8789ec7dca329a7fed45274addbe50 and a1347b280d8b8e440703a958008268e6040fe15c in to 90157618358f51148275f9e4fb50aed31b20259a, or as you see fit.

chelmuth, 2021-05-07:
@trimpim it's overdue that I join the discussion here because I agreed with @nfeske to attend to this issue a week ago but did not want to rush things and had other priorities too. As my review of your contribution is still pending and your latest fixup changes pretty many lines, I suggest to start over. Could you please prepare a new topic branch with your commits stashed as you consider appropriate? I'll remove the patch series from staging straight after finishing this posting and look forward to review your topic branch over the next week.

In the meantime I'd like to suggest to reconsider the decision to use a Timer session from lx_fs. Isn't it feasible to use Linux mechanisms for steering the rate limiting as lx_fs is a hybrid component anyway? I'm asking because I'm convinced that we should hide this aspect (timing requirement in a file system) from the integration level (run script, packaging).

chelmuth, 2021-05-12:
If you do not have pressing needs to include this feature soon, just take your time. I'll look into it again when you notify updates in this issue.

trimpim, 2021-05-27:
@chelmuth While implementing the changes to remove the dependency on a timer session, by using `setitimer()` and libc signal handling, I encountered some problems.

- At first, the Linux kernel chooses one of the threads of a program to deliver `SIGALRM` to. This led to deadlocks when I uses `signal()` to setup the signal handler, as the handler could be called while it was still running.

- Then I tried to use `sigaction()` which by default sets the signal handler to be not delivered while it still is running. With this I had the problem, that the Genode signal which I use to trigger the entrypoint (`Signal_transmitter { _notify_handler }.submit();`) not to be received.

- Next I tried to block the signal in all threads (`sigprocmask()`) and create a separate thread that uses `sigwait()` to receive the signal. With this `lx_fs` terminated without a error message. It turns out, that the signal is now delivered to the startup code of `lx_hybrid` which, if I understand it correct, starts `Component::construct()` in a new thread.

As a workaround I added the code below to `lx_hybrid_init()` in `repos/base-linux/src/lib/lx_hybrid/lx_hybrid.cc`

```c++
	sigset_t restrict { };

	if (sigprocmask(SIG_BLOCK, nullptr, &restrict) != 0) {
		Genode::raw("111111111111111111111");
	}

	sigaddset(&restrict, SIGALRM);
	if (sigprocmask(SIG_BLOCK, &restrict, nullptr) != 0) {
		Genode::raw("222222222222222222222");
	}
```

This works, when I start the run script with `strace -ff` but fails without it (both on `x86_64` and `arm_v7a`).

I'm a bit lost on what I could try next.

chelmuth, 2021-05-27:
Thanks for your intermediate results. Can you share your implementation too? From what I still remember about signal handling on Linux your approach sounds reasonable, but my knowledge is somewhat antiquated and may need a polish.

trimpim, 2021-05-27:
@chelmuth I pushed it as it is, with all the commented out stuff from my experiments.

chelmuth, 2021-05-31:
I must admit that I'm not 100% sure I got your requirements for the timing feature. Can you confirm that you like to schedule operations to adhere to some rate limiting?

From your commits I can see you already spawn an additional pthread. Could this notifier thread use `pthread_cond_timedwait` for timing and external unblock in some way?

chelmuth, 2021-05-31:
I'd also suggest change the implementation like follows.

```c++
void Notifier::trigger_notify()
{
  _notify_handler.local_submit();
}
```

chelmuth, 2021-05-31:
I have another idea after more investigation and significant pondering on the threading/execution model of the watch feature: Is there a reason against using `select()` or `poll()` with timeout in the notifier thread? If I'm not mistaken, the notifier could decide easily if the current inotify event requires immediate or delayed notification in the file-system session.

trimpim, 2021-06-01:
> I must admit that I'm not 100% sure I got your requirements for the timing feature. Can you confirm that you like to schedule operations to adhere to some rate limiting?

Yes, this is to limit the number ACKs sent when a lot of changes on the filesystem happen at the "same" time. We observed during testing, that some "CONTENT_CHANGED" ACKs got lost, if multiple components watched files via a `fs_rom` and many changes to the filesystem.

> From your commits I can see you already spawn an additional pthread. Could this notifier thread use `pthread_cond_timedwait` for timing and external unblock in some way?

I will try this.

> I'd also suggest change the implementation like follows.
> 
> ```c++
> void Notifier::trigger_notify()
> {
>   _notify_handler.local_submit();
> }
> ```

I will change this, I wasn't aware of `local_submit()`.

> I have another idea after more investigation and significant pondering on the threading/execution model of the watch feature: Is there a reason against using `select()` or `poll()` with timeout in [...]

If I understand you correct, the idea would be to wait for notifications with a timeout, and move the whole management of the pending notifications tho the thread. I like this, then we wouldn't need a separate timer and the handling for it, which simplifies the code.

trimpim, 2021-06-02:
@chelmuth your idea with the select worked out very well, all my stress tests passed.

chelmuth, 2021-06-02:
Very pleased to hear that. Do you think it's already time for review?

trimpim, 2021-06-02:
Yes, I think it is ready for review.

chelmuth, 2021-06-04:
I commented the *watch* commit inline and also merged 45aab0af2946eb4491b147d28c9760527789fabf as well as 264a994712b078169c7decc3f2912e08be90527d to staging.

chelmuth, 2021-06-08:
> @chelmuth I had to use `Signal_handler::local_submit ()` as a direct call to `watch_response()` did lead to sporadic segmentation faults.

I suspected that there may be dragons in this lair, but do you have more details about the *segmentation faults*? I must admit my file-system server skills are a bit dusted, but if *dangling references* are the cause of the faults, the implementation may have life-time management issues.

trimpim, 2021-06-08:
@chelmuth 

There is a switch in threads from the thread handling `inotify` to the components entrypoint.

Here is the output from GDB on the target.

```
/opt/gapfruit# gdb lx_fs /mnt/crash/corefiles/2021-06-07_old/lx_fs.3889.11.core 
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-buildroot-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from lx_fs...done.

warning: exec file is newer than core file.
[New LWP 3929]
[New LWP 3910]
[New LWP 3903]
[New LWP 3934]
[New LWP 3944]
[New LWP 3889]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/libthread_db.so.1".
Core was generated by `[Genode] init -> config_fs'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00028974 in File_system::Node_base::mark_as_updated (this=0xb5e08910) at /data/genode/repos/os/include/file_system/node.h:64
64	/data/genode/repos/os/include/file_system/node.h: No such file or directory.
[Current thread is 1 (Thread 0xb5cff460 (LWP 3929))]
(gdb) bt
#0  0x00028974 in File_system::Node_base::mark_as_updated (this=0xb5e08910) at /data/genode/repos/os/include/file_system/node.h:64
#1  Lx_fs::Watch_node::watch_response (this=0xb5e08910) at /data/genode/repos/os/src/server/lx_fs/watch.cc:57
#2  0x00028304 in Genode::strlen (s=0xb5e08a40 "") at /data/genode/repos/base/include/util/string.h:561
#3  Genode::Cstring::Cstring (str=0xb5e08a40 "", this=<optimized out>) at /data/genode/repos/base/include/util/string.h:561
#4  Lx_fs::Notifier::entry (this=0xb5cbadb4) at /data/genode/repos/os/src/server/lx_fs/notifier.cc:337
#5  0xb6e9ba76 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

What looks strange to me upon closer inspection of the backtrace, is that from printing an error in `entry()` (the line number is off by one compared to the last patch), the execution continues in `watch_response()` which doesn't get called in this case.

chelmuth, 2021-06-08:
Doesn't `warning: exec file is newer than core file` make the core interpretation impossible?

nfeske, 2021-06-08:
Maybe this problem is related to https://github.com/genodelabs/genode/issues/4092?

trimpim, 2021-06-08:
@chelmuth this warning is due to the fact, that I did rebuild the `lx_fs` binary anew to get the backtrace, as I had already started on a new issue (I checked out the same commit again).

@nfeske  this might be indeed related.
The way I implemented it currently is to send a signal to the entrypoint from the thread that "listens" to the `inotify` events. With my stress tests run for around 8 hours whereas with calling `watch_response()` directly fails after about 2 hours.

trimpim, 2021-06-08:
@chelmuth yes, 0646a9e can be considered for merging.

chelmuth, 2021-06-08:
Merged 5c93b4cce32eb503c74dd3c580c6a834b251b55f and fixup 11707c6a2b981054e1dd838d297807ac1889402f.

chelmuth, 2021-06-09:
The commit broke the *lx_fs* depot archive build.

```
In file included from /data/depot/genodelabs/src/lx_fs/2021-06-08/src/server/lx_fs/watch.h:23,
                 from /data/depot/genodelabs/src/lx_fs/2021-06-08/src/server/lx_fs/watch.cc:17:
/data/depot/genodelabs/api/os/2021-05-27/include/os/vfs.h:20:10: fatal error: vfs/simple_env.h: No such file or directory
   20 | #include <vfs/simple_env.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

The error made me wonder why lx_fs needs the VFS anyway. I fixed the issue by removing the artificial dependency to *vfs/vfs_handle.h* that was introduced only to reuse the `Watch_response_handler` interface in 8e74a029d4d56034dad2365895bad47e1f38bc61.

commit 5c93b4cc (2021-06-08): lx_fs: enable watch

Enable watching files via the inotify interface of the Linux Kernel.

Delivery of watches to components is staggered in order to prevent an
overflow of the ACK queue in cases when a lot of changes are made to the
file system from the Linux side.

Fixes #4070
commit 7db6f457 (2021-06-25): lx_fs: enable watch

Enable watching files via the inotify interface of the Linux Kernel.

Delivery of watches to components is staggered in order to prevent an
overflow of the ACK queue in cases when a lot of changes are made to the
file system from the Linux side.

Fixes #4070
closed 2021-06-25 by chelmuth