genodelabs/genode #4071: ldso: review the usage of Mutex in `dl_iterate_phdr`

ssumpf, 2021-04-09:
The mutex in `dl_iterate_phdr` prevents further function resolution in `callback` functions, if a function has not been relocated at this point this leads to a deadlock.

ssumpf, 2021-04-09:
695368f should be merged with 6a016dfc25f99bd68e31c92d3c25ffa2d15f26b7

nfeske, 2021-04-09:
Thanks. I squashed both commits to https://github.com/nfeske/genode/commit/c45d964bc23f4f0f370c74d8a1290c88fe974876 on staging. @ssumpf can you double-check the commit?

ssumpf, 2021-04-09:
> Thanks. I squashed both commits to [nfeske@c45d964](https://github.com/nfeske/genode/commit/c45d964bc23f4f0f370c74d8a1290c88fe974876) on staging. @ssumpf can you double-check the commit?

Looks good to me.

ssumpf, 2021-04-14:
@nfeske: 5dcee7e is a follow up to c45d964 because the latter caused a deadlock when dynamically loaded libraries (e.g., `dlopen`, `Shared_object`) threw exceptions during ELF setup. Please consider for staging.

nfeske, 2021-04-14:
Thank you, merged to staging. The reworked version of the code looks so much better.

commit c45d964b (2021-04-09): ldso: allow symbol resolution during dl_itera_phdr

When callback functions of `dl_iterate_phdr` required further jump slot
relocations this lead to a deadlock. Therefore, we allow the resolution
of further symbols from callback functions, but protect the ELF object
list during the iteration, which blocks any dynamic loading (e.g.,
dlopen/dlcose) of shared object by other threads while in program header
iteration.

fixes #4071
closed 2021-04-20 by nfeske