genodelabs/genode #4073: tool/run: minor issues

tomga, 2021-04-14:
During my experiments I found two issues in run script that I cannot workaround easily without modifications to the script itself, so I'd like to ask for consideration to accept patches from my [run_tweaks](/tomga/genode/commits/run_tweaks) branch.

First commit (e2f92c346460a38916de94ef322a76876e8c7940) allows to implement retrieval of build targets of run scripts without actually performing a build as suggested briefly in #4004 in this [comment](/genodelabs/genode/issues/4004#issuecomment-780527605). With this commit in place I can achieve result by replacing some functions from an additional plugin loaded by `run` tool. Without it `run` tool breaks if the file `init.xsd` does not exist.

Second commit (c3f629b0e01ca351a648eb9c0ccb305b6918dc3d) is a proposal (not full yet - I'll extend it if this path is accepted) that moves calls to copy files using `file copy -force` commands to a dedicated function. It allows to implement workaround for copying relative symbolic links. Currently all symbolic links created during build are absolute. If such links are copied using mentioned function they stay absolute and point to proper (original) files. In my Scons based build attempt I tried to avoid any references to absolute paths outside of Genode directory and implemented symbolic links to be relative. If such link is copied using `file copy -force` it points to exactly the same relative path but not to original file and it breaks further `run` script processing.

Would such changes be accepted? If not, do you see a better approach for my problems?

tomga, 2021-04-20:
During past week I finished my alternative Genode build system to somewhat consistent state and set up buildbot for easy testing for regressions so I decided that it is a good moment to make this code public. `run` tool plugins mentioned here earlier are:
 * [run_build_targets](/tomga/genodebuildtool/blob/master/tool/run_build_targets)
 * [run_file_copy](/tomga/genodebuildtool/blob/master/tool/run_file_copy)

To work they require changes in Genode for which I asked for opinion if they can be incorporated. Maybe with those plugins available fore review it will be more clear why changes are needed for my use case. If proposed changes will not be accepted then I'll try to find some other solution that will not require any changes in Genode tree.

Both those proposed changes are safe in my opinion. First changes execution path that currently cannot occur because if it did then it would cause tcl exception. And second is only a refactoring that moves call to tcl `file` procedure used to copy files to a dedicated procedure.

nfeske, 2021-04-21:
Sorry for the late response. I agree that both commits should do no harm.

Regarding the `file_copy` commit, there may be the risk of new instances of `file copy -force` being added in the future because the use of `file_copy` is not enforced. There are several run scripts that use `file copy -force`. Are they causing trouble for you? If yes, we should seek a way to enforce that `file copy -force` cannot be used in run scripts.

tomga, 2021-04-25:
@nfeske thanks for merging those commits.

In ffdba74a9ff1edf3c418b7652c43128b7dec0df1 I replaced rest of `file copy` operations. All of them but one had `-force` parameter. Only `repos/ports/run/verify.run` contained a `file copy` without `-force` parameter. I assumed that it was more an omission than done on purpose. Please verify if I'm not mistaken.

Given your question about looking for a way for enforcing not using `file copy -force` I wouldn't pay much attention to it right now unless you know some easy way to achieve it. Currently support for `run` scripts in my SCons based build is in initial stage and I have some more important problems to solve there about which I'll definitely need to ask for advice - probably in next days I'll be ready to formulate my doubts.

nfeske, 2021-04-27:
After thinking once more about this change, would you mind naming the function `copy_file` instead of `file_copy`? This name would fit more naturally with the naming of the existing functions (`create_boot_directory`, `install_config`, `run_genode`, etc.).

I'm sorry for coming up so late with this suggestion. Before you touched the run scripts, I admittedly wasn't fully aware of the public visibility of this change.

tomga, 2021-04-27:
No problem at all. I'll prepare this change but not earlier than tomorrow evening.

> I admittedly wasn't fully aware of the public visibility of this change.

Actually me either. Only after your previous comment I realized that my `grep` command I used for searing for `file copy` calls excluded `*.run` files completely.

tomga, 2021-04-29:
bf2dc609433b47e784994ac82c12311ade6c523c is a fixup commit that renames `file_copy` to `copy_file`. I think that it should merge on current staging along with previous one without problems.

nfeske, 2021-04-30:
Thanks @tomga. I merged both fixups to staging now.

nfeske, 2021-05-05:
The commits entered the master branch today. Shall we close this issue?

tomga, 2021-05-05:
I have some questions about discrepancy between build targets specified in run scripts and real targets needed by same run scripts to put into an image. I know cases where scripts require building more than they actually need (it was quite irritating for me when I worked on port for rpi, without configured ccache) and recently I also found case where build was performed successfully but image could not be assembled due to missing build targets.

I planned to discuss it here and possible solutions due to the fact that topic is about run scripts but can move it to a new issue or #3978 which already transformed into general discussion about build system design.

If those problems are known (I suppose they are) and you have some ideas for fixing them I'd like to here about them please. If not then I'll describe in more detail examples which exhibit wrong behavior in next few days.

nfeske, 2021-05-06:
I'm fine with keeping this issue open as a place for further discussion.

tomga, 2021-08-14:
It took me more than I expected but finally I'm back to it.

#### Testcase
```
tool/create_builddir x86_64 BUILD_DIR=build/test
```
Enable `libports`, `ports`, `dde_linux`, `dde_rump` and `gems` repositories in `build/test/etc/build.conf`. It is important to not enable `dde_ipxe` repostory.
```
cd build/test
make BOARD=pc KERNEL=nova run/ssh_terminal

...

make[1]: Leaving directory '/projects/genode/genode/build/test'
genode build completed
using 'core-nova.a' as 'core.a'
using 'ld-nova.lib.so' as 'ld.lib.so'
using 'nova_timer_drv' as 'timer'
error copying "bin/ipxe_nic_drv": no such file or directory
    while executing
"file copy -force $src $dst"
    (procedure "copy_file" line 2)
    invoked from within
```
Command finishes with error during preparing image to run scenario after the build has finished successfully.

#### Problem description

Build components contains build `drivers/nic` target to be built which is supposed to build `ipxe_nic_drv` which is expected to exist after successful build. But `drivers/nic` target does not specify for build system that `ipxe_nic_drv` is to built. It specifies that all program targets that reside in `src/drivers/nic` directories in all available repositories should be built (with limitation to required specs but it is not important here). Build succeeds as there is one such target: `os/src/drivers/nic/virtio/spec/x86/target.mk`. As far as I know there is currently no possibility to ask for build of specific program binary - only such groups of targets with common paths.

Despite of needed binary not being built it can be seen that `virtio_pci_nic` that was not needed for this scenario has been built. So current behavior is that:

 * there is no possibility to ask for building concrete program

 * unrelated build targets (related only in being in similar subdirectories) are built, which makes builds longer and potentially may cause build errors from components not needed in workflow being worked on.

#### Solution proposal

I think that there should be a way to ask for specific programs to be built. Given the proposal from #3975 to allow specifying library targets with `lib` prefix I'd propose to handle `prg` prefix for building specific programs. In this case list of components to be built would have to contain `prg/ipxe_nic_drv` target instead of generic `drivers/nic`. Current way of specifying program build targets can stay like it is now so it shouldn't generate any incompatibility.

Such change would also need to solve a problem that currently it is not easy to find rules file (`tarket.mk`) that is responsible for building a specific target (it is specified inside a `target.mk` itself or included from it). So build system after receiving a target in new form wouldn't be able to quickly find proper `target.mk` file. To resolve this I'd suggest a change so that make filename would contain target name in it like it is for library targets. I'd propose file names of form: `<target_name>.prg.mk`. Build system could easily support both variants: old `target.mk` and new `<target_name>.prg.mk` for transitional period so there would be no pressure for external parties that have their own repositories to quickly update to new form.

With those change in place it would be possible to write consistently build requirements and files for final image in run scripts - possibly even from a single list of _requirements_.

nfeske, 2021-08-16:
The placement of the ipxe NIC driver directly under `dde_ipxe/src/drivers/nic/` is a little unfortunate. In most other cases, the individual drivers are located in appropriately named sub directories in order to allow the build-system user to single them out. E.g., `make drivers/framebuffer/vesa` builds the VESA framebuffer driver but no other framebuffer driver.

It is probably a good idea to apply this convention more rigorously. In the particular case, the ipxe NIC driver should better be moved to `dde_ipxe/src/drivers/nic/ipxe/`. The build system can be left unchanged.

tomga, 2021-08-16:
@nfeske, thank you for quick response. It seems that this wasn't the best example for what I wanted to address. It showed a special case for which you provided a solution. But I wanted to present a pattern in run scripts for drivers where a _group_ target is added to `build_components` and concrete binary to `boot_modules`. And it can cause similar issues to those in the testcase I provided. It is for example for:
 * `drivers/usb_host` and `[usb_host_drv_binary]
 * `drivers/audio` and `[audio_drv_binary]`
 * `drivers/framebuffer` and `fb_sdl, fb_drv, intel_fb_drv`

I think that with proposed changes to build system it would be possible to provide in run scripts only a list for `boot_modules` and this list could be used as targets for build. It could simplify run scripts and avoid inconsistencies between build targets and boot modules.

On the other hand I understand your reluctance to make invasive changes to build system (with all consequences like tests, documentation, etc.).

Today I found that in one case, the platform driver, where in addition to `platform_drv_binary` TCL function there is `platform_drv_build_components` function which is supposed to provide proper build components for required platform driver binary.

Extending this pattern for other driver categories, if applied consistently, would allow to eliminate the discrepancy between build targets and boot modules. It would still require to provide two distinct values for each driver but the risk of building too much or not building the required target could be eliminated. Does it seem like a good idea?

nfeske, 2021-08-17:
I don't share your perception of the "discrepancy" between build targets and boot modules being a problem. As there is no one-to-one relationship between build artifacts and boot modules, the aspiration to bring both under one umbrella by introducing new formalities is not a direction I would consider.

To substantiate my perspective, I see four categories of Genode scenarios:

1. Tests for basic features that are mostly hardware-agnostic. The selection of board-specific device drivers is not a concern here.

2. Test scenarios that fall into the broad categories network appliance or interactive, which are accommodated by the board-specific drivers_interactive or drivers_nic packages respectively. These packages wrap up and hide board-specific peculiarities (driver selection, platform-driver configuration) from the run script. For those, your concern does not apply.

3. Complex scenarios that rely on a mix of drivers. (e.g., networking + block device) Here we leave the bare-bones Genode territory. Such scenarios are best covered by pkg archives for the deployment on top of Sculpt OS. For the creation of such scenarios, Goa should ultimately become the tool of choice - presenting an alternative to run scripts. I sense that your suggestion addresses this category of scenarios that still rely on (legacy) run scripts instead of having the form of deployable packages for Sculpt OS.

4. Special cases like test scenarios for a specific driver, or sophisticated integration tests (the vbox tests come in mind), or a concrete hardware-software appliance. Such scenarios are mostly tied to specific hardware (e.g., a PC with a specially prepared disk, or a concrete board with certain peripherals attached). Still, in some cases, such scenarios can solely rely on depot content and do not even contain a build step and leave the boot-modules list empty (looking at sculpt.run).

The current tooling covers each of those categories reasonably well. The third category - the leveraging of Sculpt OS as hardware-abstraction - is certainly most worthwhile to nurture when going forward. That's why I'm repeatedly advertising Goa as direction, which is actually designed to marginalize both Genode's build system and the run tool in the long run. Goa is presently still underused and underdeveloped. When it comes to contributing to Genode's tooling, using, improving, and maturing the Goa tool would be the most welcome area of development.

tomga, 2021-08-17:
Thank you for detailed response. I feel convinced now that unifying boot modules with build targets wouldn't be a good idea.

I still think that current situation in some scripts that can lead to successful build without required targets being actually built is wrong and that more targets than needed are actually built is unfortunate and in my last post I proposed to provide a fix for this situation. But I accept that you treat run scripts and run tool itself as legacy and that you want to leave it like it is now.

I think now that there is no reason to keep this issue open so I'm closing it now.

closed 2021-08-17 by tomga

nfeske, 2021-08-18:
> But I accept that you treat run scripts and run tool itself as legacy...

Please don't get me wrong. I do *not* regard run scripts and the run tool as a legacy in general. With the term "legacy" I referred to the use of run scripts for scenarios of the third category - the ones that can better be accommodated by Sculpt OS. To illustrate what I mean when speaking of "marginalization" of the run tool, let's recap the history of our low-level tests. During the transition of Genode's low-level tests to depot recipes (driven by depot_autopilot.run or test.run), more than 50 run scripts vanished. So the role of run scripts diminished. I foresee the same story repeating for scenarios of the third category.