genodelabs/genode #4076: dde_linux: pci device matching ignores subsystem id

jschlatow, 2021-04-16:
On my T490s I stumbled upon a potential issue in [lx_emul/impl/pci.h](https://github.com/genodelabs/genode/blob/master/repos/dde_linux/src/include/lx_emul/pci.h). I was wondering about the driver logging a wrong wifi card.

Currently, the driver's id table is iterated for every PCI device in order to find a matching entry based on the device id.
Yet, for wifi, the driver's id table distinguishes cards (and what firmware to load) by their subsystem id (aka subdevice id). By ignoring the subdevice id, the last entry that matches the device id is currently chosen. This explains why the driver logged an Intel Wireless-AC 9462 (subdevice id 0x42a4) instead of a 9560 (subdevice id 0x0034). In this particular case, both devices are associated with the same firmware but, looking at the id table, there are cases in which the subdevice id matters.

jschlatow, 2021-04-16:
Commit 2dfd7c3 contains a straightforward fix in lx_emul.

nfeske, 2021-04-20:
Merged as https://github.com/genodelabs/genode/commit/685fd95f172de2e251f1437657a0846f72233cfb. I slightly shortened the first line of the commit message to adhere to the usual 50-character limit.

closed 2021-05-05 by nfeske